Ysurac/openmptcprouter #1614: Slow or offline ipv4 and/or ipv6 during couple of seconds

xillibit, 2021-02-21:
## Expected Behavior

[When enabling IPv6](https://github.com/Ysurac/openmptcprouter/issues/1147), should not make offline IPv4 time to time

## Current Behavior
Time to time when i try to access diffrents websites it's slow to load the page, then it end with that : 

[image]

If i reload one time or more the page it works

When it does that if i check on https://ip.lafibre.info/ when i can access the page, it shows IPv4 offline

It's slow in some cases to access the web interface of openmptcprouter, it loads it in endless way

## Possible Solution

## Steps to Reproduce the Problem

Enable IPv6 on openmptcprouter : https://github.com/Ysurac/openmptcprouter/issues/1147#issuecomment-782806028 

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: 0.1024
  - OpenMPTCProuter platform: x86_64 (VM on Proxmox 6.3)

Ysurac, 2021-02-21:
Check State->System log.
Check also when you have the issue if you don't have any issue on VPS side.

xillibit, 2021-02-21:
I wasn't not able to access OMR by openmptcprouter, i need to access by 192.168.200.1. Under State -> System log i have that : 

> Sun Feb 21 20:38:04 2021 user.notice Shadowsocks: Rules DOWN
Sun Feb 21 20:38:10 2021 daemon.err /usr/bin/ss-local[15503]: getpeername: Socket not connected
Sun Feb 21 20:38:20 2021 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 212.XX.XX.XX)
Sun Feb 21 20:38:20 2021 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Sun Feb 21 20:38:20 2021 user.notice Shadowsocks: Rules UP
Sun Feb 21 20:38:20 2021 user.notice Shadowsocks: Reload omr-bypass rules
Sun Feb 21 20:38:20 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Feb 21 20:38:21 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Feb 21 20:38:21 2021 daemon.info dnsmasq[14861]: read /etc/hosts - 4 addresses
Sun Feb 21 20:38:21 2021 daemon.info dnsmasq[14861]: read /tmp/hosts/odhcpd - 0 addresses
Sun Feb 21 20:38:21 2021 daemon.info dnsmasq[14861]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sun Feb 21 20:38:21 2021 daemon.info dnsmasq-dhcp[14861]: read /etc/ethers - 0 addresses
Sun Feb 21 20:38:21 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Feb 21 20:38:36 2021 daemon.err /usr/bin/ss-local[15503]: getpeername: Socket not connected
Sun Feb 21 20:38:44 2021 daemon.err /usr/bin/ss-local[15503]: getpeername: Socket not connected
Sun Feb 21 20:41:53 2021 daemon.info dnsmasq-dhcp[14861]: DHCPDISCOVER(eth1) MAC_ADDRESS
Sun Feb 21 20:41:53 2021 daemon.info dnsmasq-dhcp[14861]: DHCPOFFER(eth1) 192.168.200.118 MAC_ADDRESS
Sun Feb 21 20:41:53 2021 daemon.info dnsmasq-dhcp[14861]: DHCPREQUEST(eth1) 192.168.200.118 MAC_ADDRESS
Sun Feb 21 20:41:53 2021 daemon.info dnsmasq-dhcp[14861]: DHCPACK(eth1) 192.168.200.118 MAC_ADDRESS android-ab690d03949e7416
Sun Feb 21 20:42:35 2021 daemon.err /usr/bin/ss-local[15503]: getpeername: Socket not connected
Sun Feb 21 20:43:05 2021 daemon.err /usr/bin/ss-local[15503]: getpeername: Socket not connected
Sun Feb 21 20:44:17 2021 daemon.err /usr/bin/ss-local[15503]: getpeername: Socket not connected
Sun Feb 21 20:44:44 2021 daemon.err uhttpd[6121]: luci: accepted login on / for root from 192.168.200.128
Sun Feb 21 20:45:05 2021 daemon.err /usr/bin/ss-local[15503]: getpeername: Socket not connected
Sun Feb 21 20:45:06 2021 daemon.err /usr/bin/ss-local[15503]: getpeername: Socket not connected
Sun Feb 21 20:45:08 2021 daemon.err /usr/bin/ss-local[15503]: getpeername: Socket not connected
Sun Feb 21 20:45:12 2021 daemon.err /usr/bin/ss-local[15503]: getpeername: Socket not connected
Sun Feb 21 20:45:13 2021 user.notice post-tracking-post-tracking: wan2 (wan2) switched off because ping from 192.168.8.2 error (4.2.2.1,8.8.8.8,80.67.169.12)
Sun Feb 21 20:45:13 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.8.1 dev wan2
Sun Feb 21 20:45:14 2021 daemon.err /usr/bin/ss-local[15503]: getpeername: Socket not connected
Sun Feb 21 20:45:14 2021 daemon.err /usr/bin/ss-local[15503]: getpeername: Socket not connected
Sun Feb 21 20:45:16 2021 daemon.err /usr/bin/ss-local[15503]: getpeername: Socket not connected
Sun Feb 21 20:45:23 2021 user.notice post-tracking-post-tracking: wan2 (wan2) switched up
Sun Feb 21 20:47:05 2021 daemon.info dnsmasq-dhcp[14861]: DHCPDISCOVER(eth1) MAC_ADDRESS
Sun Feb 21 20:47:05 2021 daemon.info dnsmasq-dhcp[14861]: DHCPOFFER(eth1) 192.168.200.118 MAC_ADDRESS
Sun Feb 21 20:47:05 2021 daemon.info dnsmasq-dhcp[14861]: DHCPREQUEST(eth1) 192.168.200.118 MAC_ADDRESS
Sun Feb 21 20:47:05 2021 daemon.info dnsmasq-dhcp[14861]: DHCPACK(eth1) 192.168.200.118 MAC_ADDRESS android-ab690d03949e7416

Ysurac, 2021-02-21:
I don't understand what you mean with "I wasn't not able to access OMR by openmptcprouter, i need to access by 192.168.200.1".
In the log I can see a problem with wan2, and maybe a bigger problem with shadowsocks before.

On VPS side you should check if you can do some ping in IPv4 and IPv6 for example. You can also check from router if you can ping it.

xillibit, 2021-02-21:
> I don't understand what you mean with "I wasn't not able to access OMR by openmptcprouter, i need to access by 192.168.200.1".

To access the OMR interface in my browser habitually it works by just using openmptcprouter, but now the browser try to load the page in an endless way. So i need to goes by taping the IP directly

It display that sometime then at next reload the error disappear and then the couple of minutes after the errors are back

[image]

xillibit, 2021-02-21:
On router it don't want to ping IPv6 address, i have tried with one of IPv6 of google : 

[image]

On VPS i can ping without issue the same IPv6

Ysurac, 2021-02-22:
Try to change master interface.
What is the VPS provider ?
Your connections are LTE, ADSL, fiber ? there is no wifi ?

xillibit, 2021-02-22:
I have changed the master interface
My VPS provider si scaleway
I have an ADSL connection (SFR-BOX) and an another connection with Huawei 4G router. No i don't have connection with wifi, all is connected by cable

Ysurac, 2021-02-22:
By default, when IPv6 is enabled in the wizard an ULA IPv6 (fdxx:...) is distributed on the network. If your OS doesn't block ULA IPv6 this will work.
Can you check the result of ```ping -I 192.168.10.2 -B 51.158.xxxxx``` and ```ping -I 192.168.8.2 -B 51.158.xxxxx``` from the router and if ping is stable on both ?

xillibit, 2021-02-26:
When VPN doens't answer : 

[image]

Result of ping : `ping -I 192.168.10.2 -B 51.158.xxxxx` https://gist.github.com/xillibit/586f63993c6c57410cdd83f84a4dcb43

Result of ping : `ping -I 192.168.8.2 -B 51.158.xxxxx` https://gist.github.com/xillibit/d5f4f91e02d12ada72385ae74ec20f8b

Ysurac, 2021-02-26:
Sometimes high latency... It's why there is the message, too much latency.
Seems to be on both connections, so I would say a problem with the VPS.

xillibit, 2021-02-26:
I'am using a DEV1-S instance on scaleway but it's shared, maybe a neighboord on the host server

Ysurac, 2021-02-27:
I have a DEV1-M instance on scaleway with no issues, but yes can be due to neighbour or a bad host.
You can try a ping from the VPS to your public WAN IP and to another IP to check if it's really the problem.

xillibit, 2021-03-02:
I can't ping the IP WAN of the 4G access (first try in the image), the ping on SFR ADSL IP WAN access works : 

[image]

xillibit, 2021-03-03:
It was slow since couple of hours, so i decided to disable ipv6 in wizard and now the issue has disappear : 

[image]

closed 2021-06-07 by github-actions[bot]