Ysurac/openmptcprouter #1615: ADSL latence

Bug-Storm, 2021-02-22:
Salut.. 

J'ai 2 connexion sur mon Openmptcp, une ADSL et l'autre 4G.

Le souci c'est qu'avec l'adsl j'ai un latence que peut dépasser les 200ms voir 300ms pendant quelques secondes, et cela fait que pour le Openmptcp l'adsl est down, je voudrais savoir si c'est possible de changer la latence par défaut avant d'un switch vers la 4G?

Et puis je ne comprends pas, mais même quand la 4G est en "backup" j'ai le Openmptcp qu'utilise bcp la 4G, c'est comme si j'avais laissé la 4G en mode "activé" alors que sur les options j'ai bien l'adsl en Master et la 4G en backup.. Juste que vous puissiez avoir une notion, je consomme au moins 5Gb par jours alors que la plupart du temps la 4G est Off.. Comment peut on faire ? 

Merci 
## Specifications

  - OpenMPTCProuter version:  Raspberry Pi 4 Model B Rev 1.1 
  - openmptcprouter v0.57.3 r0+15225-bfc433efd4

Bug-Storm, 2021-02-22:
Je viens de laisser le `Planificateur Multipath TCP ` en mode `redondant ` pour pouvoir atteindre une latence Assez bas et ça a l'air de fonctionner 😉.

Bug-Storm, 2021-02-22:
Et pour la 4G, j'avais oublié que j'avais mis le test pour la passerelle en dns, cela explique le fait de trop consommer, je viens de modifier et de remettre en ping, on verra dans la fin de journée

Kalimeiro, 2021-02-22:
Bonjour,

Je crois qu'il y a bcp de confusion technique dans vos messages ...

le mode "redondant" envoi les mêmes paquets, et, peu importe le nombre de connexion ... donc la 4G consomme ... la latence diminue parce que le débit de votre 4G est supérieur à votre ADSL.

pour le ping haut de l'ADSL, c'est que la vitesse de votre liaison doit être vraiment très faible, dans ce cas il vaut mieux mettre en place SQM sur la ligne ADSL...

le "test pour la passerelle en dns" qui "explique le fait de trop consommer" ... j'en doute très fortement.

Bien à vous,

Bug-Storm, 2021-02-22:
> le mode "redondant" envoie les mêmes paquets, et, peu importe le nombre de connexions ... donc la 4G consomme ... la latence diminue parce que le débit de votre 4G est supérieur à votre ADSL.

Ok merci

`La latence diminue parce que le débit de votre 4G est supérieur à votre ADSL.`
Comment peut on avoir une baisse de latence si la 4G est en backup?

`pour le ping haut de l'ADSL, c'est que la vitesse de votre liaison doit être vraiment très faible, dans ce cas il vaut mieux mettre en place SQM sur la ligne ADSL...`

Faut il donc définir une vitesse de Down/UP ?

`le "test pour la passerelle en dns" qui "explique le fait de trop consommer" ... j'en doute très fortement.`

En laissant en ping j'ai largement une baisse de consommation

xillibit, 2021-02-26:
Vous ne pourrez pas réduire vous même la latence de votre ligne ADSL, c'est inhérent par rapport aux caractéristiques de votre ligne et de votre FAI.

maxou56, 2021-03-25:
> le mode "redondant" envoie les mêmes paquets, et, peu importe le nombre de connexions ... donc la 4G consomme ... la latence diminue parce que le débit de votre 4G est supérieur à votre ADSL.

Bonjour,
Non le mode "backup" ou "remplacent", sert uniquement si les autres "wan" ne fonctionnent plus. Les paquets ne sont pas envoyer en permanence dessus.

maxou56, 2021-03-25:
> Faut il donc définir une vitesse de Down/UP ?

Oui tu peux, surtout en émission. Pour l'Adsl si tu utilises tous le UP, le Down s'écroule par exemple. 
Et la 4G est beaucoup plus rapide en UP

closed 2021-06-28 by github-actions[bot]