Ysurac/openmptcprouter #1616: Is there a Python documentation for that？

Deep-sea-boy, 2021-02-22:
I did not find any Python files, how to operate? Python is not supported？

Deep-sea-boy, 2021-02-22:
This is download image :
[image]

Front-end developer 's files path is www/ ? 
Where is back-end developer ’s files path ?
What back-end language was the project developed in ？ Roff？R？Python？Java？PHP？...

Ysurac, 2021-02-22:
Python is not used on router side because this take some space and OpenMPTCProuter can be installed on hardware with small flash. Python is used on server side.
lua is used as back-end language for the interface, and shell script for others parts.
in /www there is static files and javascript. Others interfaces files are ine /usr/lib/lua/luci.

hesenkang, 2021-02-23:
Hello, how can the front end modify the management page? Or do some personalized customization?
It looks like automatically generated code.
Please help me. @Ysurac

Deep-sea-boy, 2021-02-23:
where is the luci log file?
how can i print the log?

Deep-sea-boy, 2021-02-23:
CRUD data's path? In Mysql or other config file?

Deep-sea-boy, 2021-02-23:
[image]
Both of this are modify  /etc/config files？

Ysurac, 2021-02-23:
All log is available via ```logread```.
Yes these commands modify /etc/config files.

hesenkang, 2021-02-24:
Hello, how can the front end modify the management page? Or do some personalized customization?
/www or  /usr/lib/lua/luci ?
It looks like automatically generated code.
Please help me. @Ysurac

closed 2021-05-30 by github-actions[bot]