Ysurac/openmptcprouter #1618: Slow bandwidth

renaudbrion, 2021-02-22:
Hello, 

I have a weird comportment with my config. The bandwidth is more or less the same bandwidth with 8 providers linked that 3 providers.

- 3 providers linked : +/- 36,55 Mb/s (ovh.net) 
- 8 providers lined : +/- 36,32 Mb/s (ovh.net)

Does someone have an idea of what could happen? :-) 

Thank you!
Kr, 
Renaud

Ysurac, 2021-02-22:
Please use the template.
Check aggregated speed and speed of each link with ```omr-test-speed```,```omr-test-speed wan1```... (ctrl+c to stop after at least 2 minutes).
Check that your 8 providers doesn't share same antenna if it's LTE.

renaudbrion, 2021-02-22:
Apologies for the template. Thank you for your answer. 
I've already tested this, all seem right.

Kalimeiro, 2021-02-22:
Can you please send screen capture of System -> OpenMPTCProuter -> Status ? (check Anonymize Public IPs)

renaudbrion, 2021-02-22:
Ty for the answer

[image]

hairydogltd, 2021-02-23:
Is the OpenMPTCProuter running on hardware that is struggling to carry the traffic?

renaudbrion, 2021-02-23:
It should be OK ? This is running on SSD. 
[image]

hairydogltd, 2021-02-23:
I was mostly thinking about the speed of the network port(s)

Bug-Storm, 2021-02-24:
Je imagine que vous avez un switch, si oui peut être le souci vient de là 😉...

Et si oui, est il aménageable ?

hairydogltd, 2021-02-24:
The network port(s) on the OMR router have to handle all the internet traffic. If that port gets overloaded, that will limit the speed available.

redkite127, 2021-02-24:
Hello, and thanks for all your replies so far.
I really doubt the network ports are overloaded. We use:
- One Gig port for reaching all WANs (over a dedicated VLAN, eth1.101, eth1.102, ...)
- One Gig port for the private lan

Here are two screenshot, the first with 5 WANs (4 enabled/one master) and the second with only 2 WANs.
- top left, first big iftop is eth0 (private lan)
- second row, first column is eth1
- then it's eth1.101, eth1.102, ...

[image]
[image]

redkite127, 2021-02-24:
I'm confused :confused: ... We moved to an other VPS (for geolocation issues) last week, and I haven't speedtested it yet assuming all VPS are having a descent speed nowadays... I've just done it at it's limited to... 39Mbps :facepalm: 

I immediately switched back to Google and it improved (still not ideal) here are the same screenshot, first one with 5 WANs, second with 2.
[image]
[image]

As you can see, we do reach 72 Mbps but it's still not that awesome comparing to the theoretical sum of throughputs.
[image]

I'm also suspecting some WANs (102,107,109), which are performing well alone, to be using/sharing the same bottleneck somewhere, when used simultaneously, none of them preform as well as alone... (a little bit like 4G problems)

Are you used to such a difference between theoretical sum of speed, what does it tend to be for you? (like ratio between theor/pratic)

Bug-Storm, 2021-02-24:
Try to speed test with 3 wan, after 4wan etc..... And look up how much wan you don't have a speed limit 😉

redkite127, 2021-02-24:
Yup, doing more speedtests and monitoring VLAN interfaces will helps... and having three connections struggling on the same bottleneck is just a waste of money ><

closed 2021-05-31 by github-actions[bot]