Ysurac/openmptcprouter #1619: OpenMPTCProuter v0.58 alpha

Ysurac, 2021-02-22:
Changes:
* IPv6 WAN and VPS support
* Support for interfaces alias
* Directly upgrade from the web interface
* Option to keep installed packages after upgrade
* Update V2Ray
* Update RPI4 Firmware
* Add some monitoring data in luci-app-mptcp
* Multipath TCP update
* Many fixes

JeffWatsonGH, 2021-03-29:
Is the option "Support 16 WANs on x86_64 instead of 8" already included? With which commit was this added?

Ysurac, 2021-03-29:
I forgot to add it. This will be added in next beta.

JeffWatsonGH, 2021-03-29:
Alright. Thanks for the quick reply! :)

geeuk, 2021-03-30:
Hi, Your Images for RPI4 does not seem to work for some reason? The pi does not boot and loops, I tried both ext4 and squashfs.. I was tempted to try the wireguard method. Thanks

Ysurac, 2021-03-30:
I tested on my RPI4 this image: https://download.openmptcprouter.com/refs/tags/v0.58beta2/5.4/rpi4/targets/bcm27xx/bcm2711/openmptcprouter-v0.58beta2-r0%2B16336-b36068d35d-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz
And it's working.

glutfloc, 2021-04-14:
Hi Ysurac,

I'm currently tying to install the VPS part from script (wget -O - http://www.openmptcprouter.com/server-test/debian10-x86_64.sh | sh) and it seems that the repository is not good : 
" Err:8 https://repo.openmptcprouter.com buster/main amd64 Packages
  File has unexpected size (8500 != 8498). Mirror sync in progress? [IP: 89.44.9.58 443]
  Hashes of expected file:
   - Filesize:8498 [weak]
   - SHA256:d147724cb0489d64d6d400c3bab89cfa721ec0e66d60fcf44385865183c523b5
   - SHA1:0e178dc772267e083e460cda6338116577d42d69 [weak]
   - MD5Sum:6bd48e6464409d7570271b08588f5511 [weak]
  Release file created at: Thu, 08 Apr 2021 19:06:03 +0000
"
So I am blocked...
Can you check it ?

Regards,

Yohann

Ysurac, 2021-04-15:
It's better now ? I purged CDN cache.

glutfloc, 2021-04-15:
Yes ! 
I was able to configure one of my VPS. Thank you.
By the way I noticed that it was also the case for the current stable release of the script (https://www.openmptcprouter.com/server/debian10-x86_64.sh). I haven't checked since this morning but I guess this is also fixed for this one.
Thank you.
Yohann

Ysurac, 2021-04-15:
Yes, it was related to the update of the Debian repo, but I forgot to purge CDN cache so this give some cache inconsistency. It's now fixed for stable and snapshot.

wicadmin, 2021-04-23:
Is ModemManager still broken on the RC? What is broken about it?

Kalimeiro, 2021-04-24:
> 
> 
> Is ModemManager still broken on the RC? What is broken about it?

Wait RC2

Speedy37, 2021-04-26:
Great work! I'm on RC1 (RPi3 B+). So far so good :)

Is it me, or bbr2 congestion algorithm is very good? At least that's the feeling after 2 days.
I've 1 slow adsl link (5mbits) and 2 "good" 4G (10-80Mbits) and the latency is much more stable while the bandwidth is higher.

UDP is still not quite there (Glorytun TCP), haven't tried anything else on this release and I disabled HTTP/3 for now due to that.
Any advice, thing you want me to try.

Is it normal that I have to restart a TCP connection to get the benefits of adding a 3rd temporary 4G for a temporary boost, I had the feeling that MLTCP allowed for dynamic add/remove of ressources? The TCP connection stays up, but gets no speed boost.

Ysurac, 2021-04-26:
Yes BBR2 seems quite good even if still in alpha/
For UDP, you can try v2ray proxy that support TCP and UDP but use more CPU/memory.
It's not normal to have to restart a TCP connection, you can try another Multipath TCP scheduler in Network->MPTCP.

Ysurac, 2021-04-27:
@Speedy37 I'm not able to reproduce the TCP connection problem at least without using MacVLAN, I will test with MacVLAN used for WANs. How do you make your test ?

Speedy37, 2021-04-27:
Default settings, ipv4 only
wan1,wan2,wan3 are MacVLAN on eth0.
usb0, Normal (the one I add/remove).

I can see it on the MPTCP fullmesh almost instantly.

Bumpalert, 2021-04-27:
Hey @Ysurac  .. The beta version is almost perfect. I got only one issue, when I enabled wireguard on both master and usb interface no gateway found, first interface becomes no gateway. As I expect all 4G networks to block MPTCP, can I keep all with wireguard or master need to be left without wireguard for making the connection to the VPS? The speed addition tested successfully now with 2 wired internet gateways without wireguard. One wired and 1 4g network wrapped over wireguard successfully.

[image]

Bumpalert, 2021-04-27:
both some issue with wireguard also on usb2_tether(2nd 4g). but i am still happy that aggregation works without any special settings.

[image]

techworldthink, 2021-04-28:
is v0.58 Release source from master branch?

Ysurac, 2021-04-28:
@Bumpalert I try to fix that
@techworldthink for Release Candidate yes, it's from master branch. There is a v0.58rc1 tag.

techworldthink, 2021-04-29:
I tried to build v0.58 for my rpi3 .but got an error!

make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/ubuntu/openmptcprouter/rpi3/source/include/toplevel.mk:230: world] Error 1

How to solve this ?

Bumpalert, 2021-04-30:
I got another look and feel bug. When you have 2 vps and disable the one with IP address and enable one with domain name, the omr-iperf shows that it connected to the disabled IP address one and status shows direct output. But public ip of connected devices is actually the enabled vps which has domain name. 

[image]
[image]

I was planning to add noip sub domain names to the dynamic ip on the vps(with automatic address update from vps). Let me know whether it has any performance concerns.

Ysurac, 2021-04-30:
@Bumpalert 
* In fact without parameters omr-iperf use the VPS with the name "vps". You can check another VPS by using his name as argument
* I will fix the issue with Direct output. No problem with noip, the only problem can be dynamic IP when IP change: this will disconnect all open session.

Bumpalert, 2021-04-30:
factory reset is a problem. It says yes, but settings are not changed on web interface. Is there any thing I missed?

root@OpenMPTCProuter:~# 
root@OpenMPTCProuter:~# 
root@OpenMPTCProuter:~# 
root@OpenMPTCProuter:~# firstboot y
This will erase all settings and remove any installed packages. Are you sure? [N/y]

root@OpenMPTCProuter:~# 
root@OpenMPTCProuter:~# firstboot yes
This will erase all settings and remove any installed packages. Are you sure? [N/y]

root@OpenMPTCProuter:~# 
root@OpenMPTCProuter:~# reboot

Ysurac, 2021-04-30:
@Bumpalert this can only work on squash FS. You can also update with same image and not keep settings.

Bumpalert, 2021-05-01:
confirmed domain name in the vps server address has some issue. Same connection when ip address used working fine will fail if dynamicdns is used eventhough public ip is resolved.

[image]
[image]

any working way for using pi 4b wifi as ap? last time I had to reflash only because I tried wifi on pi.
https://github.com/Ysurac/openmptcprouter/issues/1232
https://github.com/Ysurac/openmptcprouter/wiki/Configure-Wifi

Ysurac, 2021-05-10:
Do you have errors in System logs about omr-tracker ?

Speedy37, 2021-05-10:
Nothing special.

Ysurac, 2021-05-11:
@Speedy37 I fixed sysupgrade (it was an issue on my server). For uci cpu load I have no idea, there is a delay between each run of omr-tracker and it's not the problem here as it's really UCI. What is the RPI model ?

@dooley1981 Server part is updated and rebooted ? Since 0.58 beta 7 there is a new way to detect if multipath is enabled or not and this use the server API.

Speedy37, 2021-05-11:
RPI 3B+, 64bits image.
The new SD card or reinstall seems to have fixed that uci issue.
Still, load is quite high for simple ping/http requests. Would you probably accept a PR with a rewrite of omr-tracker in python to lighten the load?

Speedy37, 2021-05-11:
A quick look at the current code tells me there isn't much to improve. I should probably track the root cause of that load first, it might simply be IO stalls.

flimo44, 2021-05-12:
Hello @Ysurac ,
i update RC2 but https://packages.openmptcprouter.com/v0.58rc2/.... not found. Impossible to install packages : http://packages.openmptcprouter.com/v0.58rc2/x86_64/packages/.

Else after each update need to upload my certificat. It would not be possible as for the parameters to restore them automatically after each update and the same for software i need to add ( agent pour VM syno... ) ?
Last point, following the change from RC1 to RC2 problem with IPV6...

I use this version : https://download.openmptcprouter.com/develop/5.4/x86_64/targets/x86/64/openmptcprouter-v0.58rc2-r0+16336-b36068d35d-x86-64-generic-ext4-combined.img.gz

Merci.

Ysurac, 2021-05-12:
@flimo44 I fixed the install package problem.
There is no change about IPv6 between RC1 and RC2
For software I made some changes and it should resintall packages, but I never tested (and while package url was not working it's normal).
For certificate, you need to add the file in System->Flash firmware, tab configuration. But default file (/etc/uhttpd.key and .crt) should be saved by default.

flimo44, 2021-05-12:
@Ysurac ,
Utl acess https it's OK in browser but in openMTCTPRouteur not OK : 
Downloading http://packages.openmptcprouter.com/v0.58rc2/x86_64/luci/Packages.gz
*** Failed to download the package list from http://packages.openmptcprouter.com/v0.58rc2/x86_64/luci/Packages.gz

Problem link htpp  and not https ?

For certificate thank you for  backup solution.

Bumpalert, 2021-05-12:
> @Speedy37 I fixed sysupgrade (it was an issue on my server). For uci cpu load I have no idea, there is a delay between each run of omr-tracker and it's not the problem here as it's really UCI. What [...]

@Ysurac How to update server part? Just run vps script again? or vps script path is changed?

Malaga82, 2021-05-14:
Great to see wireguard as MPTCO over VPN, but for it's not bypassing ISP block, only OpenVPN does: am i the only one?
Other than that, in wreguard interface i need to manually paste server PK.
For OpenVPN, when enabled from Setting Wizard, it must manually started, it stays stopped otherwise.
Will try it more im my spare time.
Thanks

flimo44, 2021-05-14:
Hello @Ysurac,

I went back to the beta6 version which is very stable at home (the beta7 also remembered). Since RC1 version too many problems. ( I'm on VM on my Nas Synology for info )

From RC1:
-IPV6 no longer works.
No ip V6 or when it seems OK, impossible to go out in IPV6. (ping KO)

-This error  in continuous system log: 
Sat May 15 00:05:20 2021 daemon.warn odhcpd [5968]: A default route is present but there is no public prefix on lan thus we don't announce a default route !
Sat May 15 00:05:20 2021 daemon.err odhcpd [5968]: Failed to send to ff02 :: 1% lan @ eth0 (Permission denied)

- Unstable connection and loss ping.

- My public IP see is my Freebox. A tracert from my PC goes through my OVH VPS (Don't understand anything).
- 
Here is my feedback from the RC version. For me beta6 and beta7 are very stable. For the moment I remain in beta6. 

(Merci et vraiment bravo pour le super travail que tu fais sur OpenMPTCProuter.)

Ysurac, 2021-05-15:
@Malaga82 I will check MPTCP over VPN again
@flimo44 I don't have any IPv6 problems, I think you have a problem in your configuration. Check that "Always announce default router" is checked in Network->Interfaces->Lan->DHCP server->IPv6 Settings. When you say public IP is your freebox, you mean from OpenMPTCProuter or from internet ? It's an IPv4 or IPv6 ?

flimo44, 2021-05-15:
Hello,
@Ysurac 
 I don't have any IPv6 problems, I think you have a problem in your configuration. ->Possible but same configuration is OK in beta6 

Check that "Always announce default router" is checked in Network->Interfaces->Lan->DHCP server->IPv6 Settings. ->  Yes it's checked

When you say public IP is your freebox, you mean from OpenMPTCProuter or from internet ? It's an IPv4 or IPv6 ? when I test my public Ip on sites like http://ip.lafibre.info/ or http://www.mon-ip.com/ it is my IP v4 freebox wan and not that of my VPS that is seen .
In beta6 (ipv6 OK) it is indeed that of the VPS.

Ysurac, 2021-05-15:
@flimo44 the only change about IPv6 after beta6 is the new "DNS64" checkbox, but this doesn't change anything when not used.
How is the OpenMPTCProuter status page ? All is green ? If it's ok, check gateway you use on PC or mobile.

flimo44, 2021-05-15:
> @flimo44 the only change about IPv6 after beta6 is the new "DNS64" checkbox, but this doesn't change anything when not used.
> How is the OpenMPTCProuter status page ? All is green ? If it's ok, check gateway you use on PC or mobile.

Yes for beta6
Yes for RC1 or RC2 : just no ipV6  and this error in continuous system log:
Sat May 15 00:05:20 2021 daemon.warn odhcpd [5968]: A default route is present but there is no public prefix on lan thus we don't announce a default route !
Sat May 15 00:05:20 2021 daemon.err odhcpd [5968]: Failed to send to ff02 :: 1% lan @ eth0 (Permission denied)

Ysurac, 2021-05-15:
@Malaga82 I found the issue: Now you can use MPTCP over VPN on all interfaces and it's aggregated, but if only one interface use it, it's not aggregated. I will fix that.
@flimo44 For TCP traffic the proxy is used (by default shadowsocks), the only way to not use it (and not use the VPN when proxy is down) is to bypass with omr-bypass.

Malaga82, 2021-05-15:
@Ysurac sorry if i was not detailed enough.
I have 3 WAN, only 1 sometimes get MPTCP blocked ( it's a virtual ISP, it can get 2 different ASN, one of two block MPCTP ), and with OpenVPN it bypasses the block, not with WireGuard.
I'm testing with 1 WAN over VPN, if you want i can test with 2 WAN over VPN.
When WireGuard is enabled for WAN, it will say "Gateway DOWN", because PK is not setted up:
[image]
and WireGuard status shows only PK, Listen Port and FW Mark, not the connection status.
Once edited WGWAN* with PK from VPS, with "wg show", it wll start "works":
[image]

flimo44, 2021-05-18:
Hello @Ysurac 

I just installed the latest upgraded RC2 on beta6 and it looks not too bad.
No more error in the system log, IPV6 OK.
Regarding my packages they are not reinstalled automatically after the update.
Only problem encountered, I can no longer reinstall the qemu-ga package because dependencies are missing.

[image]

Best regards

Ysurac, 2021-05-20:
@flimo44 these packages should be installed by default. I will check if it's an error on package manager or if there are really not available.

flimo44, 2021-05-20:
Hello @Ysurac 
Can you please  for RC3 correct the package link :

Downloading http://packages.openmptcprouter.com/v0.58rc3/x86_64/luci/Packages.gz
*** Failed to download the package list from http://packages.openmptcprouter.com/v0.58rc3/x86_64/luci/Packages.gz

Merci

Ysurac, 2021-05-21:
@flimo44 fixed

Malaga82, 2021-05-21:
@Ysurac when upgrading RC3 i have this error on VPS side:

> Reading state information... Done
> python3-jwt is already the newest version (1.7.0-2).
> python3-netaddr is already the newest version (0.7.19-1).
> python3-passlib is already the newest version (1.7.1-1).
> libuv1 is already the newest version (1.34.2-1).
> python3-uvloop is already the newest version (0.14.0+ds1-1).
> 0 upgraded, 0 newly installed, 0 to remove and 61 not upgraded.
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> python3-aiofiles is already the newest version (0.4.0-2).
> ipcalc is already the newest version (0.41-5).
> jq is already the newest version (1.5+dfsg-2+b1).
> python3-netifaces is already the newest version (0.10.4-1+b1).
> python3-psutil is already the newest version (5.5.1-1).
> python3-uvicorn is already the newest version (0.11.5-1).
> 0 upgraded, 0 newly installed, 0 to remove and 61 not upgraded.
> -- pip3 install needed python modules
> netjsonconfig 0.9.1 has requirement jinja2<3.0,>=2.9, but you'll have jinja2 3.0.1 which is incompatible.
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done

Anyway it was installing other things anyway, and seems to finish fine.

Malaga82, 2021-05-21:
I did a quick check, despite for the PK key bug on WGWAN interface that is still there, it seems that on upload side the aggregation is not working well.

With WG enabled on the WAN:
[image]

With WG disabled on the WAN:
[image]

Currently i'm outside and can't test more, but i did 5 speedtests for each WG on and off.
Will check more later.
Thanks.

flimo44, 2021-05-21:
Hello @Ysurac 

I have this error every hour in the system log in RC3 : ? 

Fri May 21 09:02:16 2021 daemon.err omr-tracker[19536]: /bin/omr-tracker: eval: line 1132: openmptcprouter.wan1.multipath: not found
Fri May 21 09:02:16 2021 daemon.err omr-tracker[19536]: /bin/omr-tracker: eval: line 1132: openmptcprouter.wan2.multipath: not found
Fri May 21 09:02:16 2021 daemon.err omr-tracker[19536]: /bin/omr-tracker: eval: line 1132: openmptcprouter.omrvpn.multipath: not found
Fri May 21 09:02:16 2021 daemon.err omr-tracker[19536]: /bin/omr-tracker: eval: line 1132: openmptcprouter.loopback.multipath: not found
Fri May 21 09:02:16 2021 daemon.err omr-tracker[19536]: /bin/omr-tracker: eval: line 1132: openmptcprouter.lan.multipath: not found
Fri May 21 09:02:16 2021 daemon.err omr-tracker[19536]: /bin/omr-tracker: eval: line 1132: openmptcprouter.omr6in4.multipath: not found
Fri May 21 09:02:16 2021 daemon.err omr-tracker[19536]: /bin/omr-tracker: eval: line 1132: openmptcprouter.LAN_FB.multipath: not found
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: Usage: uci [<options>] <command> [<arguments>]
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]:
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: Commands:
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	batch
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	export     [<config>]
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	import     [<config>]
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	changes    [<config>]
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	commit     [<config>]
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	add        <config> <section-type>
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	add_list   <config>.<section>.<option>=<string>
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	del_list   <config>.<section>.<option>=<string>
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	show       [<config>[.<section>[.<option>]]]
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	get        <config>.<section>[.<option>]
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	set        <config>.<section>[.<option>]=<value>
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	delete     <config>[.<section>[[.<option>][=<id>]]]
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	rename     <config>.<section>[.<option>]=<name>
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	revert     <config>[.<section>[.<option>]]
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	reorder    <config>.<section>=<position>
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]:
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: Options:
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	-c <path>  set the search path for config files (default: /etc/config)
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	-d <str>   set the delimiter for list values in uci show
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	-f <file>  use <file> as input instead of stdin
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	-m         when importing, merge data into an existing package
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	-n         name unnamed sections on export (default)
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	-N         don't name unnamed sections
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	-p <path>  add a search path for config change files
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	-P <path>  add a search path for config change files and use as default
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	-q         quiet mode (don't print error messages)
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	-s         force strict mode (stop on parser errors, default)
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	-S         disable strict mode
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]: 	-X         do not use extended syntax on 'show'
Fri May 21 09:03:02 2021 daemon.err omr-tracker[19537]:

Ysurac, 2021-05-21:
@flimo44 I compiled RC4 that fix this issue (a big typo...)

flimo44, 2021-05-21:
hello
RC4 not OK for me.  Many error in log , very slow.  ping OpenMptcpRouter very bad
( openmptcprouter-v0.58rc4-r0+16336-b36068d35d-x86-64-generic-ext4-combined.img ).

Malaga82, 2021-05-22:
i tested a bit, here is what i sorted:

1. WG vpn has issue regarding PK on interface side.
2. With WG i have upload issue, it seems to aggregate but upload is really slow. With WG off upload bandwidth is like 5 times more.
3. When switching from WG to OpenVPN, resultant WAN had gateway down. Had to first disable MPTCP over VPN flag for WAN, set MPTCP over VPN settings to OpenVPN, and then enable the flag.
4. Sometimes i have "No server http response after 1 sec" warning out of nowhere, and doing nothing: strange because with RC2 and  surely with 0.57 i didn't have this warning.
5. When doing speedtest i have disconnections on some services, like twitch. It cloud be ok, but i didn't noticed this on 0.57.
6. Disable server http test, in Other settings it seems not to work, you can flag it, but after saving it will be unflagged.
7. Master interface selection in Other settings sometimes it resets to "On wizard change", haven't sorted out the behaviour on that.
8. On port forwarding, i expose a wireguard VPN server for accessing home from outside, and i have to connect twice, the first try is always a failure.

Will update if will find more.
Thanks

Ysurac, 2021-05-22:
@Malaga82 VPS script is updated to latest release ? I have no issues with WG PK. I don't understand the 8, you have another wireguard VPN ?

Malaga82, 2021-05-22:
@Ysurac  i re-launched the script yesterday with RC4, and before with RC2. 
i'm on Version 0.1025-test 5.4.100-mptcp.
Well, i don't know, but everytime i use WG and it creates a WGWAN interface, it has no PK filled.
Maybe something remains in config even if VPN WAN interface is removed?
for 8. i have a wireguard server and forwarding the port for accessing lan home.
Thanks

Ysurac, 2021-05-22:
@Malaga82 can you give me the result of ```uci show network``` and ```uci show openmptcprouter``` (without VPS ip) after you set MPTCP over VPN and when PK are not filled ? And how you set it ?

Malaga82, 2021-05-28:
@Ysurac here it is:
`root@OpenMPTCProuter:~# uci show network
`
> network.loopback=interface
> network.loopback.ifname='lo'
> network.loopback.proto='static'
> network.loopback.ipaddr='127.0.0.1'
> network.loopback.netmask='255.0.0.0'
> network.loopback.multipath='off'
> network.loopback.macaddr='00:00:00:00:00:00'
> network.loopback.metric='1'
> network.globals=globals
> network.globals.multipath='enable'
> network.globals.mptcp_path_manager='fullmesh'
> network.globals.mptcp_checksum='0'
> network.globals.mptcp_debug='0'
> network.globals.mptcp_fullmesh_create_on_err='1'
> network.globals.mptcp_ndiffports_num_subflows='1'
> network.globals.mptcp_scheduler='blest'
> network.globals.mptcp_fullmesh_num_subflows='2'
> network.globals.congestion='cubic'
> network.globals.ula_prefix='fd46:da54:b5de::/48'
> network.globals.mptcp_syn_retries='3'
> network.globals.mptcp_rr_cwnd_limited='Y'
> network.globals.mptcp_rr_num_segments='1'
> network.lan=interface
> network.lan.ifname='eth0'
> network.lan.proto='static'
> network.lan.netmask='255.255.255.0'
> network.lan.ip6assign='60'
> network.lan.delegate='0'
> network.lan.ip4table='lan'
> network.lan.macaddr='d2:76:e9:19:f8:13'
> network.lan.metric='2'
> network.lan.ipaddr='192.168.1.254'
> network.lan.defaultroute='0'
> network.lan.peerdns='0'
> network.lan.multipath='off'
> network.lan.ipv6='0'
> network.lan_rule=rule
> network.lan_rule.lookup='lan'
> network.lan_rule.priority='100'
> network.wan1=interface
> network.wan1.ifname='eth1'
> network.wan1.proto='static'
> network.wan1.ip4table='wan'
> network.wan1.defaultroute='0'
> network.wan1.macaddr='1e:c0:b0:92:87:4a'
> network.wan1.ipv6='0'
> network.wan1.metric='3'
> network.wan1.peerdns='0'
> network.wan1.label='FTTC'
> network.wan1.ipaddr='192.168.2.200'
> network.wan1.netmask='255.255.255.0'
> network.wan1.gateway='192.168.2.254'
> network.wan1.dns='192.168.1.20' '94.140.15.15'
> network.wan1.addlatency='0'
> network.wan1.multipath='master'
> network.wan2=interface
> network.wan2.proto='static'
> network.wan2.ip4table='wan'
> network.wan2.defaultroute='0'
> network.wan2.macaddr='b6:ed:6f:31:be:ed'
> network.wan2.ipv6='0'
> network.wan2.peerdns='0'
> network.wan2.label='4G'
> network.wan2.netmask='255.255.255.0'
> network.wan2.ifname='eth2'
> network.wan2.dns='94.140.14.14' '94.140.15.15'
> network.wan2.ipaddr='192.168.8.200'
[... 178 lines cut ...]
> openmptcprouter.vps.password='XX'
> openmptcprouter.vps.port='65500'
> openmptcprouter.vps.ipv6='XXX'
> openmptcprouter.vps.pihole='0'
> openmptcprouter.vps.machine='x86_64'
> openmptcprouter.vps.nofwredirect='0'
> openmptcprouter.vps.available_vpn='glorytun_tcp' 'glorytun_udp' 'dsvpn' 'openvpn' 'openvpn_bonding' 'mlvpn'
> openmptcprouter.vps.redirect_ports='0'
> openmptcprouter.vps.get_config='0'
> openmptcprouter.vps.master='1'
> openmptcprouter.vps.backup='0'
> openmptcprouter.vps.lastbackup='1613207073.184747'
> openmptcprouter.vps.kernel='5.4.100-mptcp'
> openmptcprouter.vps.omr_version='0.1025-test'
> openmptcprouter.vps.admin_error='0'
> openmptcprouter.vps.ip='X'
> openmptcprouter.vps.token='XXX'
> openmptcprouter.vps.lastchange='1622222008'
> openmptcprouter.vps.set_firewall='1'
> openmptcprouter.wgwan2=interface
> openmptcprouter.wgwan2.multipath='on'
> openmptcprouter.wgwan2.vpn='1'
> openmptcprouter.wgwan2.baseintf='wan2'
> openmptcprouter.wgwan2.metric='7'
> openmptcprouter.wgwan2.state='down'

Malaga82, 2021-05-28:
[image]

[image]

Ysurac, 2021-05-28:
@Malaga82 Did you import an old config ? Did you try to check in the wizard, "Advanced settings" checkbox, "Force retrieve settings" and validate ?

Malaga82, 2021-05-28:
@Ysurac yes, i can't really not import it, as i have many things setted up and can't check one by one.
I'll try with "force retrieve settings".
BTW with WG on i have poor upload performance on that WAN: is something you are aware of?
Thanks

bobafetthotmail, 2021-06-01:
did the procedure for port forwarding change? I do not see a "Use V2Ray" box in the port forwarding https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding-using-v2ray

[36 more replies]