Ysurac/openmptcprouter #1620: Question  how about using APU2D4 and so on for OMR

vinceducat, 2021-02-22:
Is there user of this type off hardware wath is your experience ... is it ok for OMR

@Ysurac should be ok to aggregate ADSL and 4G

Thanks

Ysurac, 2021-02-22:
It support AES-NI so should give a quite good result even if CPU is quite slow.
I would expect it to be able to aggregate at least 200Mb/s.

vinceducat, 2021-02-23:
I have beetween 200Mbits/s and 300Mbits/s on the 4G way... perhaps APU 4D2 or 4D4   should be a better choice ...

Ysurac, 2021-02-23:
They are all based on same CPU. So they all should give the same speed. I never tested, I can be wrong on max speed.

vinceducat, 2021-02-23:
OK thanks your right same CPU...

smallstech, 2021-02-25:
All APU2/3/4 have the same CPU tho some have 2GB of ram and others 4. I recently got a APU2E4 for OMR and it works like a charm tho I would highly recommend building a custom firmware ([info here](https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform)) and applying apu specific optimizations see [https://advanxer.com/blog/2019/02/optimising-lede-openwrt-for-the-pc-engines-apu2/](url) and [here](https://openwrt.org/toh/pcengines/apu2).

vinceducat, 2021-02-26:
Thanks il will read this carefully ;-)

luisglass, 2021-03-05:
New to the creating image part.. In reading [this](https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform) to understand more the "building a custom firmware" for the APU. I'm not clear where the menu is to select this arch? And am missing where to find a makeconfig file in the git clone.  

I'm currently using the x86_64 precompiled version my APU3.  It's working, except having troubled getting to work with a Quectel EC25 modem card.

smallstech, 2021-03-05:
The APU is x86_64 based. The EC25 might need a kernel module, the openwrt forum might be a better place for information about getting that to work.

closed 2021-06-09 by github-actions[bot]