Ysurac/openmptcprouter #1621: Errors on first connection

desmando, 2021-02-23:
## Expected Behavior
Browse internet over tunnel

## Current Behavior
On visit visit to a new website there is a delay in the connection and chrome will respond with err_empty_response or err_connection_closed. After a refresh that site will work.

## Possible Solution
Destroying the VPC and recreating on new VM resolves problem for a while

## Steps to Reproduce the Problem

  1. Use webbrowser to visit site that hasn't been visited before
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: 0.1024 5.4.86-mptcp
  - OpenMPTCProuter platform: RPI4

Ysurac, 2021-02-23:
Shouldn't be any delay.
Do you use OpenMPTCProuter IP as DNS ?
What is your VPS provider ?

desmando, 2021-02-23:
I just checked and machine I'm on was not using router as DNS. I've set it to do so now.
Provider is AWS Lightsail.

desmando, 2021-02-23:
Problem is still happening after configuring client to use router as dns.

Ysurac, 2021-02-23:
Everything is green on status page ?

desmando, 2021-02-23:
Yes

Ysurac, 2021-02-23:
It's possible that AWS filter DNS request (many result on google search about that even if I don't know if this apply in our case).
You can try to set a custom DNS in Network->DHCP&DNS in place of "127.0.0.1#5353" and add it in omr-bypass.

desmando, 2021-02-23:
Change made. I'll monitor and report back if needed. Thanks for the help.

closed 2021-02-23 by desmando