Ysurac/openmptcprouter #1624: Traffic going only over master interface

luka-diplomski, 2021-02-24:
## Expected Behavior

Traffic going over both interfaces simulatiously, mptcp

## Current Behavior
Traffic going out only over interface that is selected as master

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version:  sorry dont know how to check, pulled latest debian9-x86_64.sh on approx 02/22/2021 20:00CET
  - OpenMPTCProuter platform: RPI4

First of all thank you for this awesome miracle! Few hours ago I didnt even know magic like this is posible, especially for dummies version gui+wizard like this :)  Been using mikrotik loadbalncing for years and finally found a Holy Grail!

But... it's not working for me. Please help!
I have 2x 4G routers connected, but only the interface that I select as master is sending/receiving traffic. Enabled interface is not receiving traffic. When I switch roles (master/enabled), then only the other one is "working". Almost like a master/backup solution. I tried changing Multipath TCP scheduler, Congestion Control, multiple restarting of RPI and VPC but nothing helps :(  Fullmesh enabled.

Only thing I can think of is that one of the interfaces (4g routers) is behind CG-NAT, could that be the issue of not-multipathing?
Its 2AM im going to sleep depresed, please help!

Thank you!

Ysurac, 2021-02-24:
You can check that MPTCP is not blocked on an interface using Network->MPTCP, "MPTCP Support Check" tab.

luka-diplomski, 2021-02-24:
Thank you for the quick reply! I already tried that but didnt get any errors, or so I thought..

my WAN1:

```
1: 192.168.2.254 
2: 192.168.225.1 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
3: 172.31.177.97 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
4: 172.28.24.244  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
5: 172.28.24.78  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
6: 172.28.24.197  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
7: 172.28.24.249  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
8: 195.29.224.145  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
9: 195.29.225.178  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
10: 149.6.30.153  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
11: 154.54.59.193  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
12: 154.54.59.186  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
13: 130.117.51.58  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
14: 154.54.37.42  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
15: 54.36.50.94  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
16: 213.251.128.114  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
17: * 
18: 213.251.128.114  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
19: * 
20: 91.121.215.117  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
21: * 
22: * 
23: * 
24: * 
25: * 
26: * 
27: * 
28: **MyVPSip** TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 0e:d1:b8:16:25:78 , SourceMAC = 18:d6:c7:ac:c1:cd , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 43 , Protocol = 0x6 , CheckSum = 0x6487 , SourceIP = myVPSip , DestinationIP = 192.168.2.1 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 29427 , SeqNumber = 2273069481 , AckNumber = 1615289215 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 43200 , CheckSum = 0xb435 , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1424 , >
< TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 3xxxxxxxxxxx668866 , >
```

wan2 (cg-nat):

```
1: 192.168.3.254 
2: * 
3: 10.84.161.173 IP::TTL IP::CheckSum 
4: * 
5: 212.91.97.220  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
6: 83.139.120.129  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
7: 212.162.29.117  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
8: * 
9: * 
10: * 
11: * 
12: * 
13: 91.121.215.117  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
14: * 
15: * 
16: * 
17: * 
18: * 
19: * 
20: * 
21: **MyVPSip** TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize -TCPOptionMPTCPCapable 

Result:
< Ethernet (14 bytes) :: DestinationMAC = c6:c9:d0:0e:b1:3e , SourceMAC = d4:72:26:3b:5c:8e , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 44 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 45 , Protocol = 0x6 , CheckSum = 0x6193 , SourceIP = MyVPSip , DestinationIP = 192.168.3.1 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 33655 , SeqNumber = 1872892241 , AckNumber = 1884849914 , DataOffset = 6 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 42340 , CheckSum = 0xcbd , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1380 , >
```

Ysurac, 2021-02-24:
On WAN2 I can see "-TCPOptionMPTCPCapable", so I would say MPTCP flag is filtered somewhere on this connection.
You can try to check "MPTCP over VPN" on this connection in the wizard.

luka-diplomski, 2021-02-24:
Thanks for the quick reply! It works!!!

I called my ISP and they removed me from CGNAT and now I get multipath tcp.
Please drink a beer for me, I paypalled it. And I'm your new grateful patreon.

THANK YOU!

closed..

closed 2021-02-24 by luka-diplomski