Ysurac/openmptcprouter #1629: Shadowsocks is not running + Can't get public IP address from shadowsocks

kinkyninja, 2021-02-25:
## Expected Behavior

This error should not occur. 
'Force retrieve keys' helps but only for the current session and after hardware reboot, it's needed again.  

As a fallback solution, I can execute 'Force retrieve settings' in command line after each reboot. 
Can you tell how to execute this from command line?

## Current Behavior

This error happens in ~50% of cases after rebooting the system (Raspberry Pi 4B + 4 modems). 
Internet starts lagging, seems like only 1 modem works when the problem is there. In ~50% cases everything works just fine. 

1 interface is master.

Checked other similar Issues here, but didn't find a solution. 

After pressing "Force retrieve all keys from server" and saving the problem goes away. 
But after the hardware reboot - it's still there, the problem occurs in 30%-50% cases. 

[image]

## Specifications

  - OpenMPTCProuter version:  v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version:  5.4.86-mptcp 0.1024
  - OpenMPTCProuter platform: Raspberry Pi 4

hairydogltd, 2021-02-25:
Why is the Raspberry Pi being rebooted? 
How are you rebooting it?

Ysurac, 2021-02-25:
Check in State->Process what is using so much CPU.

kinkyninja, 2021-02-26:
Btw, how do I "Force retrieve settings" via the command line?

kams19, 2021-02-26:
i had similar issue, had to rebuild the openmptcp router - then it works fine.

closed 2021-06-01 by github-actions[bot]