Ysurac/openmptcprouter #1630: OMR-BYPASS not working for domains

kams19, 2021-02-26:
## Expected Behavior
 

domains defined in OMR-BYPASS should follow the local WAN interface

## Current Behavior

domains defined under OMR-BYPASS still follow the route to VPS 

## Possible Solution

Not known

## Steps to Reproduce the Problem

  1. Define a new domain under domain section in OMR-BYPASS
  2. Save and Apply
  3. traceroute and see the route followed

## Context (Environment)
 Watch BBC iPlayer via local WAN
 BBC iPlayer actively blocks connection from VPS IP addresses. Hence I am trying to route that traffic locally via one of my WAN connection.

## Specifications

[image]

  - OpenMPTCProuter version:  0.57.3
  - OpenMPTCProuter VPS version:  0.1024.5.4.86-mptcp
  - OpenMPTCProuter platform:  x86 (VM on ESXi)

[image]

BBC traceroute -

C:\WINDOWS\system32>tracert -d bbc.co.uk

Tracing route to bbc.co.uk [151.101.0.81]
over a maximum of 30 hops:

  1    <1 ms    <1 ms    <1 ms  192.168.1.1
  2     2 ms     1 ms     2 ms  192.168.10.1
  3    26 ms    40 ms    32 ms  10.255.255.1
  4     *

However, teams is defined under Protocols and Services in OMR-BYPASS, that follows local WAN route :
[image]

C:\WINDOWS\system32>tracert -d teams.microsoft.com

Tracing route to s-0005.s-msedge.net [52.113.194.132]
over a maximum of 30 hops:

  1    <1 ms    <1 ms    <1 ms  192.168.1.1
  2     *        *        *     Request timed out.
  3     1 ms     1 ms     1 ms  192.168.9.1
  4     *        *        *     Request timed out.

Ysurac, 2021-02-26:
Make sure you use OpenMPTCProuter IP as DNS and reset your local DNS cache if needed.

Kalimeiro, 2021-02-26:
I approved this bug, only ASN worked.

Ysurac, 2021-02-26:
Domain bypass is working correctly. Problems are often due to some cache.
Wildcards domain bypass is supported via protocol, I can add it to domain bypass.

kams19, 2021-02-26:
i have changed DNS for this PC to that of OMR IP but the traceroute still takes VPS route:

Ethernet adapter Ethernet0:

   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : vmxnet3 Ethernet Adapter
   Physical Address. . . . . . . . . : 00-0C-29-xxxxxxx
   DHCP Enabled. . . . . . . . . . . : No
   Autoconfiguration Enabled . . . . : Yes
   IPv4 Address. . . . . . . . . . . : 192.168.1.200(Preferred)
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Default Gateway . . . . . . . . . : 192.168.1.1
   DNS Servers . . . . . . . . . . . : 192.168.10.1
   NetBIOS over Tcpip. . . . . . . . : Enabled

traceroute output after dnsflush:

C:\WINDOWS\system32>tracert -d bbc.co.uk

Tracing route to bbc.co.uk [151.101.192.81]
over a maximum of 30 hops:

  1    <1 ms    <1 ms    <1 ms  192.168.1.1
  2     1 ms     1 ms     2 ms  192.168.10.1
  3    15 ms    15 ms    15 ms  10.255.255.1
^C

Ysurac, 2021-02-26:
As I said, you may still have some cache.

Ysurac, 2021-02-26:
And what is 192.168.1.1 ?

kams19, 2021-02-26:
192.168.1.1 is the firewall/DHCP for local LAN.
Let me know how can I clear the cache?

Ysurac, 2021-02-26:
You can try to reboot client & router.

Ysurac, 2021-02-26:
As you have many protocols, it's possible that settings IP take a long time (amazon, microsoft,... have many domains/ips), this will take far less time in next release. You can check if a omr-bypass  process is running in Status->Processes, if it's the case then it's the problem.

kams19, 2021-02-26:
cant see omr-bypass under status>processes

and on the cli-

xxxxx@OpenMPTCProuter:~# service omr-bypass status
active with no instances

This is the fresh installation that i have done today. Is this normal?

Ysurac, 2021-02-26:
You can try to disable protocols bypass and check if it's working without them.

Kalimeiro, 2021-02-27:
> Let me know how can I clear the cache?

On Windows, try : ipconfig /flushdns

kams19, 2021-03-01:
> > Let me know how can I clear the cache?
> 
> On Windows, try : ipconfig /flushdns

I had done that before I tried the traceroute, so was wondering if there is anything else that needs to be done.

kams19, 2021-03-01:
> You can try to disable protocols bypass and check if it's working without them.

it is still the same even after disabling protocols. Did you try to replicate the issue at your end, is that just me who is getting this?

Ysurac, 2021-03-01:
Check with ```ipset list``` that you have bbc.co.uk IPs.
If it's the case, check in which list (should be omr_dst_bypass_wan...), and check that this rule is in iptables, via ```iptables-save | grep omr_dst_bypass_...```. You now get the "set-xmark" that you can verify in ```ip rule``` and get the "lookup" number. With this last number you can check the route table via ```ip r show table <number>```.

kams19, 2021-03-01:
> Check with `ipset list` that you have bbc.co.uk IPs.
> If it's the case, check in which list (should be omr_dst_bypass_wan...), and check that this rule is in iptables, via `iptables-save | grep [...]

root@OpenMPTCProuter:~# iptables-save | grep omr_dst_bypass_eth2
-A ssr_def_dst -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_dst -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_dst -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_dst -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_dst -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_dst -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_dst -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_dst -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_local_out -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_local_out -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_local_out -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_local_out -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_local_out -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_local_out -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_local_out -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_local_out -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_pre_src -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_pre_src -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_pre_src -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_pre_src -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_pre_src -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_pre_src -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_pre_src -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A ssr_def_pre_src -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
-A omr-bypass -m set --match-set omr_dst_bypass_eth2 dst -j MARK --set-xmark 0x5394/0xffffffff
root@OpenMPTCProuter:~# ip rule
0:      from all lookup local
0:      from 192.168.8.250 lookup 3
**0:      from 192.168.88.250 lookup 4**
0:      from 192.168.9.250 lookup 8
0:      from 10.255.255.2 lookup 1200
1:      from all fwmark 0x539 lookup 991337
1:      from all fwmark 0x5393 lookup 3
1:      from all fwmark 0x5394 lookup 4
1:      from all fwmark 0x5398 lookup 8
100:    from all lookup lan
10000:  from 192.168.10.1 lookup lan
20000:  from all to 192.168.10.1/24 lookup lan
32766:  from all lookup main
32767:  from all lookup default
90004:  from all iif lo lookup lan

root@OpenMPTCProuter:~# ip r show table 4
default via 192.168.88.1 dev eth2
192.168.88.0/24 dev eth2 scope link

this looks correct isnt it?
but still the traceroute and traffic does not follow this route

Ysurac, 2021-03-01:
This seems to be correct.
You can try to check via ```iptables -t nat -vL``` if a counter to a set-xmark.... is increasing with a RETURN with mark match increasing the same way.

kams19, 2021-03-01:
> This seems to be correct.
> You can try to check via `iptables -t nat -vL` if a counter to a set-xmark.... is increasing with a RETURN with mark match increasing the same way.

i can see the set x mark and RETURN counter increases when I browse the website. 
But it seems only the http/s traffic gets routed by this. The counter doesnt increase for ping and tracert traffic - as that traffic is still using the default route of VPS, bit strange. 
All the traffic for services under Protocol criteria takes the correct route though.

Do you agree this is a bug?

Heugy67, 2021-03-01:
Hello,
I have a similar issue but only with one domain.
The faulty domain ping by default in Ipv6.

The rule omr6_dst_bypass_wan2 is empty.

I'm very interresting to follow this issue to know if my issue is same as "Kam19"

Ysurac, 2021-03-01:
IPv6 is not supported on WAN in current release, this will be supported in next release.

closed 2021-06-04 by github-actions[bot]

elemen3, 2022-04-03:
> Domain bypass is working correctly. Problems are often due to some cache. Wildcards domain bypass is supported via protocol, I can add it to domain bypass.

I just tried `*.co.uk` and `.co.uk` but it did't work.  

How should it add a wildcard domain?

Ysurac, 2022-04-03:
To add *.co.uk, simply add "co.uk". Check that you use OpenMPTCProuter as DNS. You may also need to wait for browser/computer DNS cache to be flushed.