Ysurac/openmptcprouter #1633: Optimal performance with udp traffic?

luka-diplomski, 2021-02-28:
Hello!
I have a (n00b :) question about optimizing performance of UDP traffic.

If I understood correctly (by default) all tcp traffic goes to vps using shadowsocks/mptcp, and all of UDP trafic goes to vps through glorytun TCP multipath vpn?

I'm planning to use VPN from my LAN to nordvpn US for usage of US vod services. For now I'm using openvpn tcp tunnel and it is working sub-optimal (approx 40% bandwidth lost). Whats the recommendation of optimal bandwidth? Switch to glorytun UDP and use Openvpn UDP/or wireguard? Or use openvpn udp/wireguard through glorytun TCP?

Has anybody done some testing and is willing to share his results, or should I start a weekend benchmarking session and share my results? : )

Thanks!

henry18hn, 2021-02-28:
I am in US using a vps in my region, but my streaming is blocked unless bypassed. Im glad to do some testing but I'm not sure where to start. I have not changed any settings after startup.

Ysurac, 2021-02-28:
Some site block IPs from some VPS providers.
For the VPS, you can use what work the best in your case. By default Shadowsocks is used for TCP and Glorytun TCP for UDP/ICMP/... You can use try V2Ray as proxy that work for TCP and UDP but use more CPU, or try another VPN.

luka-diplomski, 2021-02-28:
Thank you for the reply!

Ill try different combinations (v2ray/glorytun udp/wireguard/opentcp udp/tcp) in the next few days and post here if you dont mind.. maybe somebody else find it useful.

I would try today but I'm in a countryside with 2x 4g and Sundays here mean 1-5mbps  max :( Waiting on Elon to save us :D

hairydogltd, 2021-02-28:
Waiting for Elon? Not me. I was offered the Starlink service:
Setup fee: £439
Service: £89 / month
Post & Packing: £54
A fair bit more than I'm prepared to pay for: 
"During beta, users can expect to see data speeds vary from 50Mb/s to 150Mb/s and latency from 20ms to 40ms in most locations over the next several months as we enhance the Starlink system. There will also be brief periods of no connectivity at all."
I think I'll wait for 5G to arrive.

luka-diplomski, 2021-02-28:
Send it my way please. Ill be happy to buy it.

Im paying apporx 85gpb/month for 2x 4g lines that goes from (combined)  8-30mbps by day, and 30-50mbps by night.
Also I have 4x 4g antenas on my roof that costed me around 250gbp to buy and set up :(

luka-diplomski, 2021-02-28:
Btw I figured out my problem is that openvpn only goes through one WAN interface (master) so Ill open another issue and close this one soon. TY all

hairydogltd, 2021-02-28:
Ouch.
Three 4G unlimited, £16 per month. 
EE 4G unlimited, £20 per month for first sim, £10 per month for subsequent sims.
The Three one needs careful setting up. The nearest, strongest signal is from an overloaded base and give terrible speeds. 
At twice the distance, luckily in a different direction, there's a fast base so putting the antenna behind a stone wall blocks the slow one's signal.

luka-diplomski, 2021-02-28:
I will close for now, issue is that my tunnel doesn't go multipath :(

closed 2021-02-28 by luka-diplomski

hairydogltd, 2021-02-28:
To check availability for your location, visit Starlink.com and re-enter your service address. Thank you for your interest in Starlink and your continued support!

luka-diplomski, 2021-02-28:
Sometime in 2022 :(, they are accepting 99$ preorders in my area

henry18hn, 2021-02-28:
I have 2 4g setups as well, close to $3000.00 worth of gear, almost $100 a month for 30 down maybe 10 up, when im lucky. 90 to 100 ms latency. I paid the Starlink deposit also, may be here by fall or winter