Ysurac/openmptcprouter #1636: OpenVPN TCP vpn goes only through master WAN interface (no multipathing)

luka-diplomski, 2021-02-28:
## Expected Behavior
I installed ubuntu vm on my server and I'm using it as a OpenVPN router/dns in a way that I set my TV's GW and DNS to ubuntu VM so that all traffic from (only) my TV goes through vpn tunnel. 

I'm using OpenVPN TCP VPN tunnel.

## Current Behavior
Everything works except that I can see that the traffic goes only through 1 WAN interface, that is set as master in OMR mptcp.

## Specifications

  - OpenMPTCProuter version: v0.57.3
  - OpenMPTCProuter VPS version: 0.1024 5.4.86-mptcp
  - OpenMPTCProuter platform: RPI3 x64

Ysurac, 2021-02-28:
OpenVPN TCP is the slowest available VPN.
I don't understand your configuration.
For MPTCP part, check that all is green in status page and check in Network->MPTCP that MPTCP is not blocked on any wan.

luka-diplomski, 2021-02-28:
MPTCP is working perfectly and I get bonded speeds. Checked both with built in OMR graph tools, and by doing speedtest directly connected over wifi to wan1, and then to wan 2. I get combined bandwidth! :)

On my pc (that is getting dhcp from OMR), so gw and dns is ok, and I get bonded speeds in speedtest. If I start openvpn tcp vpn connection on that PC I can see that traffic is going only through master MPTPC wan link. When I switch "master" and "enabled" wan links config, then traffic is going only through the other one.

I expected that openvpn tcp vpn connection will utilize multipath tcp, both wlans bonded? Did I underatand something wrong? If yes, what is the best way to utilize openvpn using multipath, if posible?

Thanks for reply and thank you for your patience.

Ysurac, 2021-02-28:
Ok, so a custom OpenVPN install. Where is the OpenVPN server ? Not on OpenMPTCProuter VPS part ?

luka-diplomski, 2021-02-28:
No, local machine in LAN

luka-diplomski, 2021-02-28:
If Im not mistaken best solution would be to install openvpn client to vps and initiate tunnel from there. But then all of my traffic would go to vpn? And I want only 1 of my devices(ip/mac) going through vpn..

Another option would be to use another vps with configure tunnel, and add it to OMR.. and then set the rule to force only that device through second vps?

Are there any smarter/cheaper alternative to achieve what I want to do without buying second vps, but still utilize wan bonding since my speeds are pathetic :/

Thank you!

luka-diplomski, 2021-03-02:
#1536

closed 2021-03-02 by luka-diplomski