Ysurac/openmptcprouter #1643: Support for MikroTik hEX (RB750Gr3)

milankragujevic, 2021-03-02:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
Request to support OMR for MikroTik hEX (RB750Gr3)

**Describe why you are asking for this support?**
I have the router and would either like a confirmation that it could run OMR if I tried to build for it (without any encryption of course, but I think it has hardware crypto engine according to MikroTik) or maybe it even being added as a device.

**Additional context**
Quad Core MIPS 880 MHz SoC
256 MB RAM
16 MB Flash

Ysurac, 2021-03-03:
With only 16MB flash this seems really difficult.

extart, 2021-03-03:
> With only 16MB flash this seems really difficult.

It support SD Card, can be used extroot

milankragujevic, 2021-03-29:
@Ysurac regarding this, I see in the changelog for v0.58beta1 that you added support for Ubiquiti EdgeRouter X. I want to add that it is very very similar to the MikroTik hEX v3, other than the hEX having 1/8th the Flash (but supports SD card). The SoC is the same (well, MT7621AT vs MT7621A). I would ask you to confirm if in your opinion it is possible to run OMR off an SD card on the hEX if some porting is done as per OpenWrt which already supports it. 

In any case, here are deviwiki links, you can take a look:

https://deviwiki.com/wiki/MikroTik_RouterBOARD_hEX_v3_(RB750Gr3)
https://deviwiki.com/wiki/Ubiquiti_Networks_EdgeRouter_X

Ysurac, 2021-03-29:
Maybe possible but this seems difficult, flash layout seems to allow only 1.9Mo for kernel size.

milankragujevic, 2021-03-29:
Understood. I will just buy the ER-X. Can you tell me about what speeds do you get with it, for example combining 2 WAN connections, DOCSIS 100/8 mbps and LTE 2/50 Mbps (I limit download to 2 Mbps so I only use the LTE data plan for uploads). I woud expect around 100/50 Mbps. I can disable encryption for proxy and vpn.

closed 2021-05-14 by milankragujevic