Ysurac/openmptcprouter #1644: Slow Bandwidth and iperf Problem

easycloudcompany, 2021-03-03:
Hi, on my installation seems the bandwidth is very slow and also the ping latency is haigh

I've seen similar issue here and i've tryed with iperf test on shell and on Webgui but it does not work  see following image

[image]
[image]

and some time on bond-VPS there is an error  **"impossible obtain public IP by ShadowSocks!**

[image]

My configuration is the following
System:

[image]

Ysurac, 2021-03-03:
Please use template.

There is no master interface defined, this must be the case.

For iPerf some servers can answer or not sometimes, they are limited.

To test speed, via SSH or Services->terminal there is ```omr-test-speed``` to test aggregated speed and ```omr-test-speed <wan>``` to test <wan> speed. After at least 2 minutes can be stopped via Ctrl+C.

For ping problem, you can enable SQM.

easycloudcompany, 2021-03-03:
Thanks and sorry I'll use the template if needed in future.

BR

closed 2021-06-07 by github-actions[bot]