Ysurac/openmptcprouter #1645: Problem with expired certificate of openmptcprouter

jfh75, 2021-03-03:
Hello,

I have following issue trying to reinstall my VPS : 

Ign:3 https://repo.openmptcprouter.com stretch InRelease                                                                      
Err:4 https://repo.openmptcprouter.com stretch Release                                                                       
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 2001:41d0:2:7fc0::4 443]
Hit:5 http://security.debian.org/debian-security buster/updates InRelease
Reading package lists... Done
E: The repository 'https://repo.openmptcprouter.com stretch Release' no longer has a Release file.

It seems the certificate has expired. Is there a way to go on anyway ?

Thank you for your help,

JFH

Ysurac, 2021-03-03:
Fixed.
repo.openmptcprouter.com will be moved to another server soon.

closed 2021-03-03 by Ysurac

jfh75, 2021-03-03:
The Yannick touch, no pb lasts more than 48 seconds.... ah ah !
Thank you so much, I indeed screwed my VPS, it is working fine now !

JFH