Ysurac/openmptcprouter #1646: Probleme bande passante

Redven-Biker, 2021-03-03:
Bonjour après plusieurs recherche je ne trouve pas la solution a mon problème je vous explique

j'ai tout bien renseigner et tout bien brancher j'ai fait des test de ping de mon VPS OVH qui m'annonce bien +850 mega mais le problème c'est que sans faire cette manip j'ai 20 méga et après toute la configuration effectué je suis toujours au même débit (20 méga) sachant que j'ai bien désactiver le dhcp et tout se qui était demandé j'ai suivi 3 vidéos tuto et des forum donc je pense que je ne me suis pas tromper

merci d'avance pour votre aide :D

Ysurac, 2021-03-04:
Il faut utiliser le template...

Sinon tout est vert dans la page status dans Système->OpenMPTCProuter ?
Que donne via SSH ou via Services->Terminal ```omr-test-speed``` et ```omr-test-speed wan1``` (en remplaçant wan1 par une interface existante et en faisant ça pour chaque interface). Il faut le laisser au moins 2 minutes puis arrêter via Ctrl+C

Redven-Biker, 2021-03-04:
dans l'état de OpenMPTCProuter tout est en vert ( PC / Router / VPS et wan1 ) par contre la commande omr-test-speed je ne peut pas l'executé sa me dit que la commande est invalide