Ysurac/openmptcprouter #1647: Test Ping Serveur Erreur

JeedHome44, 2021-03-04:
Bonjour,

Depuis un bon moment, je constatais des problèmes car mon wan2 passait souvent en erreur serveur après 1 seconde.

Je pensais que ça venait de la connexion entre OMR et mon routeur 4G.

J'ai regardé dans les paramètres avancé et j'ai vu qu'il y avait une option pour désactiver le test Ping vers le serveur.

Depuis que j'ai coché cette option, je constate une nette amélioration et plus aucun problème.

Je pense donc que tous les soucis que j'avais depuis plusieurs mois venaient de ce test Ping vers le serveur.

Y a-t-il un impact grâce à laisser ce test désactivé?
Est-il bugué?

Je voulais partager mon expérience si d'autres ont le même problème.

Ysurac, 2021-03-04:
Cette option arrête simplement le ping vers le serveur de la page status et sert donc à éviter d'avoir le message "pas de réponse du serveur après 1 seconde" pour les connexions bloquant le ping.

JeedHome44, 2021-03-04:
Mais si je l'active ça me bloque pas mal de chose et surtout sur le site Amazon ca rame énormément.
Et en cochant l'option pour désactiver le Ping. J'ai fait pleins de tests aujourd'hui et c'est bien mieux.
Je retrouve la vitesse de navigation web que j'avais au tout début de l'installation d'OMR.
J'avais ce problème de test Ping sur mon wan2 4G Free. Est-ce que Free le bloque et met donc en défaut le wan2 sur OMR?

Vu que ça marche bien avec le test Ping serveur désactivé, je vais laisser comme ça 😉

closed 2021-04-21 by JeedHome44