Ysurac/openmptcprouter #1648: omr-test-speed only works without specifying an interface

hairydogltd, 2021-03-04:
## Expected Behavior

When I try to test the speed of individual wan connections, the speed test should connect and test the speed
## Current Behavior

I get more or less the same error each time. 

omr-test-speed wan1
[ping the servers]
Best server is http://lon.download.datapacket.com/10000mb.bin, running test:
ipset v7.6: Syntax error: cannot parse 185.59.221.2189.187.167.4195.181.164.16: resolving to IPv4 address failed

omr-test-speed wan2
[ping the servers]
Best server is http://lon.download.datapacket.com/10000mb.bin, running test:
ipset v7.6: Syntax error: cannot parse 89.187.167.4185.59.221.21195.181.164.16: resolving to IPv4 address failed

omr-test-speed wan3
[ping the servers]
Best server is http://par.download.datapacket.com/10000mb.bin, running test:
ipset v7.6: Syntax error: cannot parse 185.93.2.9185.93.2.21185.93.2.16: resolving to IPv4 address failed

Nothing appears in the log

## Possible Solution

## Steps to Reproduce the Problem

  1. log in as root
  2. omr-test-speed wan1
  3.

## Context (Environment)

I want to test the speed of individual wan connections and I cannot

omr-test-speed (without specifying a wan) works as expected.

I am using a local DNS server, running pihole - everything seems to be working properly

## Specifications

  - OpenMPTCProuter version: 0.57.3 / 5.7.3r0 
  - OpenMPTCProuter VPS version: 0.1024.5.86-mptcp
  - OpenMPTCProuter platform: RPI4

Ysurac, 2021-03-04:
Duplicate of #1547
Resolved in develop branch, will be solved in next release.

closed 2021-03-04 by Ysurac

Ysurac, 2021-03-04:
It's not related to the issue and it's not something to fix, can be an enhancement.