Ysurac/openmptcprouter #1650: Cannot install kmod-macremapper

SonicFM, 2021-03-05:
## Expected Behavior
First, thank you for your incredible work, Ysurac!
Second, sorry if this is the wrong category.

I tried installing OpenWRTs "luci-app-dcwapd" and therefore also "dcwapd" which depends on the ["kmod-macremapper"](https://openwrt.org/packages/pkgdata/kmod-macremapper) package.

Expected Behavior is (if this package would be availabe) it installs all dependend packages and then it should be usable as desired.
[image]

Translation (if needed):
Installiert = installed
Nicht installiert = not installed
Nicht verfügbar = not available
Abhänigkeiten = Dependencies

## Current Behavior
As the ["kmod-macremapper"](https://openwrt.org/packages/pkgdata/kmod-macremapper) package is not available in your [Repo](https://download.openmptcprouter.com/release/v0.57.3/bpi-r2/targets/mediatek/mt7623/packages/). Therefore the installation of the disired packages isnt possible.
However it is available in [OpenWRTs Snapshot-Repo](https://downloads.openwrt.org/snapshots/targets/mediatek/mt7623/kmods/5.4.83-1-ae090b5438487c6fc97780493e702354/).

If this Package can be compiled for the 5.4.83+ Kernel it should be possilbe to install and use the packages as desired.

## Specifications
  - OpenMPTCProuter version: v0.57.3 r0+15225-bfc433e
  - OpenMPTCProuter VPS version: currently none (not installed)
  - OpenMPTCProuter platform: Bananapi BPI-R2

Ysurac, 2021-03-05:
The module will be compiled in next release.

closed 2021-06-09 by github-actions[bot]

SonicFM, 2021-06-16:
Hi Ysurac! :)
Thanks for your fantastic work again!

Sadly this issue did not changed with your new release v58.x.
Could you have another look into it?

I'm still getting the error: "Required dependent package kmod-macremapper is not available in any repository."