Ysurac/openmptcprouter #1651: VPS blocks ICMP

davidebono1, 2021-03-05:
## Expected Behavior
(not really expected but i was wondering if OMPTCP could work in this condition)
OMPTCP works even if VPS blocks ICMP

## Current Behavior
If the VPS provider blocks ICMP then OMPTCP fails to connect.
In the status page OMPTCP reports:
"Can't access and use server part
Can't ping server
Can't contact Server Admin Script (No result - No API script answer)"

## Specifications

  - OpenMPTCProuter version: v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: 0.1024
  - OpenMPTCProuter platform: x86_64

I'm testing a new italian provider called Loclix but unfortunately they block ICMP packets.
I was wondering if ICMP is strictly required for OMPTCP or if maybe there is some workaround that can make OMPTCP work even in this condition.
Thanks

Ysurac, 2021-03-05:
ICMP is not required on VPS part.
The error on server part is not related to ICMP problem. VPS was rebooted after install ? Can you access https://<ipofserver>:65500/ ?

davidebono1, 2021-03-05:
Thanks for the clarification!
Yes, i've rebooted the VPS multiple times. 
Can you explain how i can access https://:65500/ ? If i put it in the browser bar it just searches it with the search engine.
Thank you!

Ysurac, 2021-03-05:
I modified my comment, the text was removed ;)

davidebono1, 2021-03-05:
Thanks :)
I just tried it but it doesn't work, Firefox says "We’re having trouble finding that site" so i guess it's having problems resolving the address.
If it can help i noticed that even if on the status page it gives many errors i can still browse the web but the speed is very low 
[image]

Are there maybe any logs i can send that can help figure out the problem?

Ysurac, 2021-03-05:
Direct ouput mean that speed is not aggregated.
If you can go to https://ipofserver:65500/ (replace ipofserver by ip of server), then this may be port > 65000 blocked by provider or API not running on server (you can check using ```ps aux | grep omr``` on the VPS via SSH).

davidebono1, 2021-03-05:
Well this is embarrassing...
I literally typed https://ipofserver:65500/ ... sorry...
Ok with the correct ip of my server it loads correctly and i see the message "Welcome to OpenMPTCProuter Server part".

I tried to change the master gateway and it seems like most errors are gone
[image]
(i guess the messages "No Server ping response" on the right will probably stay due to the VPS blocking ICMP right?)

Now i can browse the web with my VPS IP but the aggregation still isn't working, looks like it's using just the master gateway.

Ysurac, 2021-03-05:
The "No Server ping repsonse after 1 second" message can be disabled in "Advanced settings" tab using "Disable server ping".
For the VPN tunnel DOWN, if the message stay you should check in "State->System log" what is related to glorytun.

davidebono1, 2021-03-05:
Thanks!
The message VPN tunnel DOWN unfortunately stays,
this is the system log up until the last mention of glorytun.
```
Fri Mar  5 14:07:55 2021 user.notice ucitrack: Setting up non-init /etc/config/fstab reload handler: /sbin/block mount
Fri Mar  5 14:07:55 2021 user.notice ucitrack: Setting up /etc/config/system reload trigger for non-procd /etc/init.d/led
Fri Mar  5 14:07:55 2021 user.notice ucitrack: Setting up /etc/config/system reload dependency on /etc/config/luci_statistics
Fri Mar  5 14:07:55 2021 user.notice ucitrack: Setting up /etc/config/system reload dependency on /etc/config/dhcp
Fri Mar  5 14:07:55 2021 user.notice ucitrack: Setting up /etc/config/glorytun reload dependency on /etc/config/glorytun-udp
Fri Mar  5 14:07:55 2021 user.notice ucitrack: Setting up /etc/config/glorytun reload dependency on /etc/config/openmptcprouter-vps
Fri Mar  5 14:07:55 2021 user.notice SQM: Stopping SQM on wan1
Fri Mar  5 14:07:55 2021 user.notice ucitrack: Setting up /etc/config/glorytun-udp reload dependency on /etc/config/openmptcprouter-vps
Fri Mar  5 14:07:55 2021 user.notice ModemManager: hotplug: remove network interface ifb4wan1: event processed
Fri Mar  5 14:07:55 2021 user.notice ModemManager: hotplug: event reported: action=remove, name=ifb4wan1, subsystem=net
Fri Mar  5 14:07:55 2021 user.notice ucitrack: Setting up /etc/config/shadowsocks-libev reload dependency on /etc/config/omr-tracker
Fri Mar  5 14:07:55 2021 user.notice SQM: Starting SQM script: simple.qos on wan1, in: 0 Kbps, out: 0 Kbps
Fri Mar  5 14:07:55 2021 user.notice SQM: Using generic sqm_start_default function.
Fri Mar  5 14:07:55 2021 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Fri Mar  5 14:07:55 2021 user.notice ucitrack: Setting up /etc/config/mlvpn reload trigger for non-procd /etc/init.d/mlvpn
Fri Mar  5 14:07:55 2021 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Fri Mar  5 14:07:55 2021 daemon.warn [7551]: <warn>  [TMP_IFB_4_SQM] invalid sysfs path read for net/TMP_IFB_4_SQM
Fri Mar  5 14:07:55 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar  5 14:07:55 2021 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Fri Mar  5 14:07:55 2021 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Fri Mar  5 14:07:55 2021 daemon.debug dnsmasq[4047]: listening on wan2(#17): fe80::45a:26ff:fe72:7c01%wan2 port 53
Fri Mar  5 14:07:55 2021 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Fri Mar  5 14:07:55 2021 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Fri Mar  5 14:07:55 2021 daemon.warn [7551]: <warn>  [TMP_IFB_4_SQM] invalid sysfs path read for net/TMP_IFB_4_SQM
Fri Mar  5 14:07:55 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar  5 14:07:55 2021 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Fri Mar  5 14:07:55 2021 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Fri Mar  5 14:07:55 2021 user.notice SQM: simple.qos was started on wan1 successfully
Fri Mar  5 14:07:55 2021 user.notice ModemManager: hotplug: add network interface ifb4wan1: event processed
Fri Mar  5 14:07:55 2021 user.notice ModemManager: hotplug: event reported: action=add, name=ifb4wan1, subsystem=net
Fri Mar  5 14:07:55 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar  5 14:07:56 2021 user.notice firewall: Reloading firewall due to ifup of wan1 (wan1)
Fri Mar  5 14:07:56 2021 user.notice ModemManager: hotplug: checking if ModemManager is available...
Fri Mar  5 14:07:56 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty28, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty28
Fri Mar  5 14:07:56 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty28, subsystem=tty
Fri Mar  5 14:07:56 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar  5 14:07:56 2021 user.notice OMR-VPS: Can't get vps token, try later
Fri Mar  5 14:07:56 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyS15, subsystem=tty, sysfspath=/sys/devices/platform/serial8250/tty/ttyS15
Fri Mar  5 14:07:56 2021 user.notice ModemManager: hotplug: event reported: action=add, name=ttyS15, subsystem=tty
Fri Mar  5 14:07:56 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar  5 14:07:56 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty56, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty56
Fri Mar  5 14:07:56 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty56, subsystem=tty
Fri Mar  5 14:07:56 2021 daemon.info glorytun-udp: starting glorytun-udp vpn instance vpn
Fri Mar  5 14:07:56 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar  5 14:07:56 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty18, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty18
Fri Mar  5 14:07:56 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty18, subsystem=tty
[... cut ...]
Fri Mar  5 14:08:16 2021 daemon.err glorytun-udp[14115]: running on device tun0 as pid 14115
```
I guess the line: "daemon.notice procd: /etc/rc.d/S90glorytun-udp: Cannot find device "tun0" might be causing my problem but in the interfaces tab i see the tun0 device connected.
Any idea what could be causing this problem?

Anyway i tried to use GlorytunTCP instead of GlorytunUDP and it works without errors so i guess i can just live with GlorytunTCP for now.
Thanks a lot for your help, truly a great project!

closed 2021-06-09 by github-actions[bot]