Ysurac/openmptcprouter #1653: operation not supported error

Ajowan84, 2021-03-05:
[image]

I have a permanent error. See screenshot 
But OMPTCPR seems to be working fine

## Specifications

  - OpenMPTCProuter version:  openmptcprouter v0.57.3 r0+15225-bfc433e
  - OpenMPTCProuter VPS version: 0.1024 5.4.86-mptcp
  - OpenMPTCProuter platform: vmware ESX : openmptcprouter-v0.57.3-r0+15225-bfc433efd4-x86-64-generic-squashfs-combined.ova

Ysurac, 2021-03-05:
It's a filesystem error and it's why I prefer ext4: I already had some strange error with squashfs images.

Ajowan84, 2021-03-05:
OK.

And what is the better choice ?
openmptcprouter-v0.57.3-r0+15225-bfc433efd4-x86-64-generic-ext4-combined-efi.ova
or
openmptcprouter-v0.57.3-r0+15225-bfc433efd4-x86-64-generic-ext4-combined.ova

Thx

Ysurac, 2021-03-05:
You don't need EFI for VMware.

Ajowan84, 2021-03-05:
Thank you very much for these quick answers.

closed 2021-03-05 by Ajowan84

Adorfer, 2021-03-07:
This messages appears when trying use a flash-filesystem on non-flash-drives (e.g. not supporting DISCARD operation)
normally all flash drives (emmc, spi, sata-SSD, NVRAM) should be able. If you use squashfs on a hypervisor, enable "discard" (or say "emulate SSD", if the option is named like that)

Ajowan84, 2021-03-07:
@Adorfer thank you for that clarification !