Ysurac/openmptcprouter #1655: Recursive DNS on VPS side.

k0gen, 2021-03-07:
Wouldn't it be a good idea to have unbound service running on VPS side?
This would be preferable use case especially when running with Pi-hole.
The UCI interface is there so how about adding an option to switch between local/remote unbound instance?

Ysurac, 2021-03-07:
Unbound is used on router side because DNS must be usable when VPS is not.

Ysurac, 2021-03-07:
For Pi-Hole there is a script available in the doc, and when it's installed Pi-Hole is used without issues.

k0gen, 2021-03-08:
I guess I should have added "for better privacy" in the beginning. 

Benefit: Privacy - as you're directly contacting the responsive servers from your VPS the traffic stays outside, no server can fully log the exact paths you're going, as e.g. the Google DNS servers will only be asked if you want to visit a Google website, but not if you visit the website of your favorite newspaper, etc.

Ysurac, 2021-03-08:
By default when VPS is enable, DNS traffic use the VPS as exit like all traffic.
Also root DNS servers are always used by default because VPS providers DNS are not always good or use some public DNS.
But if you really want that, Pi-Hole can be installed on VPS but if I remember it only use some public DNS, so not really good for privacy.

k0gen, 2021-03-08:
Yes that is correct. The idea is to become your own upstream DNS by using Unbound on VPS. Here is my VPS Pi-hole settings regarding that:
[image]