Ysurac/openmptcprouter #1656: wan connexion stop after connecting a second RP4

bastienvander, 2021-03-07:
## Expected Behavior

Hello,
I want to use OpenMPTCP with an OVH VPS to reach a Nextcloud server behind a 4G router from outside with the VPS public IP.

## Current Behavior

I have 2 different Raspberry Pi 4 connected by Ethernet to a Huawei B715S 4G router. The first runs Nextcloud and the second runs OpenMPTCP (64bits image download from the website). If I connect both at the same time on my 4G router, every device on the local network is disconnected from the Internet. But if I keep an Ethernet connection between the OpenMPTCP PI4 and the router, and use a Wi-Fi connection (instead of Ethernet) between the Nextcloud PI 4 and the router, everything works ! I also tried to use my Nextcloud server on a PI 3 with Ethernet, and it works too. So do you have an idea of why when I connected the two PI 4 by Ethernet at the same time, every device is disconnected? Than You for your help.
## Specifications

  - OpenMPTCProuter version:   Version 0.57.3
  - OpenMPTCProuter VPS version:  Version 0.1024 5.4.86-mptcp
  - OpenMPTCProuter platform:  RPI4 2Go

Ysurac, 2021-03-07:
Check that Nextcloud RPI IP is not already used and that it have no DHCP server enabled.

bastienvander, 2021-03-07:
Thank You for your answer.

I tried to disable dhcpcd.service in the Nextcloud server config, but nothing change. The local IP of the server isn't already in use ...

closed 2021-06-11 by github-actions[bot]