Ysurac/openmptcprouter #1658: rpi 4 usb boot

user747, 2021-03-07:
## Expected Behavior
run omr on sata/nvme with compatible adapter
testing and confirmed working on a vanilla snapshot build of openwrt by editing cmdline.txt and updating with ``root=PARTUUID=<UUID of the USB drive rootfs partition>``

## Current Behavior
there is an error stating the start4.elf does not support USB boot. this firmware seems to be from April 2020

## Possible Solution

update start4.elf to newest version
## Steps to Reproduce the Problem

  1. write image to ssd
  2. update cmdline.txt with correct PARTUUID
  3. firmware does not support usb boot

## Context (Environment)

ssd/nvme drives are faster and more reliable than microsd

## Specifications

  - OpenMPTCProuter version: openmptcprouter-v0.57.3-r0+15225-bfc433efd4-bcm27xx-bcm2711-rpi-4-ext4-factory.img
  - OpenMPTCProuter VPS version: n/a 
  - OpenMPTCProuter platform: rpi4

Ysurac, 2021-03-07:
Should be a newest version on OpenMPTCProuter snapshot and will be latest when released.

user747, 2021-03-07:
tested this snapshot and got same error. 

``https://download.openmptcprouter.com/develop/5.4/rpi4/targets/bcm27xx/bcm2711/openmptcprouter-v0.57.3-r0%2B15829-482c9ff289-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz``
 ``05-Mar-2021 23:02``

techdeisel, 2021-04-25:
> 
> 
> tested and confirmed working in latest snapshot
> 
> `openmptcprouter-v0.57.3-r0%2B16138-ffeb37047e-bcm27xx-bcm2711-rpi-4-ext4-factory.img` `09-Mar-2021 19:47`

could you please help me with this ?

Ysurac, 2021-04-25:
@techdeisel you tried latest snapshot or v0.58rc1 ?