Ysurac/openmptcprouter #1662: Weird behaviour of telnet or nmap when going via mptcp

gwarf, 2021-03-11:
## Expected Behavior

Connection failing when using telnet to try to access a remote port that is not open.
nmap showing closed ports as closed.
(I used openmptcprouter.com as an example, hoping it would be OK with you)

```sh
❯ telnet openmptcprouter.com 22
Trying 145.239.70.188...
Trying 2001:41d0:303:3abc::...
telnet: Unable to connect to remote host: Connection timed out
Time: 0h:02m:10s 

❯ nmap openmptcprouter.com -p 22,4242

Starting Nmap 7.60 ( https://nmap.org ) at 2021-03-11 13:39 CET
Nmap scan report for openmptcprouter.com (145.239.70.188)
Host is up (0.0021s latency).
Other addresses for openmptcprouter.com (not scanned): 2001:41d0:303:3abc::
rDNS record for 145.239.70.188: ns3086971.ip-145-239-70.eu

PORT     STATE  SERVICE
22/tcp   closed ssh
4242/tcp closed vrml-multi-use

Nmap done: 1 IP address (1 host up) scanned in 0.05 seconds
```

## Current Behavior

Telnet reports that it connected to the remote endpoint, nmap reports that ports are open.
If I use OMB-bypass to go via a specific wan the results are as expected (timeout and closed ports).

## Steps to Reproduce the Problem

```sh
❯ telnet openmptcprouter.com 22
Trying 145.239.70.188...
Connected to openmptcprouter.com.
Escape character is '^]'.
Connection closed by foreign host.

❯ telnet openmptcprouter.com 4242
Trying 145.239.70.188...
Connected to openmptcprouter.com.
Escape character is '^]'.
Connection closed by foreign host.

❯ nmap openmptcprouter.com -p 22,4242
Starting Nmap 7.91 ( https://nmap.org ) at 2021-03-11 13:37 CET
Nmap scan report for openmptcprouter.com (145.239.70.188)
Host is up (0.00081s latency).
Other addresses for openmptcprouter.com (not scanned): 2001:41d0:303:3abc::
rDNS record for 145.239.70.188: ns3086971.ip-145-239-70.eu

PORT     STATE SERVICE
22/tcp   open  ssh
4242/tcp open  vrml-multi-use

Nmap done: 1 IP address (1 host up) scanned in 0.07 seconds
```

Obviously there isn't any 22 port open to the internet there nor something on port 4242.
Similar behavior with the few hosts and ports I tested.

## Context (Environment)

Tested from different boxes running Archlinux and MacOS X Big Sur (telnet and nmap from homebrew).
Setup with 3 ADSL lines (two Sosh and one SFR), and a scaleway VPS.
IPv4 network, VNP is not activated.

## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: 0.1024 5.4.74-mptcp
  - OpenMPTCProuter platform: RPI3

Ysurac, 2021-03-17:
I didn't check how nmap test if port are open or closed, but TCP traffic on OpenMPTCProuter use by default a socks5 proxy, this can be why nmap is not able to detect port state.
For me it's not a bug in OpenMPTCProuter.

gwarf, 2021-03-21:
Thanks @Ysurac.

An important point: do you confirm it acts similarly for you and it's not something specific to my setup?

With the switch `-sT` nmap will do a TCP connect check, which tries to open a TCP connection (using the connect system call) to the port: https://nmap.org/book/scan-methods-connect-scan.html

So it's not expected to be something very fancy and I don't really get how it can lead to those results.
(The fact is that even with a `telnet` call it shows the connection as opened while it isn't.)

I've tried to look at this with whireshark, and I do see a packet that is said to be coming back :/
You can see below the full three way handshake and at the end the RST send to close the connection:

[image]

If I try with OMR-bypass to output on a specific wan, it works as expected, and the three way TCP handshake is not done, I just receive a RST packet from the remote server:

[image]

Ysurac, 2021-03-21:
By design, Shadowsocks avoid any type of handshake so with nmap or any net-scanning tool that depends on that to check if port is open doesn't work.
Shadowsocks is used by default as proxy for TCP traffic.

gwarf, 2021-03-23:
OK, thanks.
So what are the options? Using omr-bypass for the full client as I'm doing? Disabling shadowsocks (thus for all the clients?)? replacing by something else? Any other option?
What would you advise for such a situation?

closed 2021-06-28 by github-actions[bot]