Ysurac/openmptcprouter #1663: The following signatures were invalid

GrosRems, 2021-03-12:
Hello

I am trying to pass the script to the VPS server.
I first got a message telling me that the grub directory was not found. So I placed the following commands (on Debian10):
`apt-get install grub`
`update-grub`

But now I have the following answer :

root@orawa-prod:~# wget -O - https://www.openmptcprouter.com/server/debian10-x86_64.sh | sh
--2021-03-12 12:14:08--  https://www.openmptcprouter.com/server/debian10-x86_64.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 2a0b:4d07:202::1, 185.172.149.132
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|2a0b:4d07:202::1|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 55783 (54K) [application/octet-stream]
Saving to: 'STDOUT'

-                                               100%[====================================================================================================>]  54.48K  --.-KB/s    in 0.005s

2021-03-12 12:14:10 (10.1 MB/s) - written to stdout [55783/55783]

Check user...
Check Linux version...
Check architecture...
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Remove lock and update packages list...
Hit:1 http://security.debian.org buster/updates InRelease
Hit:2 http://ftp.debian.org/debian buster InRelease
Hit:3 http://ftp.debian.org/debian buster-updates InRelease
Get:4 https://repo.openmptcprouter.com stretch InRelease [2122 B]
Err:4 https://repo.openmptcprouter.com stretch InRelease
  The following signatures were invalid: EXPKEYSIG 608FD17B2B24D936 OpenMPTCProuter <contact@openmptcprouter.com>
Reading package lists... Done
W: GPG error: https://repo.openmptcprouter.com stretch InRelease: The following signatures were invalid: EXPKEYSIG 608FD17B2B24D936 OpenMPTCProuter <contact@openmptcprouter.com>
E: The repository 'https://repo.openmptcprouter.com stretch InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

Thank you very much for this tool and your help

Ysurac, 2021-03-12:
I fixed the issue.
It's strange that grub was not installed. Are you sure you can boot on a custom kernel on your VPS ?

GrosRems, 2021-03-12:
Thank you !

I am not sure. I had requested an OVH VPS but they were afraid following the fires and took another VPS without looking at your list.
It is a hostinger VPS. How to verify that custom kernels are allowed?

Ysurac, 2021-03-12:
I found this "All our VPS plans are based on OVZ virtualization technology."
OpenVZ is not supported by OpenMPTCProuter because kernel can't modified.

GrosRems, 2021-03-12:
Nice
I will cancel this VPS and take an OVH. I hope that everything will roll

Thanks again

closed 2021-03-12 by GrosRems