Ysurac/openmptcprouter #1664: Problem when trying to add wifi or usb tethering interface to bonding with raspberry 4b 64bit

bluenor, 2021-03-14:
Good morning all my dear friends, congratulations on your project which is truly amazing. I bought a raspberry 4b which is working fine, I did the test before flashing with this image to make sure there were no hardware problems:

https://download.openmptcprouter.com/release/v0.57.3/rpi4/targets/bcm27xx/bcm2711/openmptcprouter-v0.57.3-r0+15225-bfc433efd4-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz

I added three interfaces with a switch that correspond to 2 4g routers and a fixed connection. Everything works great as expected and I actually noticed an increase in performance.

The problem arose when I tried to insert the usb interface with the intention of aggregating my mobile's connection as well. Although the interface is visible in the interface list, unfortunately it is not visible in the MPTCProuter status window. If I look in the statistics there is actually no bandwidth usage.

The same thing happens when I try to tether it via wifi tethering, I see it in the wifi network list but I can't see it in the MPTCProuter status window and there is no bandwidth usage when I look at the statistics.

I also noticed that if I try to insert two wifi in tethering, even in the list of wifi connections I only see one active and even in this case it does not show me the connections in the status window of MPTCProuter. 

I followed a step by step guide on youtube to do this but it was with the raspberry 3 and not the 4. 

I didn't put this question in the bugs because I don't know if I am doing something wrong or if it is a bug. 

Is there any kind soul willing to help me understand if I'm doing something wrong?

Ysurac, 2021-03-14:
You need to put it in wan firewall zone

bluenor, 2021-03-21:
Thank you very much for the answer in fact I did so. I was led astray by a video where they said it did it automatically. Also in the video they said you could hotspot infinite phones. In reality I think that since they are all in the 192.168.43.0 network it is possible to put only one. Correct?

closed 2021-06-25 by github-actions[bot]