Ysurac/openmptcprouter #1665: No Server ping response after 1 second - When connected to one VPS and not the other

bnhf, 2021-03-14:
## Expected Behavior
Regardless of connected VPS, individual cellular links would respond to ping and there would be a green check mark next to the connection.

## Current Behavior
When switching between a VPS set up on Vultr, and my own private VPS, the same two cellular links fail to show a green check mark.  Instead, there's a yellow warning triangle and the message "No Server ping response after 1 second".  Seems like a given WAN link would either respond to a ping, or not, and would not be influenced by connections to different VPS's.

## Possible Solution
Unsure.

## Steps to Reproduce the Problem

For me, switching between the two VPS's I have configured reproduces the problem consistently.

## Context (Environment)

With the warning triangle, rather than a green check mark, it's harder to know for sure all is well with a given link.

## Specifications

  - OpenMPTCProuter version: v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: v0.1024 5.4.74-mptcp
  - OpenMPTCProuter platform: x86_64)

When connected to my original Vultr VPS:

[image]

When connected to my new private VPS, moments later, using the same two cellular links:

[image]

Ysurac, 2021-03-14:
And the server answer to ping from wans in less than 1 second ?

bnhf, 2021-03-14:
> And the server answer to ping from wans in less than 1 second ?

Sounds like I've made bad assumption.  I've always assumed that the OpenMPTCProuter end was doing the pinging.  If it's the VPS doing the pinging, then what's happening makes sense.  Perhaps I need a more powerful VM, or dedicated machine, to run the VPS?  Also I imagine there's a way to prioritize my router traffic to and from my private VPS to improve ping times as well?

Ysurac, 2021-03-14:
ping is done from router to VPS using each wan, but ping must be not blocked on VPS for that to work.

bnhf, 2021-03-14:
> ping is done from router to VPS using each wan, but ping must be not blocked on VPS for that to work.

Thanks Yannick! In my case I had to allow ICMP "Echo Request" traffic to pass through my Ubiquiti USG firewall like so:

[image]

Green check marks appeared almost immediately.

closed 2021-06-18 by github-actions[bot]