Ysurac/openmptcprouter #1666: OpenMPTCProuter: File system in read only mode

Gaou3, 2021-03-15:
Hello,
I just installed the OpenMPTCProuter system on my VPS (OVH) and a Rasperry 4 by following this tutorial: https://www.tutos-informatique.com/adsl-4g-agregation/

But here is the result:
[image]

Thank you very much for your help !

## Expected Behavior
that all lights are green in the preview panel

## Current Behavior
Red light: OpenMPTCProuter --> file system in read only
Orange light: Direct Output --> No public IP via ShadowSoks, No tocken yet available, No API Script answer

Log:
Mon Mar 15 09:47:11 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:47:11 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:47:14 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:47:17 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:47:19 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:47:20 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:47:23 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:47:24 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:47:29 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:47:31 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:47:31 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:47:33 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:47:34 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:47:36 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:47:39 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:47:40 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:47:45 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:47:48 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:47:50 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:47:51 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:47:52 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:47:52 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:47:54 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:47:55 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:00 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:01 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:48:05 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:06 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:48:09 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:48:10 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:11 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:12 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:13 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:48:15 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:20 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:21 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:48:25 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:48:25 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:28 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:48:31 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:31 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:32 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:36 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:37 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:48:40 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:48:41 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:43 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:48:46 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:51 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:51 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:51 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:48:53 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:48:55 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:48:56 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
[... 44 lines cut ...]
Mon Mar 15 09:50:11 2021 daemon.info dnsmasq-dhcp[15100]: DHCPACK(eth0) 192.168.100.109 7e:d9:8c:32:19:81
Mon Mar 15 09:50:11 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:50:11 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:50:12 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:50:13 2021 daemon.info dnsmasq-dhcp[15100]: DHCPDISCOVER(eth0) 7e:d9:8c:32:19:81
Mon Mar 15 09:50:13 2021 daemon.info dnsmasq-dhcp[15100]: DHCPOFFER(eth0) 192.168.100.109 7e:d9:8c:32:19:81
Mon Mar 15 09:50:15 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:50:15 2021 daemon.info dnsmasq-dhcp[15100]: DHCPREQUEST(eth0) 192.168.100.109 7e:d9:8c:32:19:81
Mon Mar 15 09:50:15 2021 daemon.info dnsmasq-dhcp[15100]: DHCPACK(eth0) 192.168.100.109 7e:d9:8c:32:19:81
Mon Mar 15 09:50:18 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:50:18 2021 daemon.info dnsmasq-dhcp[15100]: DHCPDISCOVER(eth0) 98:90:96:a1:a8:fd
Mon Mar 15 09:50:18 2021 daemon.info dnsmasq-dhcp[15100]: DHCPOFFER(eth0) 192.168.100.130 98:90:96:a1:a8:fd
Mon Mar 15 09:50:18 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:50:21 2021 daemon.info dnsmasq-dhcp[15100]: DHCPINFORM(eth0) 192.168.100.232 08:36:c9:7c:84:bb
Mon Mar 15 09:50:21 2021 daemon.info dnsmasq-dhcp[15100]: DHCPACK(eth0) 192.168.100.232 08:36:c9:7c:84:bb Romain_Monica
Mon Mar 15 09:50:22 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Mon Mar 15 09:50:23 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
Mon Mar 15 09:50:28 2021 daemon.err /usr/bin/ss-local[13635]: getpeername: Socket not connected
etc ...

## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: 0.57.3
  - OpenMPTCProuter platform: RPI 4

Ysurac, 2021-03-15:
Filesystem is put as read only after error on FS. This can be due to hard reboot or bad SD card.
You need to reinstall RPI, and if it's not already the case check "Check filesystems before mount" in System->Mount Points.

Gaou3, 2021-03-15:
OK, thank you.
But I did what you said: reinstall OpenMPTCProuter on my Rasperry 4, and check "Check filesystems before mount".

And I now, i have this:

[image]

Ysurac, 2021-03-15:
Validate wizard again and check key.

Gaou3, 2021-03-15:
Ok, thank you, verythink is ok now !

[image]

But when I do a speedtest, my two internet speed are not added ...

When I connect only my 4g, no problem, I have a big speed. But when I connect ADSL + 4g, it seems to be lower than when there is only 4g connected ...

An idea ?

Ysurac, 2021-03-15:
You can check in Network->MPTCP, "Bandwidth" tab if all connections are used or not during a speedtest.
You can test speed via SSH using ```omr-test-speed``` for aggregated speed or ```omr-test-speed wan1``` for wan1 speed (ctrl+c to stop after 2 minutes).

Gaou3, 2021-03-15:
Ok, thank you, it seems to be very unstable in download
[image]
And ADSL are 4G seems to be linked ... strange ...

Gaou3, 2021-03-15:
Ok, everything is ok ! Thank you very much !

closed 2021-03-15 by Gaou3