Ysurac/openmptcprouter #1668: impression de double consommation de bande passante

gmorlet, 2021-03-15:
## Expected Behavior
comment fonctionne le cheminement des paquets pour la consommation data ?

## Current Behavior
Sur un site réseau local 192.168.100.x
openmptcprouter en 192.168.100.1 RPI4

* d'un côté connecté sur une box 4G avec 200Go de data mensuel et un tel tethering 4G
* de l'autre si on peut dire , vps ovh

depuis la mise place d'open et de la partie vps, j'ai l'impression d'un quasi doublement de la conso data

donc je me demande commence chemine ou si il manque un paramètrage

Ex client, rpi4 (routeur local) tunnel vers vps qui passe par le wan et qui revient pour que le wan va chercher les datas puis ;cheminement inverse donc 1 aller/retour de la même data par le WAN

merci des explications et du règlage à appliquer si besoin

Ysurac, 2021-03-15:
duplicate of #1660

closed 2021-03-15 by Ysurac

gmorlet, 2021-03-15:
exact, désolé, je viens de me rendre compte que j'avais effectué en anglais. j'avais recherché un titre en français et je pensais que j'avais mal fait.

oui j'ai installé déjà stats et bandwith mais difficile de voir exactement