Ysurac/openmptcprouter #1669: VPS disconnects when performing speedtest

imbdb, 2021-03-15:
## Expected Behavior
Should not disconnect when doing speedtest

## Current Behavior
Server disconnects when doing speedtest

## Specifications

  - OpenMPTCProuter version: Version 0.56.4
  - OpenMPTCProuter VPS version: Debian 10 - Version 0.1024 5.4.86-mptcp
  - OpenMPTCProuter platform: Raspberry Pi4

Ysurac, 2021-03-15:
If the VPS disconnect, then it's a VPS problem.
If it's a connection problem between VPS and router, enable SQM on WANs interfaces in the wizard.

imbdb, 2021-03-15:
thanks Yannick for your response.
VPS seems stable according to the logs and it will keep showing green until I do some work on the internet like opening a website or doing a speed test.

whenever I do a speedtest I below error on VPS
```
ss-server[2187]: server recv: Connection reset by peer
ss-server[2187]: server recv: Connection reset by peer
ss-server[2187]: server recv: Connection reset by peer
ss-server[2187]: server recv: Connection reset by peer
```

I cannot enable SQM because I am using 4G dongles for bonding.

Ysurac, 2021-03-15:
You can try to enable SQM without setting a max speed.
This disconnect even for a simple website ? In this case you can try to disable TCP Fast Open in System->OpenMPTCProuter, "Advanced settings" tab.

imbdb, 2021-03-16:
Still facing the issue.
Perhaps it not fault of OpenMPTCP router but result of low connection speeds on my 4G dongles (1-2Mbps only).

Thanks Yannick for creating this awesome project.
Will love to contribute in project,

closed 2021-03-16 by imbdb