Ysurac/openmptcprouter #1671: VPS cannot be reached due to shorewall blocking

dur3x, 2021-03-15:
## Expected Behavior
Connection correctly established to VPS

## Current Behavior
Server started then ping & ssh works a while but quickly loose the connection with this in logs:

```
Mar 15 20:46:21 ks3374040 root: Shorewall started
Mar 15 20:46:21 ks3374040 kernel: [  431.938942] Shorewall:OUTPUT:REJECT:IN= OUT=eno0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=104 TOS=0x10 PREC=0x00 TTL=64 ID=37016 DF PROTO=TCP SPT=65222 DPT=4916 WINDOW=22 RES=0x00 ACK PSH URGP=0 
```

## Possible Solution
Current Workaround is to disable shorewall at boot but probably the best would be to understand what's wrong and why I'm completely rejected (ssh on port 65222 + icmp denied)

## Steps to Reproduce the Problem
Install fresh install in debian 9

## Context (Environment)
kimsufi install with debian 9

Ysurac, 2021-03-16:
Debian 9 is migrated to Debian 10 by script, so you should install Debian 10.
You should check in /etc/shorewall/params.net that correct external interface was detected.

dur3x, 2021-03-16:
I can confirm you that the script updated well the debian 9 into 10 :-)
Indeed the interface was wrong in /etc/shorewall/params.net
````
NET_IFACE=enp1s0
````
instead of
`````
NET_IFACE=eno0
`````

I suspect that enp1s0 was perhaps the interface name before the upgrade into debian 10.

Well as usual, you are the king and gave me a workaround very quicky ;-) I'm so excited to test this new setup !

closed 2021-03-16 by dur3x