Ysurac/openmptcprouter #1672: Comment désactiver l'IPv6 ?

kdesroches89, 2021-03-15:
Bonjour,
Je ne souhaite pas utiliser d'IPv6 , comment désactiver l'IPv6 complètement ?

@Ysurac  Merci par avance pour l'aide

Merci
Kilian

Ysurac, 2021-03-16:
L'IPv6 est désactivé par défaut.
Pour l'activer/désactiver, c'est dans Système->OpenMPTCProuter, et la case "paramètres avancés"

kdesroches89, 2021-03-16:
C'est déja le cas mais j'ai quand même de l'IPv6 :

[image]

[image]

Ysurac, 2021-03-16:
Voir dans Réseau->Interfaces puis LAN, onglet "Serveur DHCP", puis "Paramètres IPv6" que tout est bien sur "désactivé".
Pour l'adresse 2a01:xxx elle peut provenir d'un autre matériel comme une freebox par exemple.

kdesroches89, 2021-03-16:
C'est effectivement la freebox mais visiblement il faut relancer celle ci pour que OpenMTCPRouter prenne la main en amont . En cas de relance du routeur la freebox passe avant . La freebox n'a pas d'ipv6 activé , de dhcp6 actif ou de dns6 actif.

Merci @Ysurac

closed 2021-03-17 by kdesroches89