Ysurac/openmptcprouter #1673: Big upload difference between 2 computers ...

Gaou3, 2021-03-16:
Hello,

I tested the bandwith in download and upload in 3 different computers. Mine (where I set the network with OpenMPTCProuteur), and 2 others. And I have a big difference in upload between thses computers... It's like if my PC use the 4G upload, and the two others computers use the ADSL upload ... 

In my PC, my results:
[image]

In PC 2, the results:
[image]

In PC 3, the results:
[image]

DHCP is OpenMPTCProuteur, and these 3 computers have an automatic IP and DNS

Thank you !

## Expected Behavior
I expect the sames bandwith in upload in every computers of my network

## Current Behavior
I have big upload differences in upload between 3 computers.

## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: 0.57.3
  - OpenMPTCProuter platform: RPI4

Ysurac, 2021-03-16:
If omr-bypass is not used to bypass these computers, result should be the same. Server used for speedtest is not the same in each test, this may be the problem.

hairydogltd, 2021-03-17:
run / cmd / ipconfig -all on each computer. 
Are they all using the correct default gateway (the OMR server IP)?

Gaou3, 2021-03-18:
Hello

I think I found the problem: my 4G connection does not work in ipV6 while my ADSL connection does. Suddenly if the connection is established in IPV6 or V4 the speed is not at all the same!

It may not be very nice, but I suddenly just disabled IPV6 on the computers.

Now I have another problem. I kept only the 4G connection on the network and I made several tests going directly through the 4G router and going through OMR.

With the 4G router my speeds are around 50 up and 20 down. Using OMR I have around 20 going up and 20 going down.

I did my tests with www.speedtest.net.
I also ran speedtest on the VPS server, I got around 200/200.

I specify that OMR runs on an RPI4.

Do you have any ideas of the problem?

Thank you

Ysurac, 2021-03-18:
Use  ```omr-test-speed``` to test aggregated speed and ```omr-test-speed wan1``` to test wan1 speed, via SSH or Services->Terminal. Ctrl+C to stop after at least 2 minutes.

Gaou3, 2021-03-18:
Thank you for your help

For omr-test-speed:

Best server is http://fra.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 9536M    1  153M    0     0  1228k      0  2:12:29  0:02:08  2:10:21 1181k
  
  For omr-test-speed wan1 :
  
Best server is http://ipv4.bouygues.testdebit.info/10G.iso, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 17 9536M   17 1708M    0     0   9.8M      0  0:16:08  0:02:53  0:13:15 10.6M

Ysurac, 2021-03-18:
Everything is green in status page ?
Proxy is still default shadowsocks ?
What is the VPS provider ?

Gaou3, 2021-03-18:
I purposely connected the 4G connection only. It is very green.
My ip is not assigned by OMR so I have an orange message telling me so. But that's all.

The default proxy is shadowsocks

VPS is OVH

Ysurac, 2021-03-18:
What is 4G provider ?

Gaou3, 2021-03-18:
Free

Ysurac, 2021-03-18:
I would need full configuration, in System->OpenMPTCProuter, last tab.

Gaou3, 2021-03-18:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
dscp.@classify[3].comment='OMR vpn'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
dscp.@classify[4].proto='udp'
dscp.@classify[4].class='cs6'
dscp.@classify[4].dest_port='65001,65301'
dscp.@classify[4].comment='OMR vpn'
dscp.@domains[0]=domains
dscp.@domains[0].name='googlevideo.com'
dscp.@domains[0].class='cs4'
dscp.@domains[0].comment='Youtube'
[... 1500 lines cut ...]
v2ray.omrrouting.domain='full:omr.lan'
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.radio0.country='00'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.mode='ap'
wireless.default_radio0.ssid='OpenWrt'
wireless.default_radio0.encryption='none'
wireless.default_radio0.skip_inactivity_poll='1'
wireless.default_radio0.network='wifi'

Ysurac, 2021-03-18:
I forgot to say without VPS IPs/key

Gaou3, 2021-03-18:
Oh ... thanks for the deletion

Gaou3, 2021-03-18:
You want me to respot?

Ysurac, 2021-03-18:
No, it's ok. I don't see any error in configuration.

Ysurac, 2021-03-18:
Can you check in State->System log if you see any error ?

Gaou3, 2021-03-18:
Here are the most recent errors I see:

Thu Mar 18 15:57:27 2021 daemon.err rngd[3120]: Initializing available sources
Thu Mar 18 15:57:27 2021 daemon.err rngd[3120]: [hwrng ]: Initialized
Thu Mar 18 15:57:27 2021 cron.err crond[3407]: crond (busybox 1.31.1) started, log level 5

It seems that OMR is not on time, 3:57 pm is 4:57 pm actually

Ysurac, 2021-03-18:
It's in time, but UTC timezone.

Ysurac, 2021-03-19:
OVH network is not really stable for now, some part are really slow.

closed 2021-03-25 by Gaou3