Ysurac/openmptcprouter #1674: Instabilité de OMR impossible à comprendre

JeedHome44, 2021-03-16:
## Expected Behavior

Bonjour,

Je tourne en rond depuis des mois avec ce problème et je n'arrive pas à résoudre les soucis.
Ces dernières semaines, j'ai fait des tests sur chacun de mes 2 clients internet et ce n'est pas eux le soucis.
Il y a donc bien un problème sur OMR que je n'avais pas au tout début quand je l'ai installé et qui est apparu dès que j'ai fait la première mise à jour.

Explication:
Continuellement, j'ai un message orange sur mon wan 4G m'informant que le serveur n'a pas répondu après 1 sec.
Ça a pour effet de rendre la connexion internet ultra lente avec une latence de fou.

Pour info, je n'ai jamais touché aux paramètres de OMR à part les By-pass et je fixe tous les IP en DHCP. C'est tout.

En voyant ça, j'ai donc tenté de mettre mon routeur 4G uniquement en 700 MHz au lieu de lui mettre toutes les bandes 4G. Au début tout est au vert puis quelques heures après ca recommence.

J'ai tenté de désactivé le test ping vers le serveur. Dés que je sauvegarde, tout passe au vert, le débit revient puis ça recommence...

Ce matin, j'avais un débit pourri, j'ai donc tenté de décocher l'option permettant de privilégier la latence TCP plutôt que le débit.
Direct, j'ai retrouvé un bon débit et tout au vert.

Et rebelote, la le wan 2 est encore orange et le serveur passe même de temps en temps au rouge.

Je ne comprends pas pourquoi dés que je modifie un truc et que je sauvegarde dans système, tout passe au vert et ça très bien quelques heures puis ensuite, c'est de nouveau horrible pour naviguer sur internet...

Tout est au vert sur les config du routeur 4G et tout est au vert sur la Freebox pourtant.

Le routeur 4G était en CPL avant. J'ai tiré un long câble ethernet par le mur pour aller au grenier et pareil. C'était bien quelques heures puis encore le serveur qui ne répond pas sur la 4G.

Pourtant, j'ai testé la carte SIM du routeur 4G et ça va très bien.

OMR se met donc en défaut systématiquement avec mon agrégation et par contre, la Freebox reste au vert sauf tout à l'heure ou je l'ai vu passer à l'orange. Est-ce parce que j'ai désactivé la latence TCP?

J'aurai vraiment besoin d'une prise en main à distance si possible pour voir ce qui se passe et pourquoi ça fait ça.

S'il faut des logs spécifiques, je peux les envoyer ici.

Dites moi ce qu'il faut fournir.

Dois-je dans un premier temps remettre la latence TCP et le test Ping serveur?

## Current Behavior

Si besoin de logs spécifiques, je peux le faire pour les montrer ici. Je ne sais pas quoi montrer pour l'instant.

## Specifications

  - OpenMPTCProuter version:  dernière dispo
  - OpenMPTCProuter VPS version:  dernière dispo
  - OpenMPTCProuter platform:  Pi 4B 2Go

Ysurac, 2021-03-16:
Il me faudrait les logs dans Etat->Journal Système et Etat->Journal noyau

JeedHome44, 2021-03-16:
Merci pour la réponse rapide.

Voici les logs pour le journal système

```
Tue Mar 16 16:01:09 2021 daemon.info dnsmasq-dhcp[1708]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Tue Mar 16 16:01:09 2021 daemon.err uhttpd[6534]: luci: accepted login on / for root from 192.168.100.40
Tue Mar 16 16:01:17 2021 daemon.info dnsmasq-dhcp[1708]: DHCPREQUEST(eth0) 192.168.100.49 90:9a:4a:92:d4:8a
Tue Mar 16 16:01:17 2021 daemon.info dnsmasq-dhcp[1708]: DHCPACK(eth0) 192.168.100.49 90:9a:4a:92:d4:8a Access-Point-Salon
Tue Mar 16 16:01:29 2021 daemon.info dnsmasq-dhcp[1708]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Tue Mar 16 16:01:29 2021 daemon.info dnsmasq-dhcp[1708]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Tue Mar 16 16:01:41 2021 daemon.info dnsmasq-dhcp[1708]: DHCPDISCOVER(eth0) 70:ee:50:00:14:86
Tue Mar 16 16:01:41 2021 daemon.info dnsmasq-dhcp[1708]: DHCPOFFER(eth0) 192.168.100.18 70:ee:50:00:14:86
Tue Mar 16 16:01:41 2021 daemon.info dnsmasq-dhcp[1708]: DHCPREQUEST(eth0) 192.168.100.18 70:ee:50:00:14:86
Tue Mar 16 16:01:41 2021 daemon.info dnsmasq-dhcp[1708]: DHCPACK(eth0) 192.168.100.18 70:ee:50:00:14:86 Station-Netatmo
Tue Mar 16 16:01:43 2021 daemon.info dnsmasq-dhcp[1708]: DHCPDISCOVER(eth0) 00:e0:4c:6d:63:03
Tue Mar 16 16:01:43 2021 daemon.info dnsmasq-dhcp[1708]: DHCPOFFER(eth0) 192.168.100.39 00:e0:4c:6d:63:03
Tue Mar 16 16:01:43 2021 daemon.info dnsmasq-dhcp[1708]: DHCPREQUEST(eth0) 192.168.100.39 00:e0:4c:6d:63:03
Tue Mar 16 16:01:43 2021 daemon.info dnsmasq-dhcp[1708]: DHCPACK(eth0) 192.168.100.39 00:e0:4c:6d:63:03 Mi-Box-Salon
Tue Mar 16 16:01:49 2021 daemon.info dnsmasq-dhcp[1708]: DHCPREQUEST(eth0) 192.168.100.49 90:9a:4a:92:d4:8a
Tue Mar 16 16:01:49 2021 daemon.info dnsmasq-dhcp[1708]: DHCPACK(eth0) 192.168.100.49 90:9a:4a:92:d4:8a Access-Point-Salon
Tue Mar 16 16:01:49 2021 daemon.info dnsmasq-dhcp[1708]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Tue Mar 16 16:01:49 2021 daemon.info dnsmasq-dhcp[1708]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Tue Mar 16 16:02:02 2021 daemon.info dnsmasq-dhcp[1708]: DHCPREQUEST(eth0) 192.168.100.54 d4:a6:51:06:4f:b7
Tue Mar 16 16:02:02 2021 daemon.info dnsmasq-dhcp[1708]: DHCPACK(eth0) 192.168.100.54 d4:a6:51:06:4f:b7 Pont-Zemismart
Tue Mar 16 16:02:10 2021 daemon.info dnsmasq-dhcp[1708]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Tue Mar 16 16:02:10 2021 daemon.info dnsmasq-dhcp[1708]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Tue Mar 16 16:02:14 2021 daemon.info vnstatd[7063]: Info: SIGHUP received, flushing data to disk and reloading config.
Tue Mar 16 16:02:14 2021 user.notice omr-tracker: Launching...
Tue Mar 16 16:02:17 2021 user.notice omr-tracker: Launched
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-local[3794]: using tcp fast open
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-local[3794]: enable TCP no-delay
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-local[3794]: initializing ciphers... chacha20-ietf-poly1305
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-local[3794]: listening at 127.0.0.1:1111
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-local[3794]: tcp port reuse enabled
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-local[3794]: udprelay enabled
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-local[3794]: udp port reuse enabled
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-local[3794]: running from root user
Tue Mar 16 16:02:18 2021 user.info : using tcp fast open
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3798]: enable TCP no-delay
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3798]: initializing ciphers... chacha20-ietf-poly1305
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3798]: listening at 0.0.0.0:1101
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3798]: tcp port reuse enabled
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3798]: UDP relay enabled
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3798]: udp port reuse enabled
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3798]: running from root user
Tue Mar 16 16:02:18 2021 user.info : using tcp fast open
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3807]: enable TCP no-delay
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3807]: resolving hostname to IPv6 address first
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3807]: initializing ciphers... chacha20-ietf-poly1305
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3807]: listening at 0.0.0.0:1100
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3807]: tcp port reuse enabled
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3807]: UDP relay enabled
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3807]: udp port reuse enabled
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3807]: running from root user
Tue Mar 16 16:02:18 2021 user.info : using tcp fast open
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3800]: enable TCP no-delay
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3800]: resolving hostname to IPv6 address first
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3800]: initializing ciphers... chacha20-ietf-poly1305
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3800]: listening at 0.0.0.0:1100
Tue Mar 16 16:02:18 2021 daemon.info /usr/bin/ss-redir[3800]: tcp port reuse enabled
[... cut ...]
Tue Mar 16 16:32:00 2021 daemon.info dnsmasq-dhcp[26386]: DHCPREQUEST(eth0) 192.168.100.18 70:ee:50:00:14:86
Tue Mar 16 16:32:00 2021 daemon.info dnsmasq-dhcp[26386]: DHCPACK(eth0) 192.168.100.18 70:ee:50:00:14:86 Station-Netatmo
Tue Mar 16 16:32:03 2021 daemon.err uhttpd[6534]: luci: accepted login on / for root from 192.168.100.30
Tue Mar 16 16:32:08 2021 daemon.info dnsmasq-dhcp[26386]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Tue Mar 16 16:32:08 2021 daemon.info dnsmasq-dhcp[26386]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
```

JeedHome44, 2021-03-16:
et le journal noyau:

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.83 (runner@fv-az50-423) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0+15225-bfc433efd4)) #0 SMP Tue Dec 15 19:35:00 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 504832
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 4096 pages used for memmap
[    0.000000]   DMA32 zone: 262144 pages, LIFO batch:63
[    0.000000] percpu: Embedded 21 pages/cpu s45144 r8192 d32680 u86016
[    0.000000] pcpu-alloc: s45144 r8192 d32680 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 496944
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:7A:49:37 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x37400000-0x3b400000] (64MB)
[    0.000000] Memory: 1836992K/2019328K available (7038K kernel code, 430K rwdata, 1840K rodata, 512K init, 539K bss, 116800K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x2dc/0x3dc with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000155] Console: colour dummy device 80x25
[    0.000394] printk: console [tty1] enabled
[    0.000427] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000449] pid_max: default: 32768 minimum: 301
[    0.000608] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.000659] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.001331] Disabling memory control group subsystem
[    0.002026] ASID allocator initialised with 32768 entries
[    0.002117] rcu: Hierarchical SRCU implementation.
[    0.002739] smp: Bringing up secondary CPUs ...
[    0.003136] Detected PIPT I-cache on CPU1
[    0.003186] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.003632] Detected PIPT I-cache on CPU2
[    0.003666] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.004170] Detected PIPT I-cache on CPU3
[    0.004203] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.004272] smp: Brought up 1 node, 4 CPUs
[    0.004300] SMP: Total of 4 processors activated.
[    0.004314] CPU features: detected: 32-bit EL0 Support
[    0.004327] CPU features: detected: CRC32 instructions
[    0.004380] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.004398] CPU: All CPU(s) started at EL2
[    0.004426] alternatives: patching kernel code
[    0.005393] devtmpfs: initialized
[    0.012312] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.012353] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.014104] pinctrl core: initialized pinctrl subsystem
[    0.014395] reg-dummy reg-dummy: no of_node; not parsing pinctrl DT
[    0.015008] NET: Registered protocol family 16
[    0.016403] DMA: preallocated 1024 KiB pool for atomic allocations
[    0.016951] cpuidle: using governor ladder
[    0.016981] cpuidle: using governor menu
[    0.017253] Serial: AMBA PL011 UART driver
[... 656 lines cut ...]
[    8.999428] usbserial: USB Serial support registered for suunto
[    9.005399] usbserial: USB Serial support registered for siemens_mpi
[    9.041618] brcmfmac mmc1:0001:1: no of_node; not parsing pinctrl DT
[    9.041704] brcmfmac mmc1:0001:2: no of_node; not parsing pinctrl DT
[    9.045017] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.214032] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[... cut ...]
[    9.345475] usbcore: registered new interface driver rtl8192cu
[    9.355730] usbcore: registered new interface driver ath9k_htc
[    9.364795] kmodloader: done loading kernel modules from /etc/modules.d/*
[   16.791410] bcmgenet: Skipping UMAC reset
[   16.796722] bcmgenet fd580000.ethernet: configuring instance for external RGMII
[   16.804209] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   16.813519] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
```

JeedHome44, 2021-03-19:
Bonjour,

Savez vous d'où vient mon problème?

Dés que le wan2 affiche pas de réponse serveur après 1 sec ça fou le bordel. Je fais une modification dans les paramètres avancés puis remet comme avant et ça remarche.

Il y a donc pour moi quelque chose qui plante dans OMR entre le routeur et le VPS. Serait-ce le VPN entre les 2?

Peut on se passer de VPN entre chez nous et le VPS?

Ysurac, 2021-03-19:
Il y a vraiment beaucoup de requêtes DHCP. Ca peut saturer OMR.

JeedHome44, 2021-03-19:
J'ai beaucoup d'appareil oui. C'est quoi la solution?
Environ 180 appareils

Ysurac, 2021-03-19:
Le soucis n'est pas le nombre d'appareil, mais que certains font des demandes DHCP en permanence.

JeedHome44, 2021-03-19:
Ah bon? On peut savoir lesquels?

JeedHome44, 2021-03-19:
Mon problème d'appareils Tuya contrôlés en local avec wifilightv2 qui se déconnecte vient donc de la aussi?

Ysurac, 2021-03-19:
Au moins ceux la: 
```
DHCPDISCOVER(eth0) 00:0e:c6:b6:65:53
Tue Mar 16 16:20:57 2021 daemon.info dnsmasq-dhcp[26386]: DHCPOFFER(eth0) 192.168.100.13 00:0e:c6:b6:65:53
Tue Mar 16 16:20:58 2021 daemon.info dnsmasq-dhcp[26386]: DHCPREQUEST(eth0) 192.168.100.13 00:0e:c6:b6:65:53
Tue Mar 16 16:20:58 2021 daemon.info dnsmasq-dhcp[26386]: DHCPACK(eth0) 192.168.100.13 00:0e:c6:b6:65:53 Mi-Box-Bureau
Tue Mar 16 16:21:01 2021 daemon.info dnsmasq-dhcp[26386]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Tue Mar 16 16:21:01 2021 daemon.info dnsmasq-dhcp[26386]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Tue Mar 16 16:21:06 2021 daemon.info dnsmasq-dhcp[26386]: DHCPREQUEST(eth0) 192.168.100.49 90:9a:4a:92:d4:8a
Tue Mar 16 16:21:06 2021 daemon.info dnsmasq-dhcp[26386]: DHCPACK(eth0) 192.168.100.49 90:9a:4a:92:d4:8a Access-Point-Salon
Tue Mar 16 16:21:21 2021 daemon.info dnsmasq-dhcp[26386]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
```
Le  pire semblant être 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage et peut être qu'il engrange le soucis avec les autres.
Le soucis peut être lié à un second DHCP sur le réseau, ou un AP qui déconne.

JeedHome44, 2021-03-19:
Alors pour l'AP du salon, il est tout récent et m'a permis de passer tous les appareils du salon dessus en ethernet sur un onduleur. L'AP Salon est mis sur le wifi de l'AP Garage en 5GHz. Cet AP garage est en ethernet sur le routeur 4G dans le grenier qui est lui même en ethernet sur la Freebox en ethernet direct.
Je suis certains que ces 2 AP sont bien en mode AP.
Ensuite, j'ai 2 répéteur TP Link aux 2 extrémités de la maison. Ils sont bien aussi en mode AP et non répéteur et sont branché chacun sur une CPL gigabit. La troisième CPL arrive dans le bureau et est mis en ethernet sur la Freebox également.

Mais depuis un moment, y a un truc que je ne pige pas. J'ai un AP principal dans le bureau avec mon ancien routeur TP Link Archer C9. C'est le seul que je n'arrive pas à choper sur le LAN. Même en me connectant au wifi dessus. Si je mets Tplinkap.net, ça veut rien savoir. Je me souviens l'avoir configuré en routeur avant et avoir décoché AP.
C'est peut être lui qui fou le bordel non?
Si son mode AP n'est pas correctement fait, il peut donc avoir gardé un mode DHCP caché?
J'avais bloqué son IP et il n'apparaît pas sur OMR pourtant...
Ça peut être ca la source du problème?

Ysurac, 2021-03-19:
Je peux pas dire.
Tu peux déjà mettre l'API garage en IP fixe pour voir si il y a moins de log DHCP.

JeedHome44, 2021-03-19:
Pourtant tout est en IP fixe et j'ai même mis récemment en infinite tout à la maison. Donc l'AP Garage est en fixe normalement. Ce n'est pas le cas dans les log?
J'ai mis tout en infinite en pansant que ça allait réduire les requêtes DHCP non?

JeedHome44, 2021-03-19:
J'ai mon second Jeedom qui est dans le salon et qui fait toutes les heures du Timelapse qu'il sauvegarde sur le NAS dans le bureau. Il est sur l'AP Salon en ethernet. L'AP Salon est en mode client. Ça peut venir aussi de la l'instabilité?

JeedHome44, 2021-03-19:
Et c'est quoi l'option SQM dans la configuration de chaque wan stp?

Ysurac, 2021-03-19:
Le infinite peut être le soucis, ce n'est pas toujours supporté.
L'AP garage ne semble pas être en IP fixe.
Pas de soucis pour le jeedom je pense.
SQM (Smart Queue Management) permet d'éviter une augmentation de la latence en cas de saturation des connexions.

JeedHome44, 2021-03-19:
Les IP infinite sont fait depuis une 10ene de jour seulement. Les AP ne supportent pas ca?
Comment savoir quel appareil supporte ou non les IP infini?

Dans un premier temps, je vais remettre un zéro l'ap sur lequel je n'arrive pas à aller et voir si ça règle les soucis.

C'est compliqué la gestion d'un réseau quand même. On a l'impression que tout va bien mais non...

JeedHome44, 2021-03-20:
Mais, en fait, le LAN peut faire planter le wan2 uniquement?

Car il ne faut pas planter le wan1.

Et la, je me rends compte que c'est au bout de 24 heures environ que le wan2 revient en orange sur l'erreur serveur 1sec.

Ne serait-ce pas Free qui limite maintenant à 24 heures la connexion 4G? Et donc après, OMR voit un blocage sur le wan2?

La j'ai juste encore refait un changement et je l'ai enlevé et le wan2 est de nouveau au vert.

Autre précision: depuis que j'ai ce problème, je me rends compte que je change d'ip externe 4G sur le routeur 4G. Avant je gardais toujours la même.

Ne serait-ce donc pas cet IP qui change peut être toutes les 24 heures qui mettrait donc en défaut OMR aussi?

JeedHome44, 2021-03-20:
Ça ne serait pas possible d'avoir une option qui recharge le wan2 4G s'il voit un changement d'IP externe?

Ysurac, 2021-03-20:
Normalement un changement d'IP ne provoque qu'une perte temporaire de la connexion en question.

JeedHome44, 2021-03-20:
Y a-t-il une option pour pour recharger le wan qui a changé d'ip?
Je pense que ça serait ma solution.

Quand j'ai utilisé pour la première fois OMR j'avais pas ce problème car l'IP ne changeait pas.

Je me demande si Free n'a pas changé ca en mettant la 5G sur son réseau.

Donc l'IP tourne. Je ne sais pas si Free a la possibilité de faire une IP fixe en 4G. Je ne pense pas.

Si t'as la possibilité de faire une option comme ça que je puisse tester et te faire un retour...

JeedHome44, 2021-03-20:
[image]

Ça peut être la raison du problème ca?

Faut que je trouve quel est l'appareil avec le MAC qui rentre en conflit

JeedHome44, 2021-03-20:
Le MAC 70:EE:**:**:**:D1 c'est la Netatmo Welcome de l'entrée mis sur le wifi du routeur 4G. Ça peut être la cause globale du problème?

Ysurac, 2021-03-20:
En tout cas c'est un soucis.

JeedHome44, 2021-03-21:
Premier test: j'ai remis à zéro l'access point du bureau que je n'arrivais pas à jour. Je l'ai reconfiguré pour qu'il ai l'adresse ip 192.168.10.1, donc différente de la Freebox en 0.1 et du routeur 4G en 8.1. j'ai remis les mêmes wifi et j'ai désactivé le DHCP. Je l'ai reboot et rebranché au LAN. Pourtant, quand je mets 192.168.10.1 dans le navigateur, je n'ai pas accès aux config de l'AP. C'est un tp link Archer C9.
Y a-t-il une règle particulière à créer sur OMR pour y avoir accès?

JeedHome44, 2021-03-21:
j'ai changé la caméra Netatmo Welcome de wifi et elle fait les mêmes erreurs sur mon routeur 4G...
je ne comprends plus rien...

Ysurac, 2021-03-21:
Pour accéder à l'IP 192.168.10.1 il faut soit que le client soit dans le même sous réseau, soit que le routeur ait une interface dans ce réseau.

JeedHome44, 2021-03-21:
J'ai mis 255.255.0.0 en masque sous réseau sur l'AP Archer C9 pendant que OMR allait le voir comme ça. C'est une erreur?
Il faut faire une nouvelle interface sur OMR c'est ça?
Après, ce n'est pas forcément gênant de ne pas le voir sur OMR?

JeedHome44, 2021-03-21:
Des fois, j'ai ca au lieu d'avoir Free.

Je ne pige pas ce que j'ai...

[image]

Ysurac, 2021-03-21:
Ce n'est pas un soucis. C'est Free qui a déclaré la plage d'IP sous ce nom.

JeedHome44, 2021-03-21:
Mais ce qui m'étonne le plus, c'est que je gardais toujours le même IP 4G a une époque. Et maintenant, ça change plusieurs fois par jour j'ai l'impression...

Donc la je refais la config de l'AP principal mais je n'ai pas compris ce que tu m'as dit pour pouvoir accéder au 192.168.10.1 j'ai mis le masque 255.255.0.0 c'est une erreur?
Et bizarrement, il s'affiche sur l'App Tether de TP link maintenant mais ça plante quand je clique dessus pour y accéder.

Il faut que je fasse une règle sur OMR? Mais laquelle?

Et j'ai bien revérifié, l'AP Garage c'est un TP Link AC1200 et il est bien configuré en AP et NOK en routeur. Je ne comprends donc pas pourquoi il fait autant de demandes. Il y a juste dessus plusieurs appareils Tuya, une antenne BLEA avec un Pi zéro, quelques prises connectées...

JeedHome44, 2021-03-22:
Bon, j'ai mis la Netatmo Welcome sur le wifi Freebox au lieu du wifi Huawei et j'ai l'impression que ça ne fait plus planter la wan2.

Par contre, est-ce possible de me dire comment pouvoir consulter le 192.168.10.1 par le sous réseau 255.255.0.0 svp?

Quoi faire sur OMR pour ouvrir ca?

Merci d'avance

babyss69, 2021-03-23:
Bonjour, 
Moi cela plantais énormément.
Au début je croyais que cela venait du VPS qui avait une grande latence, (J'ai essayé 3 VPS , OVH , VULTR et même hébergement perso mais en faite cela venais de la configuration de la 4 G , le modem basculait sur de l'IPV6 , et la plus rien ne marchait.

JeedHome44, 2021-03-23:
Merci du conseil.
Je vais regarder si l'IPV6 est activée sur le routeur 4G.

J'avais fait une regle sur la Freebox mais pas le routeur 4G.

La, depuis plusieurs jours, je ne constate plus de problèmes depuis que j'ai changé ma caméra Netatmo Welcome de wifi. Elle était sur le wifi du routeur 4G mais je pense qu'il ne devait pas supporter ca...

babyss69, 2021-03-23:
Je voulais te demander , tu as tester la limitation des quotas dans openmptcprouter ?
moi je n'ai pas vu d'option pour régler le quota en fonction d'une date .
L'autre jour ma connexion master avait atteint le quota maximum de data (j'avais mis une sécurité directement chez le fournisseur pour pas faire de hors forfais), le basculement sur la 2 -ème ligne dans openmptcprouter ne c'est pas effectué , cela m'avait bloqué l'ensemble de la connexion internet .
J'ai l'impression que des que l'on s'amuse avec les connexion 4 G , il y a pas mal d'anomalies , et c'est pas forcement évidnt de les détectés .

Ysurac, 2021-03-23:
Le quota se fait sur une base mensuelle, pas possible de régler de date.
Normalement le basculement se fait mais parfois lorsque le quota est atteint la ligne a un comportement étrange que le ping fonctionne encore mais plus de trafic http.
La prochaine version implémentera une détection visant à vérifier que le trafic http fonctionne également, ce qui pourra peut être résoudre ce soucis que je n'ai pas.
Sinon, comme d'habitude en cas de soucis pour qu'il soit potentiellement corrigé il faut ouvrir une issue.

JeedHome44, 2021-03-23:
Je n'ai pas de problème de cota car je suis data illimité sur la 4G

babyss69, 2021-03-23:
Merci pour l'information Ysurac

Ysurac, 2021-03-23:
@babyss69 oui, il faudra si possible tester la bêta, quand elle sera dispo, pour me dire si ça règle le soucis ou si il faudra trouver une autre solution ;)
@JeedHome44 pour la version actuelle, il est effectivement préférable de permettre uniquement l'IPv4 sur la 4G, la prochaine version supportera portentiellement les connexions IPv4 et IPv6 simultanées.

JeedHome44, 2021-03-23:
J'ai regardé et mon routeur 4G n'a pas l'air de supporter l'ipv6

JeedHome44, 2021-03-24:
Bonjour,

La faible latence TCP est elle activée par défaut ou pas quand on installe OMR pour la première fois?

J'ai un très bon débit en download et upload la Mais une latence assez importante je trouve.

Plus de problèmes 4G pour le moment

JeedHome44, 2021-03-25:
Bonjour,

Le routeur 4G s'est encore mis en défaut orange sur OMR.

Est-ce possible que ce soit un problème hardware du routeur qui fasse ca?

Y a moyen de faire une recharge de la connexion du wan 4G s'il passe en orange? Car je fais une modification et l'enlève et c'est reparti la...

Je vais voir si en faisant un reboot du routeur 4G quotidiennement la nuit, si ça ne peut pas améliorer quelque chose...

Ysurac, 2021-03-25:
Il faudrait voir ce que détecte OpenMPTCProuter dans Etat->journal au moment ou la connexion 4G lâche.

JeedHome44, 2021-03-25:
Y a-t-il moyen d'avoir une notification sur OMR quand la connexion passe à l'orange pour voir l'heure exacte et ensuite, pouvoir regarder le journal.

Y aurait moyen d'intégrer un suivi OMR sur Jeedom tiens? Ça permettrait de pouvoir faire du monitoring! 😉

Ysurac, 2021-03-25:
oui via Services->omr-tracker en cochant mail alert et en configurant Services->Email.

JeedHome44, 2021-03-31:
Le wan2 n'arrête pas de passer sur serveur 1 sec mais ne revient pas au vert. Je ne pige pas. Pas moyen de le faire revenir automatiquement?

Ysurac, 2021-03-31:
J'ai pas compris la.

JeedHome44, 2021-03-31:
Excuses moi.

Voilà comment c'est quand tout marche bien.

Dés que je vois le wan2 planter, je te montre comment c'est affiché.

[image]

JeedHome44, 2021-03-31:
La il n'y a pas les infos vu au dessus!

[image]

JeedHome44, 2021-04-01:
Et voilà exactement mon problème.

La wan2 passe dans cette position et ne revient pas automatiquement au vert. Mais si je fais une modification dans les paramètres que je valide et ensuite que j'enlève la modif, ça revient au vert.

Quand j'ai installé OMR en octobre dernier, c'était super stable et ça n'arrivait jamais.

C'est après avoir mis à jour la premier fois OMR que ça symptôme est apparu.

De plus, le wan 1 le fait des fois mais revient sur le vert ensuite automatiquement contrairement au wan 2.

Aucune automatisation n'est possible pour repasser le wan 2 au vert automatiquement sans que j'ai besoin à chaque fois de faire une intervention manuelle?

Et quand c'est comme ça, tout rame à la maison sur internet car tout ne passe que par l'ADSL.

[image]

JeedHome44, 2021-04-01:
J'oubliais:

OMR Tracker ne m'envoie pas de mail quand ca passe dans cette position.

Je ne sais donc pas quand est-ce que ça se passe exactement

JeedHome44, 2021-04-01:
Et voilà ce qui apparaît après quand je demande d'exécuter des actions avec les Google Home quand le wan 2 est sur Orange.

[image]

N'est-ce pas le VPN Shadowsocks qui est en cause et qui fait planter mon wan2?

Est-ce possible de désactiver le VPN entre OMR et le VPS?

Ysurac, 2021-04-01:
Le soucis doit être un WAN2 qui n'est pas complètement down mais ne répond pas normalement. Ce n'est donc pas détecté par "omr-tracker" qui vérifie l'état et envoi les mails.
Shadowsocks est un proxy, il sert pour tout ce qui est TCP. Le VPN sert pour tout ce qui n'est pas TCP. Il est fortement recommandé de ne rien désactiver.
WAN2 est un routeur ethernet ? une clef USB ? autre chose ?

JeedHome44, 2021-04-01:
Mon wan2 est un routeur Huawei 4G B525s-23a.

Tu penses qu'il peut être défectueux?

Il faudrait que j'en commande un autre et que je teste de le mettre à la place et voir le résultat?

JeedHome44, 2021-04-01:
Mais pourquoi ça reste à l'orange et si je vais sur la page de config OMR et que je clique de sauvegarder, tout revient normalement. Ce n'est pas possible de faire ça sur un wan automatique s'il passe sur l'erreur serveur après 1 sec? Le wan 1 revient bien tout seul au vert mais pas le 2.

JeedHome44, 2021-04-02:
Je suis en train de me demander si ce n'est pas ca la cause du problème.

Avant octobre 2020, j'avais le même débit en download mais 20-30 en upload et 40-50ms en latence.

Depuis octobre, je monte très haut en upload mais latence pourrie.

C'est donc ça qui met en orange le wan2 non?

Ça se règle sur le routeur 4G ou c'est totalement dépendant de l'antenne?

Je parie que c'est leur histoire de 5G Free qui fou le bordel.

Une idée?

Ysurac, 2021-04-02:
Le soucis est effectivement la latence (pouvant dépasser la seconde) et surtout les paquets perdus (les croix rouges).

JeedHome44, 2021-04-02:
J'ai tenté de mettre le routeur 4G uniquement sur la bande 700 ou 1800 MHz mais ça ne descend pas sous 200ms

C'est donc l'antenne Free qui privilégie l'upload à la latence?

Ysurac, 2021-04-02:
Ca semble étonnant.
Le test de débit a été fait directement avec le Modem sans passer par OpenMPTCProuter ?

JeedHome44, 2021-04-02:
Non
Je me mets sur le wifi de la maison et j'obtiens ca...
Quand je suis arrivé sur OMR, j'arrivais même à mettre le wan 4G en maître et c'était ultra stable.
C'est lors de la première mise à jour OMR que c'est parti en vrille et depuis, le wan2 est super instable.

Je fais comment pour tester le débit uniquement du routeur 4G vu qu'il est sur le LAN OMR?

[92 more replies]

closed 2021-04-20 by JeedHome44