Ysurac/openmptcprouter #1675: Position des routeurs dans le réseau + routeur supplémentaire suivant openMPTCProuter

94374n, 2021-03-16:
Bonjour ! Je viens ici avec quelques schémas pour savoir si j'ai bien compris le fonctionnement du système au niveau de la position des routeurs. Je vais emménager dans un appartement qui n'a que l'ADSL et j'aimerais utiliser openmptcprouter pour y ajouter une connexion 4G sans perdre tout ce qui concerne l'ip et la redirection des ports. N'ayant pas encore emménagé ni reçu le nouveau routeur du FAI ainsi que le routeur 4G, je ne peux que planifier sans tester par moi-même...

Tout d'abord j'aimerais être sûr de bien avoir compris. Est-ce que cette configuration de routeurs est correcte ?

[image]

Une fois mis en place, avec quel routeur dois-je gérer la redirection de ports ainsi que les ip assignées manuellement au DPCH si besoin ?
Aussi est-ce que la vitesse de connexion du VPS va brider les connexions locales entre les machines ?

J'ai pour habitude de passer mon routeur FAI en DMZ et de mettre derrière un autre routeur (Asus dans mon cas) pour profiter d'un wifi plus efficace et de configurations plus poussées (redirection de ports et définir certaines IP manuellement principalement), comme ceci :

[image]

Est-ce que je peux faire pareil derrière l'installation comme ceci pour avoir une configuration similaire à celle que j'ai habitude d'avoir ? Est-ce possible de mettre tout le reste en DMZ et de connecter tous mes appareils, gérer les ports et le DHCP sur mon routeur Asus ?

[image]

Merci d'avance pour vos réponses 🙏

Ysurac, 2021-03-17:
Si la Box FAI a un switch, oui le routeur avec OpenMPTCProuter peut être banché dessus.  Le DHCP peut être désactivé dessus donc pas de soucis pour mettre le routeur Asus, mais pour qu'OMR-ByPass fonctionne (permet de sortir directement vers la box FAI ou la 4G pour certains sites) il faut que l'Asus utilise OpenMPTCProuter en DNS.
Les redirections de ports du VPS vers OpenMPTCProuter se font depuis OpenMPTCProuter qui peut gérer aussi le DHCP et les baux statiques si besoin.
Le VPS n'intervient pas dans les connexions locales qui restent locale.
OpenMPTCProuter peut pas vraiment être mis en DMZ ou bridge et c'est pas recommandé afin de limiter le trafic entrant entre le VPS et le routeur.

94374n, 2021-03-17:
Comment utiliser OpenMPTCProuter en DNS ? Juste inscrire 192.168.100.1 dans le champ DNS du routeur Asus ?
Pas moyen donc de gérer la redirection de ports et les baux statiques via le routeur Asus ? Obligation de me connecter à OpenMPTCProuter via 192.168.100.1 ?

Aussi si je laisse tomber l'idée de rajouter mon routeur Asus, est-il possible d'utiliser le routeur 4G tel qu'il est placé dans mon schéma (car il est bien plus puissant que ma box FAI) en tant que point d'accès sachant que mon but est de pouvoir faire du port forwading malgré mon réseau 4G ?

Ysurac, 2021-03-17:
Oui juste en mettant 192.168.100.1 dans le DNS du routeur.
Les baux statiques pas de soucis que ça soit fait depuis l'Asus.
Pour la redirection de ports, comme ils sont ouvert selon le besoin sur le VPS oui ça doit être faire depuis OpenMPTCProuter. Il est possible de rediriger tous les ports mais ce n'est pas une bonne idée.

Pas de soucis d'utiliser un routeur 4G en point d'accès, c'est ce que je fais pour mon wifi.

94374n, 2021-03-17:
Merci beaucoup pour ces réponses, tout est très clair !

closed 2021-03-17 by 94374n