Ysurac/openmptcprouter #1676: Problem creating VLANs

Rookie805, 2021-03-16:
## Expected Behavior
Trying to create VLAN interface as a WAN on Raspberry Pi4b. In Network -> Interfaces -> Add new interface:
name: vlan3
Protocol: DHCP client
Interface type: Normal
Interface: custom = eth0.3 -> Create interface

Expected to get IP address from DHCP server in VLAN 3 and setting this as one of the WAN interfaces.

## Current Behavior
In Network -> Interfaces -> Add new interface:
name: vlan3
Protocol: DHCP client
Interface type: Normal
Interface: custom = eth0.3 -> Create interface

After interface is created - it doesn't get IP address from DHCP server in VLAN3. Traffic counter on this interface is counting only TX packets and no RX (=0). Another device with configured DHCP client on VLAN3 connected to this switch immediately gets IP address from DHCP server in VLAN3.

Tue Mar 16 20:14:44 2021 daemon.notice netifd: Interface 'vlan3' is enabled
Tue Mar 16 20:14:44 2021 daemon.notice netifd: VLAN 'eth0.3' link is up
Tue Mar 16 20:14:44 2021 daemon.notice netifd: Interface 'vlan3' has link connectivity
Tue Mar 16 20:14:44 2021 daemon.notice netifd: Interface 'vlan3' is setting up now
Tue Mar 16 20:14:44 2021 daemon.notice netifd: vlan3 (6190): udhcpc: started, v1.31.1
Tue Mar 16 20:14:44 2021 user.notice ModemManager: hotplug: add network interface eth0.3: event processed
Tue Mar 16 20:14:44 2021 user.notice ModemManager: hotplug: event reported: action=add, name=eth0.3, subsystem=net
Tue Mar 16 20:14:44 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Mar 16 20:14:44 2021 daemon.notice netifd: vlan3 (6190): udhcpc: sending discover
Tue Mar 16 20:14:45 2021 daemon.info vnstatd[19674]: Info: Interface "eth0.3" enabled.
Tue Mar 16 20:14:47 2021 daemon.notice netifd: vlan3 (6190): udhcpc: sending discover
Tue Mar 16 20:14:47 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Tue Mar 16 20:14:50 2021 daemon.notice netifd: vlan3 (6190): udhcpc: sending discover
<END OF LOG>

Strings about ModemManager confuses me, as I didn't selected it as an interface type.

I also tried to create an interface using CLI:
ip link add link eth0 name eth0.3 type vlan id 3
ip link set dev eth0.3 up

with the same result.

## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: 0.1024
  - OpenMPTCProuter platform: RPI4

I'm not an expert in linux at all, so detailed instructions would be much appreciated. I would gladly provide more info if required (any cli output or logs). Thank you in advance.

Ysurac, 2021-03-17:
All defined interfaces use VLAN or some use MacVLAN ? In the last case this can be a conflict.
For ModemManager, it's not related as it's only used with ModemManager set as protocol.

Rookie805, 2021-03-21:
Problem solved. It was not OMR's fault. 

Just for history: WAN interfaces was connected to LTE modems in bridge mode. And WAN's IP was locking to the MAC address of device which got lease first (and that was not OMR, but my router, because I switched back and forth while checking). After that no other device got offered IP by DHCP, even if my router released it.

Solution was to re-enable modems LTE interfaces with  OMR as only DHCP client.

Thank you for your time and the whole project!

closed 2021-03-21 by Rookie805