Ysurac/openmptcprouter #1677: OpenMpTCProuter on a corporate network

GrosRems, 2021-03-17:
Hello

I am not sure I really understand how OpenMPTcpRouter works, so I have a question.

On a corporate network with several switches (Cisco SG250 type for example), do the routers need a specific configuration?

Is it necessary to give them an IP on the same network as the RPI ?

Thank you

Kalimeiro, 2021-03-18:
Hello,

Integrating OpenMPTCProuter in a corporate network is possible, my experience is positive on this subject.

On the other hand, in view of your questions on the subject, I strongly suggest to contact an IT systems and networks company if you plan to use OMR.

Best regards,

GrosRems, 2021-03-18:
Hello

I understand your answer, it's sure that my question is a little stupid.
The problem is that in France companies do not really want to install this type of solution.
And to tell the truth, I still have a license in computer science, I am supposed to know the network.

I'll try to ask a smarter question.

On a network without vlan (all ports untaged), the switches should transmit the frames of the different subnetworks without problem. OMR acts as a router and therefore allows communication between subnetworks. If my understanding is correct as long as a machine is using OMR as a gateway with manual IP or OMR DHCP, link aggregation should work.

Am I forgetting something for this to work?

Thank you

Kalimeiro, 2021-03-18:
Bonjour :-)

Je connais bien les problèmes et les mentalités en France ... ! :-) la question n'est pas stupide mais comme je ne sais pas à qui j'ai à faire, installer OMR dans une entreprise et bénéfique mais pas sans risque, aucun problème pour moi d'installer OMR dans les entreprises ... (je suis à mon compte), la satisfaction est certaine, mais il faut comprendre que OMR est encore en beta, car il est en développement.

à partir du moment ou OMR est défini en tant que passerelle/dns par défaut à la place de ton ancien routeur dans les paramètres de ton système d'exploitation alors oui l'ordinateur bénéficiera de l'agrégation MPTCP. le DHCP de OMR fera la même chose que précédemment mais automatiquement à condition de ne pas déjà avoir un serveur DHCP sur le réseau sinon c'est la pagaille assuré.

si tu as plusieurs sous réseaux, et que tu as déjà un DHCP, il faut que ton OMR ai plusieurs IP dans chacun des sous réseaux, le mieux étant de partir sur du VLAN qui fonctionne très bien avec OMR et de laisser OMR gérer le DHCP.

Je peux t'aider si tu le souhaite, j'ai déjà réalisé ce genre de configuration et ça fonctionne très bien, y compris avec de l'IPv6, accessoirement faire du VPN SITE-TO-SITE avec OMR est aussi possible je l'ai déjà réalisé pour un client et ça fonctionne drôlement bien ... !

GrosRems, 2021-03-19:
Hi

We will be yelled at if we discuss here in French ...
I asked my question to know if there were any particular points of vigilance, in the end I started and I did not actually encounter any particular problem.
I take note of what you say about VLANs, I have not tried it so far but I will do some tests because indeed it sounds interesting.

Thank you anyway

Salut

On va se faire engueuler si on discute ici en Français...
Je posais ma question pour savoir s'il y a avait des points de vigilance particulier, au final je me suis lancé et je n'ai effectivement pas spécialement rencontré de problème.
Je prend note de ce que tu dis sur les VLAN, je ne l'ai pas tenté jusqu'ici mais je ferais des tests parce qu'effectivement ca parait intéressant.

Merci en tout cas

closed 2021-06-22 by github-actions[bot]