Ysurac/openmptcprouter #1680: Bonding not working anymore

robertoglxda, 2021-03-20:
## Expected Behavior
I have two LTE connections that I want to bond. This was working until some weeks ago.

## Current Behavior
When the 'master' connection is down, I get no internet connectivity. The second connection is configured as 'backup' on OMR.

## Possible Solution
No solution yet.

## Steps to Reproduce the Problem
I'm not sure of how to reproduce it.

## Context (Environment)
On the server, I get this log every minute:
`Mar 20 17:19:04 VPN kernel: omr-bonding: option mode: unable to set because the bond device is up
Mar 20 17:19:04 VPN omr-service[998]: /usr/local/bin/omr-service: line 82: echo: write error: Device or resource busy
Mar 20 17:19:04 VPN omr-service[7607]: Error: Device can not be enslaved while up.
Mar 20 17:19:04 VPN kernel: omr-bonding: (slave ovpnbonding1): slave is up - this may be due to an out of date ifenslave
Mar 20 17:19:04 VPN omr-service[7609]: Error: Device can not be enslaved while up.
Mar 20 17:19:04 VPN kernel: omr-bonding: (slave ovpnbonding2): slave is up - this may be due to an out of date ifenslave
Mar 20 17:19:04 VPN omr-service[7611]: Error: Device can not be enslaved while up.
Mar 20 17:19:04 VPN kernel: omr-bonding: (slave ovpnbonding3): slave is up - this may be due to an out of date ifenslave
Mar 20 17:19:04 VPN omr-service[7613]: Error: Device can not be enslaved while up.
Mar 20 17:19:04 VPN kernel: omr-bonding: (slave ovpnbonding4): slave is up - this may be due to an out of date ifenslave
Mar 20 17:19:04 VPN omr-service[7615]: Error: Device can not be enslaved while up.
Mar 20 17:19:04 VPN kernel: omr-bonding: (slave ovpnbonding5): slave is up - this may be due to an out of date ifenslave
Mar 20 17:19:04 VPN omr-service[7617]: Error: Device can not be enslaved while up.
Mar 20 17:19:04 VPN kernel: omr-bonding: (slave ovpnbonding6): slave is up - this may be due to an out of date ifenslave
Mar 20 17:19:04 VPN omr-service[7619]: Error: Device can not be enslaved while up.
Mar 20 17:19:04 VPN kernel: omr-bonding: (slave ovpnbonding7): slave is up - this may be due to an out of date ifenslave
Mar 20 17:19:04 VPN omr-service[7621]: Error: Device can not be enslaved while up.
Mar 20 17:19:04 VPN kernel: omr-bonding: (slave ovpnbonding8): slave is up - this may be due to an out of date ifenslave
Mar 20 17:19:04 VPN omr-service[7624]: RTNETLINK answers: File exists
Mar 20 17:19:04 VPN omr-service[7625]: RTNETLINK answers: File exists
Mar 20 17:19:04 VPN omr-service[7626]: RTNETLINK answers: File exists`

## Specifications

  - OpenMPTCProuter version: 0.57.1
  - OpenMPTCProuter VPS version: 0.1023 5.4.81-mptcp
  - OpenMPTCProuter platform: RPI4 4GB RAM

Ysurac, 2021-03-20:
I don't support old release. There is new release to fix previous releases ;)

robertoglxda, 2021-03-20:
You are right. I just updated OpenMPTCProuter to version  0.57.3.
I also did run the update/install script on the VPS following the simple command here:
https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS
However, I still get the report 'version 0.1023 5.4.81-mptcp (Latest available version 0.1024)' from the openmptcprouter->status page. I also rebooted the VPS and the RPI4.

The issue described remains.

closed 2021-06-24 by github-actions[bot]