Ysurac/openmptcprouter #1683: MPTCP router speed issue

netajeepatil, 2021-03-22:
Hi, 
I am trying to aggregate 3 SIM cards with MPTCP router ( Version：v0.57.3 ), with VPS on ubuntu server 20.04 LTS 
I am getting a speed maximum of 29 mbps.  ( I have 3 Sim cards, 2 from vodafone & 1 Jio). Per sim speed if I do speedtest, I am getting 25mbps, each  on 2 vodafone sim & 10 mbps ( on Jio sim).  

After aggregation, my speed is not exceeding 29mbps. Please assist, if there is any specific settings, which I have missed or needs to be checked, which I may have overlooked. I am using glorytun TCP
Please check the screens below fyr.

[image]
[image]
[image]
[image]
[image]

My intention is to aggregate have 7 SIM cards .

Ysurac, 2021-03-22:
Latency is quite high.
Speed can be limited by antenna, so if all providers use the same you may not get a better speed.

netajeepatil, 2021-03-22:
Thanks for response. I am using two Vodafone 4G LTE mobile network and one Jio 4G LTE mobile network. 
Do you mean that Instead of all LTE network I have to choose different ISP (DSL / Cable Network). Please help me out.

Ysurac, 2021-03-22:
I mean that Vodafone 4G LTE can be limited by Vodafone 4G antenna capacity, so if you run speedtest on the 2 SIM at same time you may get only 25/2 on each. I think Jio have it's own antenna.
You can check in Network->MPTCP, bandwith tab if all cards are used or if a card is not used at all.

netajeepatil, 2021-03-23:
I have connected Vodafone 4G SIm on WAN1 interface ( capping of 16mbps), 100mbps lease line on WAN2 interface & 30 Mbps Dedicated ILL connected on WAN3 interface as a output I am getting Max 70 Mbps Download & 55-59 mbps upload speed. 
Please check attached snap for reference.
[image]

Please correct me if anything wrong,  as per me if 15+100+30=145mbps input for all wan interface , after aggregation output needs to get  approx. 100mbps but result getting max 70mbps.
Please suggest any changes needs to be done in config  to get much better speed.

Adorfer, 2021-03-23:
i suspect that there is something like:
1) Master should always the connection with lowes latency 
2) aggregated bandwith will probably not exceed "master*4 in speedtest"

Ysurac, 2021-03-23:
Master only need to be on the connection more stable, even if it's not the lowest latency connection.
Aggregated bandwidth is not related at all to master connection. Master connection is only used for initial connection.
For this case, I already said to check bandwidth graph to check if all WANs are used.

netajeepatil, 2021-03-24:
Please see attached bandwidth snaps where i used two wans with ILL  lines only (WAN1 - 100 MBPS & WAN2 30 MBPS DEDICATED ILL) -  please suggest .

[image]
[image]
[image]
[image]

MPTCP Support Check for WAN1 & WAN2
MPTCP Support Check – WAN1
1: 192.168.10.2 
2: 192.168.42.129 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
3: 172.18.3.1 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
4: 45.64.192.113 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
5: 45.64.193.85  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
6: 103.205.124.81  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
7: 27.109.1.149  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
8: 72.14.204.217 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
9: 10.23.193.126 TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
10: 216.239.54.84 TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
11: 108.170.248.162  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
12: 108.170.229.13 TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize [Extra headers: ICMPExtension ICMPExtensionObject ICMPExtensionMPLS ] 
13: 209.85.242.12  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
14: 209.85.252.101 TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
15: * 
16: * 
17: * 
18: * 
19: * 
20: * 
21: * 
22: * 
23: * 
24: 35.247.182.85 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 50:65:f3:51:19:c3 , SourceMAC = 60:32:b1:d2:30:e0 , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 57 , Protocol = 0x6 , CheckSum = 0x9cca , SourceIP = 35.247.182.85 , DestinationIP = 192.168.10.1 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 37461 , SeqNumber = 442393981 , AckNumber = 742657278 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 43400 , CheckSum = 0x21c3 , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1394 , >
< TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 14015209356131660191 , >

MPTCP Support Check – WAN2
1: 45.64.192.97 
2: 45.64.193.85  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
3: 103.205.124.81  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
4: 27.109.1.149  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
5: 72.14.204.217 IP::TTL IP::CheckSum 
6: 10.252.182.158 IP::DiffServicesCP IP::TTL IP::CheckSum 
7: 142.250.214.98 IP::DiffServicesCP IP::TTL IP::CheckSum 
8: 108.170.248.194 IP::DiffServicesCP IP::TTL IP::CheckSum 
9: 108.170.225.145 IP::DiffServicesCP IP::TTL IP::CheckSum [Extra headers: ICMPExtension ICMPExtensionObject ICMPExtensionMPLS ] 
10: 72.14.236.223 IP::DiffServicesCP IP::TTL IP::CheckSum 
11: 66.249.94.149 IP::DiffServicesCP IP::TTL IP::CheckSum 
12: * 
13: * 
14: * 
15: * 
16: * 
17: * 
18: * 
19: * 
20: * 
21: 35.247.182.85 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 48:ee:0c:bb:35:7a , SourceMAC = 4c:5e:0c:7f:7e:e7 , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 60 , Protocol = 0x6 , CheckSum = 0x76d1 , SourceIP = 35.247.182.85 , DestinationIP = 45.64.192.98 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 50679 , SeqNumber = 1031026232 , AckNumber = 2135378151 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 43400 , CheckSum = 0x8e8c , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1420 , >
< TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 14630316630742674456 , >

Speed test results-
[image]
[image]

Please suggest if anything missed by me, how can i improve.

**One more query I want to ask can we bring VPS server in mptcp router network  (LAN network) to avoid cloud dependency and improve speed.**

Ysurac, 2021-03-24:
WAN1 and WAN2 support Multipath TCP. What is VPS max internet network speed ?
I would need via SSH or Services->Terminal, the result of ```omr-test-speed wan1```, ```omr-test-speed wan2```,... and ```omr-test-speed```. Stop after at least 2 minutes with Ctrl+C. This check speed on interfaces wan1,wan2,... and aggregated speed by downloading a big file.
VPS server must be in a network that have a direct internet connection with at least the max aggregated speed.

netajeepatil, 2021-03-24:
My VPS server  is in google cloud

Services-terminal snap – unable to run any cmd
[image]

Connected to router by putty-ssh and below are the results. 

PACKAGE:     openmptcprouter
 VERSION:     v0.57.3
 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Tue Jan  5 07:02:46 UTC 2021
-----------------------------------------------------

root@OpenMPTCProuter:~# omr-test-speed wan1
Select best test server...
ping: SO_BINDTODEVICE wan1: No such device
host: speedtest.frankfurt.linode.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: speedtest.tokyo2.linode.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: speedtest.singapore.linode.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: speedtest.newark.linode.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: speedtest.atlanta.linode.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: speedtest.dallas.linode.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: speedtest.fremont.linode.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: speedtest.tele2.net - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: www.ovh.net - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: speed.hetzner.de - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: ipv4.bouygues.testdebit.info - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: par.download.datapacket.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: nyc.download.datapacket.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: ams.download.datapacket.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: fra.download.datapacket.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: lon.download.datapacket.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: mad.download.datapacket.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: prg.download.datapacket.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: sto.download.datapacket.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: vie.download.datapacket.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: war.download.datapacket.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: atl.download.datapacket.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: chi.download.datapacket.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: lax.download.datapacket.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: mia.download.datapacket.com - ping:
ping: SO_BINDTODEVICE wan1: No such device
host: nyc.download.datapacket.com - ping:
Best server is http://proof.ovh.net/files/10Gio.dat, running test:
curl: (45) Couldn't bind to 'wan1'

[... 65 lines cut ...]
host: speedtest.fremont.linode.com - ping: 233
host: speedtest.tele2.net - ping: 309
host: www.ovh.net - ping:
host: speed.hetzner.de - ping: 320
host: ipv4.bouygues.testdebit.info - ping: 313
host: par.download.datapacket.com - ping: 331
host: nyc.download.datapacket.com - ping: 280
host: ams.download.datapacket.com - ping: 890
host: fra.download.datapacket.com - ping: 322
host: lon.download.datapacket.com - ping: 333
host: mad.download.datapacket.com - ping: 325
host: prg.download.datapacket.com - ping: 315
host: sto.download.datapacket.com - ping: 346
host: vie.download.datapacket.com - ping: 322
host: war.download.datapacket.com - ping: 345
host: atl.download.datapacket.com - ping: 317
host: chi.download.datapacket.com - ping: 272
host: lax.download.datapacket.com - ping: 246
host: mia.download.datapacket.com - ping: 287
host: nyc.download.datapacket.com - ping: 280
Best server is http://speedtest.singapore.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--   100 86160    0 86160    0     0   228k      0 --:--:-- --:--:-- --:--:--  2100  403k    0  403k    0     0   260k      0 --:--:--  0:00:01 --:--:--  2100 2360k    0 2360k    0     0  1015k      0 --:--:--  0:00:02 --:--:-- 10100 8241k    0 8241k    0     0  2480k      0 --:--:--  0:00:03 --:--:-- 24100 14.5M    0 14.5M    0     0  3443k      0 --:--:--  0:00:04 --:--:-- 34100 20.5M    0 20.5M    0     0  3952k      0 --:--:--  0:00:05 --:--:-- 42100 27.0M    0 27.0M    0     0  4380k      0 --:--:--  0:00:06 --:--:-- 57100 33.7M    0 33.7M    0     0  4717k      0 --:--:--  0:00:07 --:--:-- 64100 40.9M    0 40.9M    0     0  5038k      0 --:--:--  0:00:08 --:--:-- 67100 48.3M    0 48.3M    0     0  5310k      0 --:--:--  0:00:09 --:--:-- 69100 56.0M    0 56.0M    0     0  5560k      0 --:--:--  0:00:10 --:--:-- 72100 62.0M    0 62.0M    0     0  5601k      0 --:--:--  0:00:11 --:--:-- 71100 68.7M    0 68.7M    0     0  5711k      0 --:--:--  0:00:12 --:--:-- 71100 75.7M    0 75.7M    0     0  5819k      0 --:--:--  0:00:13 --:--:-- 71100 81.1M    0 81.1M    0     0  5795k      0 --:--:--  0:00:14 --:--:-- 66100 86.1M    0 86.1M    0     0  5757k      0 --:--:--  0:00:15 --:--:-- 61100 91.4M    0 91.4M    0     0  5737k      0 --:--:--  0:00:16 --:--:-- 60100 96.6M    0 96.6M    0     0  5710k      0 --:--:--  0:00:17 --:--:-- 57100  101M    0  101M    0     0  5687k      0 --:--:--  0:00:18 --:--:-- 53100  107M    0  107M    0     0  5687k      0 --:--:--  0:00:19 --:--:-- 53100  112M    0  112M    0     0  5678k      0 --:--: [... cut ...] --  0:03:00 --:--:-- 38100 1002M    0 1002M    0     0  5661k      0 --:--:--  0:03:01 --:--:-- 42100 1007M    0 1007M    0     0  5659k      0 --:--:--  0:03:02 --:--:-- 45100 1013M    0 1013M    0     0  5662k      0 --:--:--  0:03:03 --:--:-- 50100 1019M    0 1019M    0     0  5664k      0 --:--:--  0:03:04 --:--:-- 54100 1022M    0 1022M    0     0  5651k      0 --:--:--  0:03:05 --:--:-- 52100 1024M    0 1024M    0     0  5646k      0 --:--:--  0:03:05 --:--:-- 5034k
root@OpenMPTCProuter:~#

hairydogltd, 2021-03-24:
> I would need via SSH or Services->Terminal, the result of omr-test-speed wan1, omr-test-speed wan2,..

Does that mean that these first commands are now working? 
Where can I get the updated code to use them? 
Is it safe to use code from the Develop branch?

Ysurac, 2021-03-24:
@hairydogltd commands are working when download website have only one IP. Fixed release for all case is here https://github.com/Ysurac/openmptcprouter-feeds/raw/develop/openmptcprouter/files/bin/omr-test-speed and can be downloaded on any release.

@netajeepatil not easy to read... For wan1, wan2 this need to be real physical interface, you can check name of interface using for example ```ip a``` via SSH.

hairydogltd, 2021-03-24:
Thanks. That sorted it! :)

netajeepatil, 2021-03-25:
Please check below result for ip a  - 

root@OpenMPTCProuter:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group                                                                                         default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: ip6tnl0@NONE: <NOARP,80000> mtu 1452 qdisc noop state DOWN group default qlen                                                                                         1000
    link/tunnel6 :: brd ::
3: sit0@NONE: <NOARP,80000> mtu 1480 qdisc noop state DOWN group default qlen 10                                                                                        00
    link/sit 0.0.0.0 brd 0.0.0.0
4: gre0@NONE: <NOARP,80000> mtu 1452 qdisc noop state DOWN group default qlen 10                                                                                        00
    link/gre 0.0.0.0 brd 0.0.0.0
5: gretap0@NONE: <BROADCAST,MULTICAST,80000> mtu 1462 qdisc noop state DOWN grou                                                                                        p default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
6: erspan0@NONE: <BROADCAST,MULTICAST,80000> mtu 1450 qdisc noop state DOWN grou                                                                                        p default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
7: ip6gre0@NONE: <NOARP,80000> mtu 1448 qdisc noop state DOWN group default qlen                                                                                         1000
    link/gre6 :: brd ::
8: bond0: <BROADCAST,MULTICAST,MASTER,80000> mtu 1500 qdisc noop state DOWN grou                                                                                        p default qlen 1000
    link/ether d2:f6:29:8d:3e:ed brd ff:ff:ff:ff:ff:ff
9: teql0: <NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void
10: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP,80000> mtu 1500 qdisc fq_codel state                                                                                         UP group default qlen 1000
    link/ether 00:e0:4c:12:88:57 brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global eth0
       valid_lft forever preferred_lft forever
11: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP gro                                                                                        up default qlen 100
    link/ether 50:65:f3:51:19:c3 brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.1/24 brd 192.168.10.255 scope global eth1
       valid_lft forever preferred_lft forever
12: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOW                                                                                        N group default qlen 100
    link/ether 48:ee:0c:bb:35:7a brd ff:ff:ff:ff:ff:ff
    inet 45.64.192.98/29 brd 45.64.192.103 scope global eth2
       valid_lft forever preferred_lft forever
13: eth3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP gro                                                                                        up default qlen 100
    link/ether 00:50:b6:48:00:35 brd ff:ff:ff:ff:ff:ff
    inet 192.168.12.1/24 brd 192.168.12.255 scope global eth3
       valid_lft forever preferred_lft forever
20: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq_code                                                                                        l state UNKNOWN group default qlen 100
    link/none
root@OpenMPTCProuter:~#

netajeepatil, 2021-03-26:
plz suggest if any other inputs needs from my end.

Ysurac, 2021-03-26:
Same request with ```omr-test-speed``` but as I said, use real interfaces so this seems to be eth1, eth2 and eth3

closed 2021-06-29 by github-actions[bot]