Ysurac/openmptcprouter #1684: Email notifications when WAN is offline

hrmtc, 2021-03-22:
Hi.

How can I configure email notifications when some of my configured WAN interfaces become offline?
What software package would help me to achive this?

Thanks!

Ysurac, 2021-03-22:
You can check "Mail alert" in Services->omr-tracker on defaults section or by adding wan interfaces you want to be notified.
Mail configuration is in Services->E-Mail

Kalimeiro, 2021-03-24:
And be sure to set in Settings ->DNS recursive -> DHCP -> WAN DNS to "Use upstream"
Then, set in interface WAN1 DNS of you'r ISP.

Ysurac, 2021-03-24:
Should not be needed. By default root servers DNS are used and they are available when VPS is online or not and from every WANs.
The only (rare) case where this can be a problem is when external DNS is blocked by ISP.

hrmtc, 2021-03-25:
> You can check "Mail alert" in Services->omr-tracker on defaults section or by adding wan interfaces you want to be notified.
> Mail configuration is in Services->E-Mail

Thanks! This seems to be exactly what I was looking for.

closed 2021-03-25 by hrmtc