Ysurac/openmptcprouter #1686: Huawei E8372 detected as storage

xemles, 2021-03-23:
## Expected Behavior

Plug my Huawei E8372, should be detected as router.

## Current Behavior

Plugging my Huawei E8372, detected as USB Storage:

`Tue Mar 23 14:17:56 2021 kern.info kernel: [  855.891241] usb 2-1: new high-speed USB device number 3 using xhci-hcd
Tue Mar 23 14:17:56 2021 kern.debug kernel: [  856.072190] usb 2-1: no of_node; not parsing pinctrl DT
Tue Mar 23 14:17:56 2021 kern.debug kernel: [  856.078373] usb-storage 2-1:1.0: no of_node; not parsing pinctrl DT
Tue Mar 23 14:17:56 2021 kern.info kernel: [  856.084766] usb-storage 2-1:1.0: USB Mass Storage device detected
Tue Mar 23 14:17:56 2021 kern.info kernel: [  856.091198] scsi host2: usb-storage 2-1:1.0
Tue Mar 23 14:17:57 2021 kern.notice kernel: [  857.112038] scsi 2:0:0:0: CD-ROM            HUAWEI   Mass Storage     2.31 PQ: 0 ANSI: 2
Tue Mar 23 14:17:57 2021 kern.debug kernel: [  857.120588] sd 2:0:0:0: no of_node; not parsing pinctrl DT
Tue Mar 23 14:17:57 2021 kern.notice kernel: [  857.126342] scsi 2:0:0:1: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
Tue Mar 23 14:17:57 2021 kern.debug kernel: [  857.134615] sd 2:0:0:1: no of_node; not parsing pinctrl DT
Tue Mar 23 14:17:57 2021 kern.warn kernel: [  857.140803] sd 2:0:0:1: Power-on or device reset occurred
Tue Mar 23 14:17:57 2021 kern.notice kernel: [  857.157279] sd 2:0:0:1: [sda] Attached SCSI removable disk`

I checked #1519 which seemed to have the same problem, but since I'm pretty new to OpenMPTCPRouter, not sure I wanna rollback to 0.54 just to make sure it works.

lsusb gives me: Bus 002 Device 003: ID 12d1:1f01

And on `ifconfig` I can confirm it doesnt count it as an interface.

I tried `opkg install usb-modeswitch` but it tells me the package is incorrect and as mentioned in #1519 :

> usb-modeswitch package is installed on v0.57 by default.

But I don't see much showing me it is...
So I don't really know where I'm going with this one :')

## Specifications

  - OpenMPTCProuter version: v0.57
  - OpenMPTCProuter VPS version: v0.1024
  - OpenMPTCProuter platform: Linksys WRT3200ACM

Ysurac, 2021-03-23:
Disk space on WRT3200ACM is limited, so usb-modeswitch is not installed by default. You can install it via web interface after refreshing package list.

xemles, 2021-03-23:
I refreshed the package list but it doesn't seem to show, which is the issue

Ysurac, 2021-03-23:
The problem is "has no valid architecture, ignoring" error.
Can you give me the URLs in ```/etc/opkg/customfeeds.conf``` ?

xemles, 2021-03-23:
`src/gz openwrt_luci http://packages.openmptcprouter.com/v0.57.3/arm_cortex-a9_vfpv3/luci
src/gz openwrt_packages http://packages.openmptcprouter.com/v0.57.3/arm_cortex-a9_vfpv3/packages
src/gz openwrt_base http://packages.openmptcprouter.com/v0.57.3/arm_cortex-a9_vfpv3/base
src/gz openwrt_routing http://packages.openmptcprouter.com/v0.57.3/arm_cortex-a9_vfpv3/routing
src/gz openwrt_telephony http://packages.openmptcprouter.com/v0.57.3/arm_cortex-a9_vfpv3/telephony
`

I tried downloading the package manually from [](https://downloads.openwrt.org/releases/packages-18.06/arm_cortex-a9_vfpv3/base/usb-modeswitch_2017-12-19-f40f84c2-1_arm_cortex-a9_vfpv3.ipk) but still no success, first gave me:
`Collected errors:
 * wfopen: https://downloads.openwrt.org/releases/packages-18.06/arm_cortex-a9_vfpv3/base/usb-modeswitch_2017-12-19-f40f84c2-1_arm_cortex-a9_vfpv3.ipk: No such file or directory.
 * pkg_init_from_file: Failed to extract control file from https://downloads.openwrt.org/releases/packages-18.06/arm_cortex-a9_vfpv3/base/usb-modeswitch_2017-12-19-f40f84c2-1_arm_cortex-a9_vfpv3.ipk.`
Then since it said No such file or directory on the router, I downloaded the file manually and uploaded it to [transfer.sh](http://transfer.sh/13cdSt/usb-modeswitch_2017-12-19-f40f84c2-1_arm_cortex-a9_vfpv3.ipk)

And then I downloaded that transfer.sh file which gave me lots of packages with `no valid architecture` and:

`Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for usb-modeswitch found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package usb-modeswitch.`

Ysurac, 2021-03-23:
I think wrong arch is used.
Try to set url to "http://packages.openmptcprouter.com/v0.57.3/arm_cortex-a9_vfpv3-d16/luci" (same for packages, base, routing,... just add -d16")

xemles, 2021-03-23:
Yup, it did install !
I'll keep the issue opened just in case I don't manage to make the Stick work, I'll close it if I manage to make it work

xemles, 2021-03-23:
Weird thing, since the Stick works, the two other gateways appear down
[image]

Ysurac, 2021-03-23:
I will fix packages URL for next release.
For the gateway this can be normal (sometimes usb dongle doesn't allow gateway ping), or not.
If not, check the macvlan is not enabled where it shouldn't and that interfaces are wan firewall zone.

xemles, 2021-03-23:
> I will fix packages URL for next release.
> For the gateway this can be normal (sometimes usb dongle doesn't allow gateway ping), or not.
> If not, check the macvlan is not enabled where it shouldn't and that interfaces are wan firewall zone.

Well, that's the thing, the only one replying is the USB Dongle now, the two others are actual routers

xemles, 2021-03-23:
I have no idea what happened but after putting a backup of what I believe were the exact same settings and adding the interface again, it worked.

closed 2021-03-23 by xemles