Ysurac/openmptcprouter #1687: Comportemennt instable

bebertb2r, 2021-03-23:
## Expected Behavior

Bonjour le comportement n'est pas stable dans le temps, n'utilise qu'un seul port WAN au bout de quelques heures

## Current Behavior

Bonjour,
Je ne comprend pas vraiment ce qu'il se passe

J'ai un routeur huawei sur réseau FREE
un routeur huawei sur réseau SFR
un routeur ADSL SFR au débit inhumain de 8mb
un VPS OVH, débit max 500mb

Aléatoirement, je reste "scotché" sur un des 3 WAN, le hasard ou la fatalité ayant une préférence pour le WAN ADSL.
Également, les test de débits m'indiquent aléatoirement un provider différent, rarement mon fournisseur VPS (OVH).
La répartition de l'utilisation des wan m'interroge car l'utilisation est inversement proportionnelle au débit  (Je viens de redémarrer mais elle a été répartie de la même manière après une journée entière de télétravail)

ADSL Reçu: 37.90 MB (30580 Pqts.)
4G FREE Reçu: 2.99 MB (4384 Pqts.)
4G SFR Reçu: 6.75 MB (9422 Pqts.)
omrvpm Reçu: 1.07 MB (3581 Pqts.)
LAN Reçu: 59.15 MB (87997 Pqts.)

J'ai décoché l'option  "Activer la faible latence TCP"
Les DCHP sont tous désactivés sur les routeurs
Parfois aprés un redémarrage, tout fonctionne correctement, c'est à dire que je me vois de l’extérieur avec mon IP OVH, et les débits sont bien aggrégrés.
Tout est OK dans l'aperçu réseau (à part votre ip n'a pas été attribué par ce routeur, bien que pas d'autre dhcp que OMR)

Avez vous une idée de comment résoudre ce problème ?
Merci

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

Ysurac, 2021-03-23:
La WAN ayant la plus faible latence est utilisé en priorité, puis si la connexion est saturée celle avec une latence plus élevée et ainsi de suite.
Pour tester le débit agrégé il y a ```omr-test-speed``` sur le routeur, accessible via SSH ou Services->Terminal (ctrl+c pour arrêter après au moins 2 minutes), ```omr-test-speed wan1``` permet de tester la vitesse que de wan1.
Si l'IP externe n'est pas celle du VPS alors c'est que le proxy et le VPN ne fonctionnent pas, ce qui est visible dans la page status. Il y a également des informations sur la raison dans les logs dans Etat->Journal Système.

bebertb2r, 2021-03-23:
Tout est OK dans la page statut, la latence est la moins élevé sur les interfaces 4G pour autant c'est l'ADSL qui est utilisé.

J'ai redémarré et j'ai passé ma 4G FREE en mode Master, et la l'aggrégation à refonctionné, je me voyais bien en ip OVH
J'ai fait des dizaine de tests, et surveiller ma log, mon ip est redevenu en "free mobile".
Tout les WANs sont quand même exploités.
Le logs n'ont pas bougés entre temps. (le dernier  Tue Mar 23 21:08:14 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0)

J'attend de voir sur la durée si tout mes WANs continuent a être exploités. J'observe.
En tout cas je salue le boulot qui a été fait et la qualité de cet outil.

(la capture ci dessous a été fait avant le changement )
[image]

Ysurac, 2021-03-24:
IPv6 est bien désactivé pour tous les WANs ?

bebertb2r, 2021-03-24:
OUI IPv6 désactivé sur tous les routeurs,
Ce soir même constat, j'ai perdu l'aggrégation et IP non VPS OVH

Ysurac, 2021-03-24:
Et quand il y a perte la page status reste tout en vert avec l'IP du VPS affiché ?
La page status fait les tests en direct, donc si elle arrive bien à avoir l'IP du VPS dans ce cas le soucis n'est pas au niveau du routeur. Il n'y a pas un autre serveur DHCP activé ? Il faudrait vérifer la passerelle utilisée sur le PC client quand ça arrive.

bebertb2r, 2021-03-25:
Problème résolu via un remontage complet de ma VM et reparamètrage

closed 2021-03-25 by bebertb2r