Ysurac/openmptcprouter #1688: Problème d'instabilité de liaison entre le routeur et le VPS

shinjii, 2021-03-24:
## Expected Behavior
Bonjour a tous,
J'ai un problème d'instabilité de liaison entre le routeur et le vps.

## Current Behavior
Alors,
le VPS est chez OVH.
ca fait environ 2 ans que je suis sous OMR et j'ai jamais eu de souci comme ca.
je m'explique:

depuis que j'ai récupère mon vps ovh suite aux incendie j'ai un problème de perte de liaison entre OMR et VPS.
Aprés un reboot du routeur ca fonction correctement quelques heures et ensuite ca recommence.
voici une capture du problème :
[image]
ca varie de connecte a non connecte (le ipv6 était juste un test, le pb est le même sans)

je n'ai pas l'impression que le vps soit en cause car un simple reboot du routeur ou un changement de paramètre avec réinitialisation de la liaison suffi a retrouver de la stabilité pour quelques heures.

j'ai changer le disque dur hier et réinstalle OMR avec la sauvegarde faite juste avant, même résultat.

la seule chose que je n'ai pas faite c est de réinstaller le vps (le mien chez ovh n'avais pas brulé mais avait juste été éteins)

j'ai du rebooter ce matin pour avoir une connexion correcte pour faire ce poste, donc si vous voulez des log, dites moi les quelles et je vous donne ca ce soir.

Merci pour votre aide, car la je ne comprend vraiment pas.

## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: 0.1024
  - OpenMPTCProuter platform: X86

Ysurac, 2021-03-24:
Il faudrait voir dans Etat->Journal Système ce qui est détecté.

shinjii, 2021-03-24:
ok, je fais passer ca ce soir, la c 'est stable pour le moment suite au reboot.

shinjii, 2021-03-24:
en cas j'ai deja ca si ca peux aider:

Wed Mar 24 06:51:57 2021 user.notice OMR-VPS: Can't get vps token, try later
Wed Mar 24 06:51:58 2021 daemon.info omr-tracker-v2ray: Server (51.77.145.163) seems down, no answer to ping
Wed Mar 24 06:52:07 2021 daemon.info dnsmasq-dhcp[20251]: DHCPREQUEST(eth0) 192.168.100.54 60:01:94:e0:50:4b
Wed Mar 24 06:52:07 2021 daemon.info dnsmasq-dhcp[20251]: DHCPACK(eth0) 192.168.100.54 60:01:94:e0:50:4b 054-Charge_Tablette
Wed Mar 24 06:52:07 2021 daemon.warn dnsmasq-dhcp[20251]: not giving name localhost.lan to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:52:07 2021 daemon.warn dnsmasq-dhcp[20251]: not giving name localhost to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:52:19 2021 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 74.82.42.42)
Wed Mar 24 06:52:19 2021 daemon.info omr-tracker-v2ray: Reload V2Ray rules
Wed Mar 24 06:52:19 2021 user.notice v2ray: Rules UP
Wed Mar 24 06:52:19 2021 user.notice v2ray: Reload omr-bypass rules
Wed Mar 24 06:52:19 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Mar 24 06:52:34 2021 user.notice omr-bypass: Reload dnsmasq...
Wed Mar 24 06:52:34 2021 daemon.info dnsmasq[20251]: read /etc/hosts - 4 addresses
Wed Mar 24 06:52:34 2021 daemon.info dnsmasq[20251]: read /tmp/hosts/odhcpd - 5 addresses
Wed Mar 24 06:52:34 2021 daemon.info dnsmasq[20251]: read /tmp/hosts/dhcp.cfg01411c - 54 addresses
Wed Mar 24 06:52:34 2021 daemon.info dnsmasq-dhcp[20251]: read /etc/ethers - 0 addresses
Wed Mar 24 06:52:34 2021 daemon.warn dnsmasq-dhcp[20251]: not giving name localhost.lan to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:52:34 2021 daemon.warn dnsmasq-dhcp[20251]: not giving name localhost to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:52:34 2021 user.notice omr-bypass: OMR-ByPass is running
Wed Mar 24 06:52:41 2021 daemon.info dnsmasq-dhcp[20251]: DHCPREQUEST(eth0) 192.168.100.23 60:01:94:ab:17:00
Wed Mar 24 06:52:41 2021 daemon.info dnsmasq-dhcp[20251]: DHCPACK(eth0) 192.168.100.23 60:01:94:ab:17:00 023-Volets1
Wed Mar 24 06:52:41 2021 daemon.warn dnsmasq-dhcp[20251]: not giving name localhost.lan to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:52:41 2021 daemon.warn dnsmasq-dhcp[20251]: not giving name localhost to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:52:54 2021 daemon.info dnsmasq-dhcp[20251]: DHCPREQUEST(eth0) 192.168.100.14 38:aa:3c:6f:16:09
Wed Mar 24 06:52:54 2021 daemon.info dnsmasq-dhcp[20251]: DHCPACK(eth0) 192.168.100.14 38:aa:3c:6f:16:09 014-PlayerChambreParentale
Wed Mar 24 06:52:54 2021 daemon.warn dnsmasq-dhcp[20251]: not giving name localhost.lan to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:52:54 2021 daemon.warn dnsmasq-dhcp[20251]: not giving name localhost to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:53:17 2021 user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps, no server API answer)
Wed Mar 24 06:53:32 2021 user.notice SQM: Stopping SQM on wan1
Wed Mar 24 06:53:32 2021 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev wan1 ingress
Wed Mar 24 06:53:32 2021 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Wed Mar 24 06:53:32 2021 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev wan1 root
Wed Mar 24 06:53:32 2021 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Wed Mar 24 06:53:32 2021 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev ifb4wan1 root
Wed Mar 24 06:53:32 2021 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Wed Mar 24 06:53:32 2021 user.notice ModemManager: hotplug: remove network interface ifb4wan1: event processed
Wed Mar 24 06:53:32 2021 user.notice ModemManager: hotplug: event reported: action=remove, name=ifb4wan1, subsystem=net
Wed Mar 24 06:53:32 2021 user.notice SQM: Starting SQM script: simple.qos on wan1, in: 0 Kbps, out: 0 Kbps
Wed Mar 24 06:53:32 2021 user.notice SQM: Using generic sqm_start_default function.
Wed Mar 24 06:53:33 2021 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Wed Mar 24 06:53:33 2021 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Wed Mar 24 06:53:33 2021 daemon.warn [6022]: <warn>  [TMP_IFB_4_SQM] invalid sysfs path read for net/TMP_IFB_4_SQM
Wed Mar 24 06:53:33 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Mar 24 06:53:33 2021 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Wed Mar 24 06:53:33 2021 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Wed Mar 24 06:53:33 2021 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
[... cut ...]
Wed Mar 24 06:54:02 2021 daemon.info v2ray[18030]: A unified platform for anti-censorship.
Wed Mar 24 06:54:02 2021 daemon.info v2ray[18030]: 2021/03/24 05:54:02 [Info] v2ray.com/core/common/platform/ctlcmd: <v2ctl message>
Wed Mar 24 06:54:02 2021 daemon.info v2ray[18030]: v2ctl> Read config:  /var/etc/v2ray/v2ray.main.json
Wed Mar 24 06:54:02 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Mar 24 06:54:04 2021 user.notice omr-tracker: Launched
Wed Mar 24 06:54:04 2021 daemon.info glorytun: starting glorytun vpn instance vpn

shinjii, 2021-03-24:
alors ca vient de le refaire, voila les log:

Wed Mar 24 06:55:03 2021 daemon.info dnsmasq[5414]: using only locally-known addresses for domain local
Wed Mar 24 06:55:03 2021 daemon.info dnsmasq[5414]: using only locally-known addresses for domain invalid
Wed Mar 24 06:55:03 2021 daemon.info dnsmasq[5414]: using only locally-known addresses for domain bind
Wed Mar 24 06:55:03 2021 daemon.info dnsmasq[5414]: using only locally-known addresses for domain use-application-dns.net
Wed Mar 24 06:55:03 2021 daemon.info dnsmasq[5414]: using nameserver 127.0.0.1#5353
Wed Mar 24 06:55:03 2021 daemon.info dnsmasq[5414]: using only locally-known addresses for domain lan
Wed Mar 24 06:55:03 2021 daemon.info dnsmasq[5414]: read /etc/hosts - 4 addresses
Wed Mar 24 06:55:03 2021 daemon.info dnsmasq[5414]: read /tmp/hosts/odhcpd - 5 addresses
Wed Mar 24 06:55:03 2021 daemon.info dnsmasq[5414]: read /tmp/hosts/dhcp.cfg01411c - 54 addresses
Wed Mar 24 06:55:03 2021 daemon.info dnsmasq-dhcp[5414]: read /etc/ethers - 0 addresses
Wed Mar 24 06:55:03 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost.lan to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:55:03 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:55:03 2021 user.notice omr-tracker: Launching...
Wed Mar 24 06:55:06 2021 user.notice omr-tracker: Launched
Wed Mar 24 06:55:07 2021 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.1.1.1)
Wed Mar 24 06:55:11 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Wed Mar 24 06:55:13 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Wed Mar 24 06:55:55 2021 daemon.info dnsmasq-dhcp[5414]: DHCPREQUEST(eth0) 192.168.100.14 38:aa:3c:6f:16:09
Wed Mar 24 06:55:55 2021 daemon.info dnsmasq-dhcp[5414]: DHCPACK(eth0) 192.168.100.14 38:aa:3c:6f:16:09 014-PlayerChambreParentale
Wed Mar 24 06:55:55 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost.lan to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:55:55 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:56:55 2021 daemon.info dnsmasq-dhcp[5414]: DHCPREQUEST(eth0) 192.168.100.14 38:aa:3c:6f:16:09
Wed Mar 24 06:56:55 2021 daemon.info dnsmasq-dhcp[5414]: DHCPACK(eth0) 192.168.100.14 38:aa:3c:6f:16:09 014-PlayerChambreParentale
Wed Mar 24 06:56:55 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost.lan to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:56:55 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:57:55 2021 daemon.info dnsmasq-dhcp[5414]: DHCPREQUEST(eth0) 192.168.100.14 38:aa:3c:6f:16:09
Wed Mar 24 06:57:55 2021 daemon.info dnsmasq-dhcp[5414]: DHCPACK(eth0) 192.168.100.14 38:aa:3c:6f:16:09 014-PlayerChambreParentale
Wed Mar 24 06:57:55 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost.lan to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:57:55 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:58:27 2021 daemon.info dnsmasq-dhcp[5414]: DHCPREQUEST(eth0) 192.168.100.7 e8:5a:8b:e2:9e:91
Wed Mar 24 06:58:27 2021 daemon.info dnsmasq-dhcp[5414]: DHCPACK(eth0) 192.168.100.7 e8:5a:8b:e2:9e:91 007-RedmiNote9Pro
Wed Mar 24 06:58:27 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost.lan to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:58:27 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:58:32 2021 daemon.info dnsmasq-dhcp[5414]: DHCPREQUEST(eth0) 192.168.100.7 e8:5a:8b:e2:9e:91
Wed Mar 24 06:58:32 2021 daemon.info dnsmasq-dhcp[5414]: DHCPACK(eth0) 192.168.100.7 e8:5a:8b:e2:9e:91 007-RedmiNote9Pro
Wed Mar 24 06:58:32 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost.lan to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:58:32 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:58:55 2021 daemon.info dnsmasq-dhcp[5414]: DHCPREQUEST(eth0) 192.168.100.14 38:aa:3c:6f:16:09
Wed Mar 24 06:58:55 2021 daemon.info dnsmasq-dhcp[5414]: DHCPACK(eth0) 192.168.100.14 38:aa:3c:6f:16:09 014-PlayerChambreParentale
Wed Mar 24 06:58:55 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost.lan to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 06:58:55 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
[... cut ...]
Wed Mar 24 08:02:34 2021 daemon.info dnsmasq-dhcp[5414]: DHCPREQUEST(eth0) 192.168.100.2 ac:9e:17:83:4f:17
Wed Mar 24 08:02:34 2021 daemon.info dnsmasq-dhcp[5414]: DHCPACK(eth0) 192.168.100.2 ac:9e:17:83:4f:17 002-ShinjiiPC
Wed Mar 24 08:02:34 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost.lan to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1
Wed Mar 24 08:02:34 2021 daemon.warn dnsmasq-dhcp[5414]: not giving name localhost to the DHCP lease of 192.168.100.177 because the name exists in /etc/hosts with address 127.0.0.1

Ysurac, 2021-03-24:
Dans Système->OpenMPTCProuter, il faudrait essayer de mettre WAN1 en maitre et wan2 en activé.

shinjii, 2021-03-24:
Tu crois que le problème pourrait juste venir de ça ?

shinjii, 2021-03-25:
Bon ben apparemment ca l'air d 'être ok en inversant le master du wan 1 et 2.

Bon j'ai maintenant un second problème, je pensait qu'il était lié au premier mais ca n'as pas l'air d être le cas.
je m'explique : 

après un reboot j'arrive a aller sur le site d'amazon.fr, l'application mobile Android fonctionne aussi et prime sur mes boitier Android également (ceux ci ont un bypass pour passer que sur le modem 4g, sinon prime détecte un proxy )

Apres quelques heures tout cela n'est plus accessible, j 'ai du "ERR_CONNECTION_TIMED_OUT" pour le site amazon.fr, l'application mobile me dit qu'elle n'arrive pas a se connecter et prime me dit de contrôler ma connexion internet.

internet est ok pour le reste , ip tv ect.

une idée?

ps : Promis après celui la c 'est fini -_-'

Ysurac, 2021-03-25:
Il faut vérifier que le DNS est bien l'IP d'OpenMPTCProuter partout.

shinjii, 2021-03-25:
Ok je vérifie ça sur chaque appareils? 
Pourtant c est openMPTCProuteur mon DHCP, donc il devrai me distribuer le bon DNS.
C est possible que ça change au fil du temps? Car après reboot tout rentre dans l' ordre mais ça dure que quelques heures tout au plus.

shinjii, 2021-03-25:
dans la conf réseau le dns est ok. c'est bien celui de openMPTCProuteur.

a cet instant amazon.fr marche sur mon pc mais pas sur mon téléphone par exemple alors que le dns est ok sur les 2.
j'ai passé mon tel en ipfixe et mi le même dns en 1 et 2 mais toujours pas de amazon.fr...

j'ai du mal a comprendre , et ca se trouve ca marchera plus sur le pc dans quelque instant...

closed 2021-06-28 by github-actions[bot]