Ysurac/openmptcprouter #1689: Journal système

gblanchet03, 2021-03-24:
@Ysurac 
J'ai ce genre de message dans le journal système. Comment empêcher cela ?
Wed Mar 24 09:44:11 2021 authpriv.warn dropbear[4808]: Login attempt for nonexistent user
Wed Mar 24 09:44:12 2021 authpriv.info dropbear[4808]: Exit before auth from <177.155.248.68:33402>: Disconnect received
Wed Mar 24 09:44:21 2021 authpriv.info dropbear[6250]: Child connection from 221.181.185.198:56981
Wed Mar 24 09:44:22 2021 authpriv.info dropbear[6250]: Exit before auth from <221.181.185.198:56981>: Disconnect received
Wed Mar 24 09:44:25 2021 authpriv.info dropbear[6863]: Child connection from 213.6.130.133:38224
Wed Mar 24 09:44:28 2021 authpriv.warn dropbear[6863]: Bad password attempt for 'root' from 213.6.130.133:38224
Wed Mar 24 09:44:28 2021 authpriv.info dropbear[6863]: Exit before auth from <213.6.130.133:38224>: (user 'root', 1 fails): Disconnect received
Wed Mar 24 09:44:46 2021 authpriv.info dropbear[9760]: Child connection from 188.166.185.16:46480
Wed Mar 24 09:44:47 2021 authpriv.warn dropbear[9760]: Bad password attempt for 'root' from 188.166.185.16:46480
Wed Mar 24 09:44:48 2021 authpriv.info dropbear[9760]: Exit before auth from <188.166.185.16:46480>: (user 'root', 1 fails): Disconnect received
Wed Mar 24 09:45:07 2021 authpriv.info dropbear[12882]: Child connection from 121.5.79.2:50782
Wed Mar 24 09:45:14 2021 authpriv.warn dropbear[12882]: Login attempt for nonexistent user
Wed Mar 24 09:45:15 2021 authpriv.info dropbear[12882]: Exit before auth from <121.5.79.2:50782>: Disconnect received

Cordialement.
Gérard
OMR sur OTB+

Ysurac, 2021-03-24:
En ne redirigeant pas vers le port 22 d'OpenMPTCProuter. Il y a certainement un soucis de configuration au niveau du firewall, par défaut tout est bloqué en entrée depuis le VPS et depuis les WANs.

gblanchet03, 2021-03-24:
J'ai désactivé la redirection du port 22. Et le sifflet a été coupé !!
En fait je n'ai pas besoin d'un accès SSH depuis le VPS vers OMR. 
Merci et bonne journée.
Gérard

closed 2021-03-25 by Ysurac