Ysurac/openmptcprouter #1691: Unable to access GUI with 0.57.3 VM Vmware - Get message "ERR_CONNECTION_REFUSED"

zhang91-fr, 2021-03-24:
## Expected Behavior
I should get login page from this address "http://192.168.100.1/cgi-bin/luci/"

## Current Behavior
I get message "ERR_CONNECTION_REFUSED"
I used the file named : openmptcprouter-v0.57.3-r0+15225-bfc433efd4-x86-64-generic-ext4-combined.vmdk
## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: Not installed
  - OpenMPTCProuter platform: Vmware workstation 16.0 running on Windows 10 Pro
  - Virtual network NAT (subnet address 192.168.100.0) sharing the same network adaptor below
  - Host network physical address : 192.168.200.0/24 - Gateway 192.168.200.1
  - I can reach the VM router by ping but I cannot access to GUI interface through a browser on the host (Chrome)
  - The VM router is loaded and got shell with command line prompt  - root@(none):/#

Adorfer, 2021-03-24:
have you tried manually passing to https? 
(at least in openwrt there is some kind of issue if luci-ssl/https is available on the router)

zhang91-fr, 2021-03-24:
> have you tried manually passing to https?
> (at least in openwrt there is some kind of issue if luci-ssl/https is available on the router)

I did. I got the same message http or https.

Ysurac, 2021-03-25:
Command line shouldn't be "root@none", should be "root@OpenMPTCProuter". So the image seems to be not correctly initialized.

zhang91-fr, 2021-03-26:
> Command line shouldn't be "root@none", should be "root@OpenMPTCProuter". So the image seems to be not correctly initialized.

I'm using a different image and I got the same status at root.
I've change hypervisor and using Hyper V with the right image.
Now, command line prompt is now it should be.
I got "connexion refused" when I am trying to get login GUI though the host.
VM share the host's same network card using NAT
I am wondering if the GUI accept connection from client though Nat connection ?

closed 2021-07-01 by github-actions[bot]