Ysurac/openmptcprouter #1693: Aggregate an android phone in access point or USB

Gaou3, 2021-03-25:
Hello, and thank you for your work on this project ! I have two 4G connection and 1 ADSL, and everything works perfectly ! (I had a lot of problems due to a defective SD card, and I took a lot of time to understand this SC card was the problem ... but after that, everything works fine !)

Ok, my new request is I want to add a smartphone as a new router. I need a step by step tutorial because I'm a newby in networking, and I don't find the way to do this easily on the web...

I read that USB connection is better than Wifi Access Point. Is it true ?

How can I connect my smartphone to OMR as a new router USB / WiFi Acces Point ? How can I put a static IP to my phone (I know how to do this if I have to connect my phone to a Wifi box, but I don't know how to do this if it's my phone the Wifi router ...) ? What is the process ? If there is a tutorial, can you send it to me ?

Thank you

## Expected Behavior
I want to aggregate an android phone as easily as a new router.

## Current Behavior
I don't know the way to do this aggregation

## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: 0.57.3
  - OpenMPTCProuter platform: RPI4

Ysurac, 2021-03-29:
You can check this doc: https://openwrt.org/docs/guide-user/network/wan/smartphone.usb.tethering
All needed packages are already installed.

Gaou3, 2021-03-30:
Thank you, but when I plug my phone with USB cable to the raspberry, I can't activate the USB tethering on my phone, the option to check is grey ...
It's like my phone is not detected by raspberry.
You said me that all needed packages are already installed to detect the USB, but do I have to do something ?

Ysurac, 2021-03-30:
Did you try tethering on a computer ?
You can also check that data is allowed on your phone and with another USB cable.

Gaou3, 2021-03-30:
Yes, I tried tethering on my computer and it's run. No problem, I can check USB tethering.
But I can't do it when the same USB cable is plugged into my raspberry.

Ysurac, 2021-03-30:
You can check in State->kernel log what is detected when the phone is plugged.

Gaou3, 2021-03-30:
Ok, my phone is plugged, and this is my kernel log:

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.83 (runner@fv-az50-423) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0+15225-bfc433efd4)) #0 SMP Tue Dec 15 19:35:00 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12032 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000] percpu: Embedded 21 pages/cpu s45144 r8192 d32680 u86016
[    0.000000] pcpu-alloc: s45144 r8192 d32680 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996912
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:EF:0F:73 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x37400000-0x3b400000] (64MB)
[    0.000000] Memory: 3828792K/4050944K available (7038K kernel code, 430K rwdata, 1840K rodata, 512K init, 539K bss, 156616K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x2dc/0x3dc with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000157] Console: colour dummy device 80x25
[    0.000396] printk: console [tty1] enabled
[    0.000428] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000450] pid_max: default: 32768 minimum: 301
[    0.000642] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.000725] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001382] Disabling memory control group subsystem
[    0.002076] ASID allocator initialised with 32768 entries
[    0.002168] rcu: Hierarchical SRCU implementation.
[    0.002782] smp: Bringing up secondary CPUs ...
[    0.003174] Detected PIPT I-cache on CPU1
[    0.003225] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.003667] Detected PIPT I-cache on CPU2
[    0.003701] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.004199] Detected PIPT I-cache on CPU3
[    0.004232] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.004301] smp: Brought up 1 node, 4 CPUs
[    0.004329] SMP: Total of 4 processors activated.
[    0.004343] CPU features: detected: 32-bit EL0 Support
[    0.004356] CPU features: detected: CRC32 instructions
[    0.004407] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.004425] CPU: All CPU(s) started at EL2
[    0.004453] alternatives: patching kernel code
[    0.005593] devtmpfs: initialized
[    0.012504] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.012543] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.014288] pinctrl core: initialized pinctrl subsystem
[    0.014570] reg-dummy reg-dummy: no of_node; not parsing pinctrl DT
[    0.015171] NET: Registered protocol family 16
[    0.016536] DMA: preallocated 1024 KiB pool for atomic allocations
[    0.017078] cpuidle: using governor ladder
[    0.017109] cpuidle: using governor menu
[    0.017380] Serial: AMBA PL011 UART driver
[    0.018860] bcm2835-mbox fe00b880.mailbox: mailbox enabled
[... 754 lines cut ...]
[88069.328399] pstate: 60400005 (nZCv daif +PAN -UAO)
[88069.333185] pc : dst_release+0xb4/0xc0
[88069.336925] lr : dst_release+0xb4/0xc0
[88069.340663] sp : ffffffc01006b8e0
[88069.343968] x29: ffffffc01006b8e0 x28: ffffff80f6060000 
[88069.349271] x27: ffffffc010629928 x26: ffffff80f1fc0800 
[88069.354573] x25: ffffff80f36d7100 x24: 0000000000000008 
[88069.359876] x23: ffffffc01006b9d0 x22: ffffff80f36d7100 
[... cut ...]
[88069.396995] x9 : ffffffc01006b8e0 x8 : 65646e7520657361 
[88069.402297] x7 : 656c65725f747364 x6 : ffffffc010a33235 
[88069.407600] x5 : 00ffffffffffffff x4 : 0000000000000000 
[88069.412902] x3 : 0000000000000000 x2 : 00000000ffffffff 
[88069.418205] x1 : ffffffc0e6fff000 x0 : 0000000000000015 
[88069.423508] Call trace:
[88069.425946]  dst_release+0xb4/0xc0
[88069.429361]  0xffffffc008fcd260
[88069.432493]  nf_hook_slow+0x50/0xf8
[88069.435974]  ip_sublist_rcv+0x7c/0x1e0
[88069.439713]  ip_

Ysurac, 2021-03-30:
Can you try to plug it when OpenMPTCProuter is already running ? and check kernel log at this time.

Gaou3, 2021-03-31:
I don't understand ... OpenMPTCProuter is running since 8 days ...
[image]

Ysurac, 2021-03-31:
Unplug the phone from USB, then plug it again and check log.

Gaou3, 2021-03-31:
ok:

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.83 (runner@fv-az50-423) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0+15225-bfc433efd4)) #0 SMP Tue Dec 15 19:35:00 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12032 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000] percpu: Embedded 21 pages/cpu s45144 r8192 d32680 u86016
[    0.000000] pcpu-alloc: s45144 r8192 d32680 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996912
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:EF:0F:73 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x37400000-0x3b400000] (64MB)
[    0.000000] Memory: 3828792K/4050944K available (7038K kernel code, 430K rwdata, 1840K rodata, 512K init, 539K bss, 156616K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x2dc/0x3dc with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000157] Console: colour dummy device 80x25
[    0.000396] printk: console [tty1] enabled
[    0.000428] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000450] pid_max: default: 32768 minimum: 301
[    0.000642] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.000725] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001382] Disabling memory control group subsystem
[    0.002076] ASID allocator initialised with 32768 entries
[    0.002168] rcu: Hierarchical SRCU implementation.
[    0.002782] smp: Bringing up secondary CPUs ...
[    0.003174] Detected PIPT I-cache on CPU1
[    0.003225] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.003667] Detected PIPT I-cache on CPU2
[    0.003701] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.004199] Detected PIPT I-cache on CPU3
[    0.004232] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.004301] smp: Brought up 1 node, 4 CPUs
[    0.004329] SMP: Total of 4 processors activated.
[    0.004343] CPU features: detected: 32-bit EL0 Support
[    0.004356] CPU features: detected: CRC32 instructions
[    0.004407] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.004425] CPU: All CPU(s) started at EL2
[    0.004453] alternatives: patching kernel code
[    0.005593] devtmpfs: initialized
[    0.012504] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.012543] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.014288] pinctrl core: initialized pinctrl subsystem
[    0.014570] reg-dummy reg-dummy: no of_node; not parsing pinctrl DT
[    0.015171] NET: Registered protocol family 16
[    0.016536] DMA: preallocated 1024 KiB pool for atomic allocations
[    0.017078] cpuidle: using governor ladder
[    0.017109] cpuidle: using governor menu
[    0.017380] Serial: AMBA PL011 UART driver
[    0.018860] bcm2835-mbox fe00b880.mailbox: mailbox enabled
[... 754 lines cut ...]
[88069.328399] pstate: 60400005 (nZCv daif +PAN -UAO)
[88069.333185] pc : dst_release+0xb4/0xc0
[88069.336925] lr : dst_release+0xb4/0xc0
[88069.340663] sp : ffffffc01006b8e0
[88069.343968] x29: ffffffc01006b8e0 x28: ffffff80f6060000 
[88069.349271] x27: ffffffc010629928 x26: ffffff80f1fc0800 
[88069.354573] x25: ffffff80f36d7100 x24: 0000000000000008 
[88069.359876] x23: ffffffc01006b9d0 x22: ffffff80f36d7100 
[88069.365178] x21: ffffff80ef2a3e00 x20: 00000000ffffffff 
[... cut ...]
[88069.396995] x9 : ffffffc01006b8e0 x8 : 65646e7520657361 
[88069.402297] x7 : 656c65725f747364 x6 : ffffffc010a33235 
[88069.407600] x5 : 00ffffffffffffff x4 : 0000000000000000 
[88069.412902] x3 : 0000000000000000 x2 : 00000000ffffffff 
[88069.418205] x1 : ffffffc0e6fff000 x0 : 0000000000000015 
[88069.423508] Call trace:
[88069.425946]  dst_release+0xb4/0xc0
[88069.429361]  0xffffffc008fcd260
[88069.432493]  nf_hook_slow+0x50/0xf8
[88069.435974]  ip_sublist_rcv+0x7c/0x1e0
[88069.439713]  ip_

Gaou3, 2021-05-06:
I managed to create a USB tethering with one smartphone, but is it possible to create a second usb tethering with a second smartphone connected to the raspberry?
Because when I connect a second smartphone with a second Tethering interface, they share the same IP ...

closed 2021-08-10 by github-actions[bot]