Ysurac/openmptcprouter #1694: Using two ncm usb lte connections not working

geeuk, 2021-03-25:
Hi,

I have spent days trying to use two lte me909s usb pcie cards. The issue I have is only one device works at a time as one uses only usb0 with ttyusb02 and other which i choose ttyusb05 always says device not ready and stuck on ncm-wan2 somehow i can't get one more usb1 working. After i reboot sometimes it fights to get choosen which modem to pick.

Is there any instruction to setup two lte modems via usb.

Ysurac, 2021-03-26:
Check kernel log.

geeuk, 2021-03-27:
hi sorry it was the issue with power supply and current of max 1.2Amp.. It worked with external psu but i need 12v powered hub because even if i have 5Amp psu to pi via gpio it still can only draw 1.2Amp and ME909s is very power hungry.

closed 2021-06-30 by github-actions[bot]