Ysurac/openmptcprouter #1695: OMR-Bypass configuration for local streaming services

codex70, 2021-03-26:
## Expected Behavior

When using the OpenMPTCProuter, there are certain services that I would like to be available directly through a specific WAN connection rather than going through the VPN tunnel.

Some of the services I need to use from my local provider are geo-blocked and I would like to be able to access those services directly through one of the specific WAN interfaces. In particular, there are streaming services from my internet service provider that I can no longer access as the VPS is hosted in a different country.

## Current Behavior

All traffic is going through the VPS and local geo-restricted services are no longer available.

I believe it should be possible to use OMR-Bypass to achieve what I am trying to set up, but I'm not sure how to do this. I have entered the specific domains into the OMR-Bypass configurations, but this doesn't seem sufficient. I also see that you can use OMR-Bypass for similar services (such as Netflix), but obviously the local language services I would like to use are not available:
[image]

## Specifications
  - OpenMPTCProuter version: 5.4.86-mptcp 0.1024
  - OpenMPTCProuter VPS version: 0.1024 5.4.86-mptcp
  - OpenMPTCProuter platform: x86_64
  
[image]

I am using a wifi mesh router which has been assigned a static IP address of 192.168.100.118 that then acts as the DHCP server for the rest of the network. The reason for this is because I have a number of specific rules and configurations set up on this router which would be complicated to replicate.

I have set the OpenMPTCProuter to be the default gateway and DNS server for that router.

Ysurac, 2021-03-26:
If you use some VLAN there is an issue in current release.
Else check again the DNS is the good one, if yes add a domain like "ifconfig.co" and check that you have WAN public ip.
If yes, then the service may use some other domain than what you set, use your web browser developer tools to find what it use.

codex70, 2021-03-26:
Thanks for the quick reply. If I have set this up correctly:
[image]
then it isn't working as it's still showing UK and should show Spain:
[image]
I will take another look at the configuration, try again and will let you know.

Ysurac, 2021-03-26:
And IP is VPS one ?
Also changes can take some times if you visited the website before, because DNS request can be in a cache.

codex70, 2021-03-26:
You are right, using a different browser this is now showing Spain.

Ysurac, 2021-03-26:
Cool.
So it's only wrong domains set for your streaming services. It's not always easy to find what domain they use... As I said, I use browser developer tools, in network tab there is all domains that the website call.

codex70, 2021-03-26:
Thanks, I will look at that and let you know how I get on.

codex70, 2021-03-27:
Hi, you've been amazingly helpful so far, and it looks like I've nearly got it working. In the browser it mostly works (I think I'm still missing some domains), but with an android app, I have to resort to using a packet capture tool and adding IP addresses. This has been less successful, but could probably get it working with more effort.

With that in mind, is it possible to add wildcards for domain names eg:
[image]
and if so, what is the format as the above didn't seem to help.

Also if I have a range of IP addresses, for example:
[image]
what is the format I should use?

It may be, in the end, that it's too easily broken by them changing domain names and IP addresses, but I seem to be pretty close now.

codex70, 2021-03-29:
Another quick question, if I want to use the packet capture utility to see the destination IP addresses of the service I am trying to connect to, which interface should I use?

[image]

I've made good progress with the android app, and with the browser, but then streaming to a chromecast or using the apps on the TV I still have problems. I tried using the packets captured from "eth0", but this didn't seem to help.

Ysurac, 2021-03-29:
No need for wilcard. To bypass all from *.amazon.com, you only need to set "amazon.com" for example.
To find domains, you can use Network->DHCP&DNS, "Log queries". This will be easier/better than packet capture.

codex70, 2021-03-29:
Fantastic, thanks again so much for all of your help. That makes much more sense. I now don't need any of the IP addresses I was using and also managed to significantly reduce the number of domains. Most importantly, it appears to be working on all of my devices.

Massive thanks for such quick and helpful responses!

closed 2021-07-02 by github-actions[bot]