Ysurac/openmptcprouter #1696: Monitoring Jeedom

JeedHome44, 2021-03-26:
**Is your feature request related to a problem? Please describe.**
 Faire un plugin Jeedom pour monitorer OMR sur Jeedom 

**Describe the solution you'd like**
 Cela permettrait d'automatiser le reboot d'un wan si problème... avoir des notifications autre que mail pour OMR par Jeedom...

J'ai déjà la Freebox et le routeur Huawei sur Jeedom. Il me manque OMR! 

**Describe alternatives you've considered**

**Additional context**

Ysurac, 2021-03-26:
Je n'utilise pas Jeedom actuellement, donc il y a peu de chance que je fasse un plugin prochainement. Désolé.
Par contre il y a une API sur OpenMPTCProuter et j'ai l'intention de l'enrichir prochainement, donc faire ce genre de chose sera de plus en plus simple.

JeedHome44, 2021-03-26:
Ah donc c'est possible de faire un script qui interroge toutes les minutes OMR et faire des script pour recharger un wan ou autre?

Ysurac, 2021-03-26:
oui

JeedHome44, 2021-03-26:
Ah tu as un lien pour ça? Ça m'intéresse de faire un virtuel Jeedom avec un cron 1 minute 😉

Ysurac, 2021-03-26:
Tiens je peux ajouter une option pour redémarrer un wan quand down dans omr-tracker aussi...

Ysurac, 2021-03-26:
La doc pour l'API: https://openwrt.org/docs/techref/ubus#access_to_ubus_over_http

JeedHome44, 2021-03-26:
Oh donc on peut voir l'état de chacun des wan et le redémarrer? Ça serait juste top ça! Je vais potasser ca 😉

JeedHome44, 2021-03-27:
Ah c'est pas une API OMR? C'est autre chose non?

Tu pourrais me donner un exemple pour que je comprenne mieux ce qu'il faut que je fasse?

Merci

JeedHome44, 2021-03-28:
> Tiens je peux ajouter une option pour redémarrer un wan quand down dans omr-tracker aussi...

Donc OMR peut se charger tout seul de redémarrer un wan? c'est top ca! comment ca?

Ysurac, 2021-03-28:
J'ai ajouté l'option dans la version beta, dans omr-tracker.

JeedHome44, 2021-03-28:
ok la je vais regarder plus en détail l'api pour comprendre. ca a l'air assez complexe. tu saurais me donner un exemple d'appel d'info du wan1 par exemple par l'api?

JeedHome44, 2021-03-28:
En gros, pour le monitoring sur jeedom, c'est juste voir l'état de chaque wan, du serveur et de OMR. et pouvoir agir dessus dans un premier temps. vois si ca marche comme ca déjà

closed 2021-07-02 by github-actions[bot]